istoreos/istoreos #1095: 端口环回失效，域名映射ip也失效

s949492225, 2023-11-03:
### (1) 具体问题

端口环回失效又开始了，遍寻历史上的解决方案都无效。
退而求其次，用域名映射ip生效了几个小时，发现又不管用了

最终经过反复测试，端口100转发路由80端口

在环回和域名映射都开启的情况下，
用域名：100无法访问
用域名：80可以访问。

### (2) 路由器型号和固件版本

主机名	iStoreOS
型号	Default string Default string
架构	Intel(R) N100
目标平台	x86/64
固件版本	iStoreOS 22.03.5 2023102711 / LuCI istoreos-22.03 branch git-23.271.10849-b445bc4
内核版本	5.10.176
本地时间	2023-11-03 16:17:07
运行时间	17h 11m 51s
平均负载	0.11, 0.08, 0.09

### (3) 详细日志和/或截图
日志不知道怎么搞。

jjm2473, 2023-11-09:
你说的是端口转发吗？是不是在内网测试的？你的域名绑定的是ipv4还是ipv6？

benzo970210, 2023-11-14:
我现在也是这个情况，n100 端口转发不了，但是docker 自带的端口能够访问，我单独转发的端口就无法访问
如图
[image]
但是docker 应用自带的就没问题[home.storemi.cn:8096](url)

s949492225, 2023-11-21:
ipv4

closed 2023-11-27 by s949492225

kuolemax, 2024-01-28:
> ipv4

你好，请问你解决了吗， 我也遇到这个问题了