istoreos/istoreos #1097: Passw-update搭配smartdms导致WX加载巨慢

veen1992, 2023-11-04:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
安装本地安装包Passwall-update版本，搭配smartdns（设置上游dnsmaaq），会在2天内导致网页加载很慢很慢，特别是微信加载巨卡，图片都无法加载，重启就恢复正常，过一段时间又会导致同样的问题，
### (1) 具体问题
Smartdns单独使用设置接管dnsmasq上游，会导致网络卡顿，搭配adguard home（设置adguard home接管dnsmasq接管上游），也会不定期的导致网络卡顿，甚至Adguard Home会卡死无法解析（不跑过滤，重启恢复）
A：

### (2) 路由器型号和固件版本

A：系统版本：
内核：5.10.176
版本：iStoreOS 22.03.5 2023102711
型号1：J4125处理器五口英特尔i226
型号2：i3 4002U 双网口自建设备
### (3) 详细日志和/或截图

A：wan口拨号设置自动获取dns，smartdns设置电信第一dns自动获取的61开头和200开头的dns分组走内，第二dns设置谷歌8.8.8.8以及8.8.4.4分组走外，关闭DHCP/DNS里面的dns缓存，只设置smartdns缓存5000条，
开启adguard home（0.179.40版本）或者不开。均会产生网络卡顿的问题

jjm2473, 2023-11-09:
你确定不是ipv6导致的，有没有关掉ipv6测试。还有为什么要使用smartdns，passwall自己不会分流dns吗？

veen1992, 2023-11-09:
> 你确定不是ipv6导致的，有没有关掉ipv6测试。还有为什么要使用smartdns，passwall自己不会分流dns吗？

绝对不是IPV6，因为上上下下全都关闭掉了IPV6，而且SMART和ADG里面都勾选了不解析IPV6

jjm2473, 2023-11-09:
那就先别用这么多插件，只开passwall试试。如果要去广告，用adblock或者adbyby之类基于dns拦截的，真不知道smartdns和adguard有什么用。

jjm2473, 2023-11-09:
如果出现卡顿，就看看内存和CPU占用情况。

jjm2473, 2023-11-09:
如果adbyby或adblock等基于dns的能满足去广告需求，就别用其他基于流量的，性能差，有些还要安装证书。油管还能加速？不都是跑代理的吗，pass对于油管的流量默认都是跑代理，那速度就取决于你的翻墙节点。难道使用smartdns以后，油管还能直连不成？

jjm2473, 2023-11-09:
出现卡顿的时候看看内存和CPU占用情况，可能就是某个插件死循环或者内存泄露了

veen1992, 2023-11-09:
> 出现卡顿的时候看看内存和CPU占用情况，可能就是某个插件死循环或者内存泄露了

有可能，后面观察一下

veen1992, 2023-11-09:
> 如果adbyby或adblock等基于dns的能满足去广告需求，就别用其他基于流量的，性能差，有些还要安装证书。油管还能加速？不都是跑代理的吗，pass对于油管的流量默认都是跑代理，那速度就取决于你的翻墙节点。难道使用smartdns以后，油管还能直连不成？

smartdns有分流作用，可能走了正确的DNS地址会让油管轻松找到核心服务器，加载速度和响应速度也会在smart和adg的加持下更快，体验挺好的，所以一直在用，就是最近确实可能冲突了，也更新过smartdns的版本，也更新过Adg的版本，
1.pass指向单独用smrtdns的话DNS不会泄露，但会出现卡顿。另外smartdns配合adg就会出现泄露，
2.除非adg设置“作为dnsmasq上游服务器”也不会泄露，但也会出现adg直接卡死不过滤不刷新数据的情况，然后整体网络就会卡。
只有设置adg为“53端口替换dmsmasq”就不会卡，但会泄露DNS。

jjm2473, 2023-11-09:
不用那么在意dns泄漏，pass默认不也是先过黑名单和GFW列表吗，列表里的域名查询肯定不会发给运营商的DNS服务器，剩下的能有多少域名需要经过DNS才能确认是否代理的。

veen1992, 2023-11-10:
> 不用那么在意dns泄漏，pass默认不也是先过黑名单和GFW列表吗，列表里的域名查询肯定不会发给运营商的DNS服务器，剩下的能有多少域名需要经过DNS才能确认是否代理的。

我终于解决了，坑居然在chrome浏览器，更新smartdns需要先卸载zh_cn的界面包，然后清理浏览器缓存新的界面才出来，全新的smartdns是完全不同的布局，调好就不再出现泄露和卡顿了

jjm2473, 2023-11-11:
靠，smartdns这么垃圾的吗，连界面缓存都没处理？

closed 2024-02-02 by jjm2473