istoreos/istoreos #1098: 无法连接VPN该如何解决

Richard1817, 2023-11-05:
## 2. 详细叙述

### (1) 具体问题

A：我的windows电脑使用这个路由器，无法使用PPTP，连接公司的VPN

### (2) 路由器型号和固件版本

A：软路由 N100
iStoreOS 22.03.5 2023102711

### (3) 详细日志和/或截图

A：是不是需要打开路由器的PPTP服务，请问如何打开

jjm2473, 2023-11-09:
打开防火墙设置，在区域lan的右边点“编辑”，在“连接跟踪设置”的“连接跟踪助手”里面选择PPTP看有没有用：
[image]

jjm2473, 2023-11-10:
测试了没？

Richard1817, 2023-11-10:
已经不用这个了 刷了op 正常了

Amoystyle, 2023-11-21:
> 打开防火墙设置，在区域lan的右边点“编辑”，在“连接跟踪设置”的“连接跟踪助手”里面选择PPTP看有没有用： [image]

我使用你说的这个确实能成功连上VPN

jjm2473, 2023-11-24:
新固件会默认打开端口转发助手，但只对新刷机有效，老用户还是按这个方式解决

closed 2023-11-24 by jjm2473