istoreos/istoreos #1102: [BUG]通过首页下载服务快速配置Aria2失败

HessianZ, 2023-11-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
通过【下载服务】的快速配置按钮对Aria2进行配置，提示“路径错误，必须选择硬盘的路径或者/root的路径”。
即使配置时提供的路径都是其他硬盘上的也不行。
但是同样的设置在高级配置中自行配置后启用Aria2后使用正常。

### (1) 具体问题

A：无法快速配置Aria2。

### (2) 路由器型号和固件版本

A：幻网N6000-NAS + iStoreOS 22.03.5 2023102711

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-11-09:
把分区挂载到/mnt/data试试

jjm2473, 2023-11-09:
测试了，就是因为没把硬盘挂载到/mnt里面。为了避免硬盘没挂载的情况下把根目录写满，还是挂载到/mnt里面吧

closed 2023-11-09 by jjm2473

HessianZ, 2023-11-09:
> 测试了，就是因为没把硬盘挂载到/mnt里面。为了避免硬盘没挂载的情况下把根目录写满，还是挂载到/mnt里面吧

好的，感谢回复。

这块也许可以改进一下提示内容。另外这种限制如果能做成全局开关就好了，对于像我这种盲目自信的玩家可能希望有一点更大的自由度。

jjm2473, 2023-11-09:
这自由度没啥用啊，挂载到/mnt和根目录有什么区别？如果哪天你的硬盘坏了，结果aria2还是把文件下载到/data，会发生什么？根目录可能被写满，导致更多的问题，例如系统配置文件损坏。但是/mnt不一样，mnt本身是个tmpfs，最多占用内存4MB空间，如果/mnt/data没挂载，就算写到/mnt/data也最多用掉4MB内存，不会导致系统的故障。

HessianZ, 2023-11-10:
> 这自由度没啥用啊，挂载到/mnt和根目录有什么区别？如果哪天你的硬盘坏了，结果aria2还是把文件下载到/data，会发生什么？根目录可能被写满，导致更多的问题，例如系统配置文件损坏。但是/mnt不一样，mnt本身是个tmpfs，最多占用内存4MB空间，如果/mnt/data没挂载，就算写到/mnt/data也最多用掉4MB内存，不会导致系统的故障。

好吧，有道理，我可能还要适应一下。主要是硬盘都挂到/mnt下不太符合直觉，而且硬盘坏这种场景还是极少发生，同时我也可以把/data做成tmpfs。

jjm2473, 2023-11-10:
@HessianZ 不一定硬盘坏，可能文件系统损坏，也可能挂载的速度太慢（例如有些大容量的机械硬盘初始化时间比较长），挂载前文件就已经被aria2打开了。尤其是aria2的配置文件路径和下载路径不在同一个分区的情况下，配置还在，但是下载目录没挂载，aria2还能读取到之前的下载任务。