istoreos/istoreos #1103: 旁路由模式下  每20分钟就会大量请求一个 地址 导致 主路由 cpu100%

876529270, 2023-11-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：旁路由模式下  每20分钟就会大量请求一个 地址 导致 主路由 cpu100% ，不知道是什么应用请求的

### (2) 路由器型号和固件版本

A：
[image]

### (3) 详细日志和/或截图

A：
[image]

876529270, 2023-11-08:
今天 请求这个ip
[image]

876529270, 2023-11-08:
今天请求这个 地址
[image]

876529270, 2023-11-08:
[image]

876529270, 2023-11-09:
有点奇怪  啥也没动 今天 又正常了 昨晚11点之后 到现在 都是 正常的

jjm2473, 2023-11-09:
https://www.ip2location.com/demo/172.247.174.21
https://www.ip2location.com/demo/23.225.223.234

ISP | CloudRadium L.L.C
Domain | cloudradium.com

不知道啥服务

ojj90, 2023-11-09:
同样的情况啊，装了好几个istoreos，就只有一个出现这种情况，有人知道吗

jjm2473, 2023-11-09:
你们最好分享下自己安装了哪些服务，例如哪些docker容器。直接截图docker容器页面和后台页面服务菜单。还有，是不是翻墙了，用的是不是同一家VPS或者机场

ojj90, 2023-11-09:
我这边原理上是有网心云和迅雷的商店装的插件，也是刚配置没几天的系统，后来出现问题也以为是他们的问题，就把全部docker关掉，但是问题依旧。

876529270, 2023-11-09:
> 你们最好分享下自己安装了哪些服务，例如哪些docker容器。直接截图docker容器页面和后台页面服务菜单。还有，是不是翻墙了，用的是不是同一家VPS或者机场

确实用docker装过v2Ray，但是后来把容器删掉了，这个也影响吗。
docker命令是
docker run -d \
  --restart=always \
  --privileged \
  --network=host \
  --name v2raya \
  -e V2RAYA_ADDRESS=0.0.0.0:2017 \
  -v /lib/modules:/lib/modules:ro \
  -v /etc/resolv.conf:/etc/resolv.conf \
  -v /etc/v2raya:/etc/v2raya \
  mzz2017/v2raya

876529270, 2023-11-09:
> 你们最好分享下自己安装了哪些服务，例如哪些docker容器。直接截图docker容器页面和后台页面服务菜单。还有，是不是翻墙了，用的是不是同一家VPS或者机场

我刚看了一下 /etc/resolv.conf 这个文件  旁路由模式下  这个文件正常吗
[image]

jjm2473, 2023-11-09:
> 我刚看了一下 /etc/resolv.conf 这个文件 旁路由模式下 这个文件正常吗

@876529270 正常，默认都是dnsmasq提供dns服务的，就算是旁路由模式，那也是dnsmasq代理本机dns请求

jjm2473, 2023-11-09:
> 确实用docker装过v2Ray，但是后来把容器删掉了，这个也影响吗。

不影响，除非里面有病毒。 不过，为什么会有“/lib/modules”？openwrt的内核模块都是加载了的，根本不需要让其他应用加载。

jjm2473, 2023-11-09:
你们执行下 `ps w` 看看有没有奇怪的进程

jjm2473, 2023-11-09:
还有，你们有没有改过istoreos的默认密码，没改密码可能被黑客攻击

jjm2473, 2023-11-09:
你们的机器是不是都是x86的？大部分病毒都是只有x86版本的

jjm2473, 2023-11-09:
还有，执行`cat /etc/crontabs/root` 看看结果

876529270, 2023-11-09:
> cat /etc/crontabs/root

[image]

876529270, 2023-11-09:
> 你们执行下 `ps w` 看看有没有奇怪的进程

root@iStoreOS:~# ps w
  PID USER       VSZ STAT COMMAND
    1 root      1388 S    /sbin/procd
    2 root         0 SW   [kthreadd]
    3 root         0 IW<  [rcu_gp]
    4 root         0 IW<  [rcu_par_gp]
    6 root         0 IW<  [kworker/0:0H-kb]
    8 root         0 IW<  [mm_percpu_wq]
    9 root         0 SW   [rcu_tasks_trace]
   10 root         0 SW   [ksoftirqd/0]
   11 root         0 IW   [rcu_sched]
   12 root         0 SW   [migration/0]
   13 root         0 SW   [cpuhp/0]
   14 root         0 SW   [cpuhp/1]
   15 root         0 SW   [migration/1]
   16 root         0 SW   [ksoftirqd/1]
   18 root         0 IW<  [kworker/1:0H-ev]
   19 root         0 SW   [cpuhp/2]
   20 root         0 SW   [migration/2]
   21 root         0 SW   [ksoftirqd/2]
   23 root         0 IW<  [kworker/2:0H-kb]
   24 root         0 SW   [cpuhp/3]
   25 root         0 SW   [migration/3]
   26 root         0 SW   [ksoftirqd/3]
   28 root         0 IW<  [kworker/3:0H-ev]
   29 root         0 IW<  [netns]
  337 root         0 IW   [kworker/1:1]
  367 root         0 SW   [oom_reaper]
  368 root         0 IW<  [writeback]
  370 root         0 SW   [kcompactd0]
  375 root         0 IW<  [pencrypt_serial]
  377 root         0 IW<  [pdecrypt_serial]
  379 root         0 IW<  [cryptd]
  385 root         0 IW<  [kintegrityd]
  387 root         0 IW<  [kblockd]
  388 root         0 IW<  [blkcg_punt_bio]
  530 root         0 IW   [kworker/2:1-eve]
  765 root         0 IW<  [ata_sff]
  794 root         0 SW   [watchdogd]
  898 root         0 IW<  [kworker/3:1H-kb]
  931 root         0 SW   [kswapd0]
  937 root         0 IW<  [kthrotld]
  991 root         0 IW<  [acpi_thermal_pm]
 1080 root         0 IW<  [iscsi_eh]
 1086 root         0 IW<  [iscsi_conn_clea]
 1097 root         0 IW<  [nvme-wq]
 1104 root         0 IW<  [nvme-reset-wq]
 1108 root         0 IW<  [nvme-delete-wq]
 1119 root         0 SW   [scsi_eh_0]
 1120 root         0 IW<  [scsi_tmf_0]
 1124 root         0 SW   [scsi_eh_1]
 1125 root         0 IW<  [scsi_tmf_1]
 1159 root         0 IW<  [uas]
 1163 root      4364 S    /usr/sbin/ddnstod -u 2f189dd8-dfb5-417f-ab7e-08ca0a0afce5 -x 0 -F
 1208 root         0 IW<  [ipv6_addrconf]
 1209 root         0 IW<  [kworker/1:1H-kb]
 1255 root         0 IW<  [kworker/2:1H-kb]
 1258 root         0 SW   [scsi_eh_2]
 1259 root         0 IW<  [scsi_tmf_2]
 1282 root         0 IW<  [tpm_dev_wq]
 1371 root         0 SW   [jfsIO]
 1372 root         0 SW   [jfsCommit]
 1373 root         0 SW   [jfsCommit]
 1374 root         0 SW   [jfsCommit]
 1375 root         0 SW   [jfsCommit]
 1376 root         0 SW   [jfsSync]
 1379 root         0 IW<  [xfsalloc]
 1380 root         0 IW<  [xfs_mru_cache]
[... 122 lines cut ...]
13802 root       832 S    odhcp6c -s /lib/netifd/dhcpv6.script -P0 -t120 br-lan
15355 root      2956 S    bash /usr/local/bin/task shufflewzc_faker2_main/jd_dongDongFarm_task.js
15911 root      317m S    node jd_dongDongFarm_task.js
17677 root         0 IW   [kworker/3:0-eve]
17698 root      703m S    /usr/bin/containerd-shim-runc-v2 -namespace moby -id 0c531b7cc9846c0c36d30dcc89b926058ef0e945cf30bf4303d1
17719 root       204 S    /package/admin/s6/command/s6-svscan -d4 -- /run/service
17740 root       208 S    s6-supervise s6-linux-init-shutdownd
17742 root       196 S    /package/admin/s6-linux-init/command/s6-linux-init-shutdownd -c /run/s6/basedir -g 3000 -C -B
17779 root       208 S    s6-supervise svc-qbittorrent
17780 root       208 S    s6-supervise s6rc-oneshot-runner
17781 root       208 S    s6-supervise s6rc-fdholder
17789 root       184 S    /package/admin/s6/command/s6-ipcserverd -1 -- /package/admin/s6/command/s6-ipcserver-access -v0 -E -l0 -i
17895 911       297g S    /usr/bin/qbittorrent-nox --webui-port=8080
17980 root      2956 S    bash /usr/local/bin/task shufflewzc_faker2_main/jd_fruit_task.js
18188 root      322m S    node jd_fruit_task.js
19957 root       788 S    /usr/bin/wsdd2 -i br-lan -N iStoreOS -G WORKGROUP -b vendor:default-string-default-string,model:Default_s
21050 root         0 IW   [kworker/0:0-eve]
21413 root       748 S    /sbin/askfirst /usr/libexec/login.sh
24477 root         0 IW   [kworker/1:0-eve]
26198 root      2956 S    bash /usr/local/bin/task shufflewzc_faker2_main/jd_fruit_task.js
27294 root      324m S    node jd_fruit_task.js
27372 root      2596 S    bash
29183 root         0 IW   [kworker/3:2]
29212 root         0 IW   [kworker/0:1]
30198 root         0 IW   [kworker/u8:2-ev]

jjm2473, 2023-11-09:
@876529270 没看到什么可疑的进程，但是青龙和京东还是关掉看看。执行 `ls -l /root/.ssh/` 看看有没有文件，还有改掉默认密码。

eddyclay, 2023-11-09:
我都折腾一小时了 也没找到什么进程 发送的流量

eddyclay, 2023-11-09:
我这个是主路由 一会就卡主上不来

eddyclay, 2023-11-09:
[image]

eddyclay, 2023-11-09:
[image]

eddyclay, 2023-11-09:
[https://eddyclay-hk.oss-cn-hongkong.aliyuncs.com/ufo/%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE%202023-11-09%20202127.png](url)

eddyclay, 2023-11-09:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

eddyclay, 2023-11-09:
[image]

eddyclay, 2023-11-09:
查半天 也只知道一个叫做 htop 的进程 不断用udp 发送大量数据 我卡的上不来了

876529270, 2023-11-09:
> @876529270 没看到什么可疑的进程，但是青龙和京东还是关掉看看。执行 `ls -l /root/.ssh/` 看看有没有文件，还有改掉默认密码。

改密码 docker都关了 也没用  这个目录下面没有文件

jjm2473, 2023-11-09:
@eddyclay htop是用来查看进程的，你可以执行htop看看哪个进程占用的cpu高

876529270, 2023-11-10:
顶不住了，重装系统了

jjm2473, 2023-11-11:
看看进程啊，万一发现了个病毒或者漏洞

876529270, 2023-11-13:
@jjm2473 怎么办呢，重装之后还是有问题，docker 装了qbittorrent、jellyfin、nastools ，我把软件配置备份一下，发你  看看  能不能复现问题

876529270, 2023-11-13:
@jjm2473 [backup-iStoreOS-2023-11-13.tar.gz](https://github.com/istoreos/istoreos/files/13331678/backup-iStoreOS-2023-11-13.tar.gz)

eddyclay, 2023-11-13:
我原来以为是某个软件或者passw出问题 重新装了 什么软件都没开 也这样 直接启用原来的固件了

jjm2473, 2023-11-13:
@876529270 直接用top命令看进程的cpu使用率

jjm2473, 2023-11-13:
@eddyclay 你在用finalshell连接之前会这样吗

jjm2473, 2023-11-13:
@876529270 从你提供的文件来看，应该是中毒了，比较明显的不明文件是 `/etc/profile.d/bash.cfg` ，是个可执行文件，而且足足有5MB，文件hash值 `c14b0aa9f855639d4bebe2e1f501d63fcec39cd9b46b48944866a277eb0dcdd3`， 查看 [VirusTotal 的报告](https://www.virustotal.com/gui/file/c14b0aa9f855639d4bebe2e1f501d63fcec39cd9b46b48944866a277eb0dcdd3/detection) 。

正常情况下`/etc/profile.d`里面应该只有几个脚本。

jjm2473, 2023-11-13:
在后台的状态-系统进程里面，看看有没有 bash.cfg 这个进程

876529270, 2023-11-13:
@jjm2473 /etc/profile.d/bash.cfg 这个文件 是刷过系统就存在了  
刷的这个固件 [istoreos-22.03.5-2023102711-x86-64-squashfs-combined-efi.img.gz](https://fw0.koolcenter.com/iStoreOS/x86_64_efi/istoreos-22.03.5-2023102711-x86-64-squashfs-combined-efi.img.gz)

jjm2473, 2023-11-13:
@876529270 你确定不是因为你的网络里面还有其他病毒，所以不断被感染？

jjm2473, 2023-11-13:
@876529270 我教你怎么验证。首先把软路由断网，然后只用电脑连接软路由，进后台不保留配置刷机，刷完以后马上改密码（尽量复杂点），然后再联网。除非你的电脑也中毒了，不然这样肯定就不会再被感染。

closed 2024-02-02 by jjm2473