istoreos/istoreos #1105: 日志一直提示错误

cc12593014, 2023-11-08:
Wed Nov  8 19:15:49 2023 kern.err kernel: [102347.560255] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:15:50 2023 kern.err kernel: [102347.600252] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:15:50 2023 kern.err kernel: [102347.640251] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:15:50 2023 kern.err kernel: [102347.680247] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:15:50 2023 kern.err kernel: [102347.720247] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:15:50 2023 kern.err kernel: [102347.720250] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)
Wed Nov  8 19:16:21 2023 kern.err kernel: [102378.919672] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:21 2023 kern.err kernel: [102378.959665] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:21 2023 kern.err kernel: [102378.989663] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:21 2023 kern.err kernel: [102379.019663] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:21 2023 kern.err kernel: [102379.059662] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:21 2023 kern.err kernel: [102379.059665] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)
Wed Nov  8 19:16:51 2023 kern.err kernel: [102409.009110] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:51 2023 kern.err kernel: [102409.049106] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:51 2023 kern.err kernel: [102409.079103] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:51 2023 kern.err kernel: [102409.119120] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:51 2023 kern.err kernel: [102409.159101] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:16:51 2023 kern.err kernel: [102409.159104] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)
Wed Nov  8 19:17:22 2023 kern.err kernel: [102439.728535] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:22 2023 kern.err kernel: [102439.758531] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:22 2023 kern.err kernel: [102439.798528] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:22 2023 kern.err kernel: [102439.838528] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:22 2023 kern.err kernel: [102439.878527] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:22 2023 kern.err kernel: [102439.878530] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)
Wed Nov  8 19:17:53 2023 kern.err kernel: [102471.087952] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:53 2023 kern.err kernel: [102471.127950] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:53 2023 kern.err kernel: [102471.157946] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:53 2023 kern.err kernel: [102471.187945] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:53 2023 kern.err kernel: [102471.217945] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:17:53 2023 kern.err kernel: [102471.217949] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)
Wed Nov  8 19:18:24 2023 kern.err kernel: [102502.437365] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:18:24 2023 kern.err kernel: [102502.477361] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:18:24 2023 kern.err kernel: [102502.517360] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:18:24 2023 kern.err kernel: [102502.547360] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:18:24 2023 kern.err kernel: [102502.587357] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
Wed Nov  8 19:18:24 2023 kern.err kernel: [102502.587360] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)
日志一直提示这个错误

jjm2473, 2023-11-09:
不是让你提交issue的时候提供固件版本和机型信息吗？

closed 2023-11-09 by jjm2473

cc12593014, 2023-11-20:
固件 iStoreOS 22.03.5 2023102711 / LuCI openwrt-22.03 branch git-23.292.78392-9f66674
机型CWWK CW-ADLN-6L  
CPU Intel(R) N100

jjm2473, 2023-11-20:
你这个机器有TYPE-C接口吗，看起来日志都是跟它有关。接上hdmi看看还有没有错误。如果这个错误不影响使用就先别管了，如果会影响使用，可以降级固件到istoreos-22.03.5-2023092209，关掉i915-oot驱动。

cc12593014, 2023-11-22:
插入hdmi显示器,日志就不报了.

jjm2473, 2023-11-24:
OK，感谢反馈。