istoreos/istoreos #1106: R4S刷最新几个版本iStoreOS无法启动，sys灯不亮

DeadpoolRain, 2023-11-11:
## 2. 详细叙述

### (1) 具体问题

A：R4S刷最新的几个版本iStoreOS无法启动，系统sys灯不亮，刷其他固件一切正常，换新卡重刷之后仍然如此

### (2) 路由器型号和固件版本

A：NanoPi R4S 固件版本istoreos-22.03.5-2023102711-r4s-squashfs.img.gz

### (3) 详细日志和/或截图

A：

DeadpoolRain, 2023-11-11:
补充一个：刷istore-21.02.3-2023060914版本的正常，istore-22.xxx的都无法启动

jjm2473, 2023-11-13:
暂时没有机器调试

DeadpoolRain, 2023-11-13:
已经解决了，可以先从老版本刷进去，然后再升级到最新版。直接把最新版固件烧进TF卡无法进系统，这个很奇怪。

jjm2473, 2023-11-13:
那就是uboot的问题，从老版本升级的时候不会升级uboot

Roxy09099, 2024-01-08:
> It has been solved. You can flash it from the old version first, and then upgrade to the latest version. It is very strange that I cannot enter the system if I directly burn the latest version of the firmware into the TF card.

How do you upgrade version 21 to 22? upload files via luci? or OTA?

DeadpoolRain, 2024-01-09:
> > It has been solved. You can flash it from the old version first, and then upgrade to the latest version. It is very strange that I cannot enter the system if I directly burn the latest version of the firmware into the TF card.
> 
> How do you upgrade version 21 to 22? upload files via luci? or OTA?

OTA

Roxy09099, 2024-01-09:
> OTA

Even though I have installed the curl package and connected to the internet, when I click on OTA update, I still get a curl error

jjm2473, 2024-01-10:
@Roxy09099 You should not install curl.

jjm2473, 2024-01-10:
@Roxy09099 Install 'systools' from iStore, then execute 'Reset rom pkgs', then wait for reboot.

closed 2024-02-02 by jjm2473

jjm2473, 2024-02-23:
根据[用户反馈](https://github.com/istoreos/istoreos/issues/1210#issuecomment-1952755038)，openwrt官方固件也有这个问题，那就让openwrt官方修复吧 ，istoreos的r4s固件目前使用的uboot来自openwrt官方