istoreos/istoreos #1110: tailscale 组网失败，提示tailscale和tailscaled的版本号不一致

xmasrain, 2023-11-14:
### (1) 具体问题

A：启动服务时提示：Warning: client version "1.32.3-1 (OpenWrt)" != tailscaled server version "1.52.1-t75d3c9385-g3e9627f3b"

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-11-21:
卸载重装试试

jjm2473, 2023-11-21:
istore不是有个tailscale的插件了吗，直接使用插件看看有没有问题

xmasrain, 2023-11-21:
就是用的插件，后来我按照这个文章《OpenWrt安装配置Tailscale C ZhiYunHaiBo 智云海博 (zyh8.com)<https://www.zyh8.com/archives/135/openwrt%E5%AE%89%E8%A3%85%E9%85%8D%E7%BD%AEtailscale/>》重装了，就好了

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

Jesn, 2024-01-26:
建议官网把Tailscale的版本升级下，Tailscale版本太低了，derp节点会经常报错

jjm2473, 2024-02-01:
@Jesn tailscale是openwrt仓库提供的，要升级只能反馈给openwrt

closed 2024-02-02 by jjm2473