istoreos/istoreos #1117: 1020 1027版本缺少RTL8211F驱动

hx801217, 2023-11-18:
荐片盒子 RK3566，替换T68M等RK3588机型DTB能稳定运行到0928版本，但升级到1020 1027则找不到网卡。

正确的信息：
root@iStoreOS:~# ethtool -i eth0
driver: st_gmac （看不出什么mod驱动网卡）
version: Jan_2016
firmware-version: 
expansion-rom-version: 
bus-info: 
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no

[   15.111282] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-1:00] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   15.124540] dwmac4: Master AXI performs any burst length
[   15.124591] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   15.124673] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   15.124695] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[   17.177378] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

错误的信息：
[   15.205111] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-1:00] driver [Generic PHY] (irq=POLL)
[   16.215185] rk_gmac-dwmac fe010000.ethernet: Failed to reset the dma
[   16.215236] rk_gmac-dwmac fe010000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   16.215247] rk_gmac-dwmac fe010000.ethernet eth0: stmmac_open: Hw setup failed

能否发一个RTL8211F的驱动，我自己装试试

jjm2473, 2023-11-21:
需要在dtb里面写好phy的id，具体参考其他使用rtl8211f的机型的dtb

closed 2023-11-21 by jjm2473

jjm2473, 2023-11-21:
RTL8211F驱动绝对是有的，其他使用这个PHY的机器都能正常驱动千兆网口

hx801217, 2023-11-21:
> RTL8211F驱动绝对是有的，其他使用这个PHY的机器都能正常驱动千兆网口

之前0928是好的，升级1020就不行了，应该是源码变更的缘故吧。原DTB奉上，请大佬帮看看，俺们实在水平有限。
[JP tvbox.zip](https://github.com/istoreos/istoreos/files/13422839/JP.tvbox.zip)

hx801217, 2023-11-21:
获取PHY的脚本是不是变了

jjm2473, 2023-11-21:
你有重新刷机吗，要升级uboot

hx801217, 2023-11-21:
有的，您是指1020后，所有35xx都升级了UBOOT吗

hx801217, 2023-11-21:
> 你有重新刷机吗，要升级uboot

这机型只有EMMC没有SD卡，所有不存混用问题，我刚才也研究大佬置顶的混用说明，UBOOT我也没改，还是官方的，0928版本是运行非常正常，所以想请问是不是1020以后对初始化 GMAC istoreos这边改变了以前的策略，或者基于ADC判断的策略有所改变

jjm2473, 2023-11-24:
10月8日为了支持rk3528，更新了rk的内核 commitid: f6e69e21e6 ，可能是因为这个。uio-gmac肯定是没用上的，然后ADC只是用来判断机型，你的板子应该也这样，只要启动以后显示的机型正确就跟ADC无关了。

jjm2473, 2023-11-27:
> 怪不到呢，可能识别不到3568，当成3528了，网卡不在一个GPIO上，识别不到网卡。 fe010000本来应该是3566的网卡位置

你想多了，rk3566的dts配置跟rk3568一模一样，只是删减了部分io，rk3568都没问题。

NJYZ-96, 2023-12-14:
> > 10月8日为了支持rk3528，更新了rk的内核 commitid: [f6e69e2](https://github.com/istoreos/istoreos/commit/f6e69e21e6bc2001f16910a8ed0a135856b42741) ，可能是因为这个。uio-gmac肯定是没用上的，然后ADC只是用来判断机型，你的板子应该也这样，只要启动以后显示的机型正确就跟ADC无关了。
> 
> 怪不到呢，可能识别不到3568，当成3528了，网卡不在一个GPIO上，识别不到网卡。 fe010000本来应该是3566的网卡位置

请问关于荐片盒子 RK3566；有后续更新的固件吗？目前我找到最新的也是istoreos-22.03.5-2023092815-jptvbox-squashfs