istoreos/istoreos #1121: 可以添加 ARM GPU drivers吗

leux0, 2023-11-21:
可不可以为RK3588系列添加GPU支持，想docker android调用GPU来挂游戏，但缺少GPU支持。
希望可以添加drm-panfrost之类的支持，感谢感谢

https://github.com/coolsnowwolf/lede/commit/c8e9d227fd2707ba8df28c1ee26e4baa41e301a1

ks-bao, 2023-11-21:
我记得 RK3588 我们是支持 Jellyfin 硬解的啊。这个是 GPU 还是其他的？

leux0, 2023-11-21:
感谢，能驱动 /dev/mali0 这个吗，docker android会用到，谢谢

jjm2473, 2023-11-21:
你怎么跑docker android的？waydroid？anbox？

leux0, 2023-11-21:
下面这几个docker都可以调用/dev/mali0
registry.cn-hangzhou.aliyuncs.com/whsyf/dobox:rk3588-202303017

chisbread/rk3588-gaming:redroid-firefly

shangzebei/rk3588:latest

jjm2473, 2023-11-21:
哦，之前在firefly看过这个。不过arm的gpu驱动都是各家厂商自己提供的，linux内核有自带的驱动吗？

leux0, 2023-11-23:
谢谢回复，这个rk3588的内核可以驱动mali0：https://github.com/unifreq/linux-5.10.y-rk35xx
闭源firmware部分[mali_csffw.bin](https://github.com/unifreq/openwrt_packit/blob/master/files/firmware/mali_csffw.bin)

leux0, 2023-11-24:
我使用自带的内核编译CONFIG_DRM_PANFROST时在编译IOMMU_IO_PGTABLE处报错，而PANFROST依赖CONFIG_IOMMU_IO_PGTABLE  
[image]

jjm2473, 2023-11-28:
CONFIG_DRM_PANFROST 不是drm显示相关的吗，应该跟3D加速没关系吧。根据rk3568和rk3588的dts，3D加速的驱动应该是 bifrost 或 mali400  https://github.com/armbian/linux-rockchip/tree/rk-5.10-rkr6/drivers/gpu/arm ，这些驱动在原版的 linux 里面是没有的

jjm2473, 2023-11-28:
OK，PANFROST 确实支持 arm,mali-bifrost https://github.com/torvalds/linux/blob/df60cee26a2e3d937a319229e335cb3f9c1f16d2/drivers/gpu/drm/panfrost/panfrost_drv.c#L772

jjm2473, 2023-11-28:
OK，改一下clock和中断的名称就行了
```
&gpu {
	interrupt-names = "gpu", "mmu", "job";
	clock-names = "clk_mali", "bus";
};
```

leux0, 2023-11-28:
感谢感谢

jjm2473, 2023-11-29:
@leux0 有办法测试panfrost吗

leux0, 2023-11-29:
这有个ubuntu下的GPU配置说明：https://github.com/ChisBread/rk3588-gaming-step-by-step/blob/main/gpu-envs.md

下面这几个docker都宣称可以调用RK3588的GPU

docker run -itd --rm --privileged \
  --pull always \
  -v /mnt/nvme0n1p1/data:/data \
  -v /dev/net/tun:/dev/tun \
  -v /dev/mali0:/dev/mali0 \
  -p 5501:5555 \
  --name dobox1 \
  registry.cn-hangzhou.aliyuncs.com/whsyf/dobox:rk3588-202303017
----------------------------------------------------------

docker run -itd --rm --privileged \
    --pull always \
    -v /data:/data \
    -p 5555:5555 \
    chisbread/rk3588-gaming:redroid-firefly \
    androidboot.redroid_height=1600 \
    androidboot.redroid_width=2560 \
    androidboot.redroid_dpi=480 \
androidboot.redroid_gpu_mode=host \
--mount=type=bind,source=/dev/mali0,destination=/dev/mali0
--------------------------------------------

docker pull --platform=arm64 shangzebei/rk3588:latest

docker run -d --privileged --name test \
    -p 5555:5555 \
    -v /dev/mali0:/dev/mali0 \
    shangzebei/rk3588 androidboot.redroid_gpu_mode=mali

jjm2473, 2023-11-29:
但是看起来这些都要/dev/mali0。
我今天还在rk3588上测试，发现panfrost不能驱动gpu，昨天rk3568倒是可以

leux0, 2023-11-29:
touch /home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176/.configured
rm -f /home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176/vmlinux /home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176/System.map
make -C /home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176 KCFLAGS="-fmacro-prefix-map=/home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/leux/istoreos/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Nov 28 12:14:03 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/leux/istoreos/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/leux/istoreos/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.176 CC="ccache_cc" Image dtbs modules
make[5]: Entering directory '/home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/iommu/iommu.o
drivers/iommu/iommu.c: In function '__iommu_map_pages':
drivers/iommu/iommu.c:2415:16: error: 'const struct iommu_ops' has no member named 'map_pages'
 2415 |         if (ops->map_pages) {
      |                ^~
drivers/iommu/iommu.c:2416:26: error: 'const struct iommu_ops' has no member named 'map_pages'
 2416 |                 ret = ops->map_pages(domain, iova, paddr, pgsize, count, prot,
      |                          ^~
In file included from ./include/linux/dev_printk.h:14,
                 from ./include/linux/device.h:15,
                 from drivers/iommu/iommu.c:9:
drivers/iommu/iommu.c: In function '__iommu_map':
drivers/iommu/iommu.c:2436:39: error: 'const struct iommu_ops' has no member named 'map_pages'
 2436 |         if (unlikely(!(ops->map || ops->map_pages) ||
      |                                       ^~
./include/linux/compiler.h:78:45: note: in definition of macro 'unlikely'
   78 | # define unlikely(x)    __builtin_expect(!!(x), 0)
      |                                             ^
drivers/iommu/iommu.c: In function '__iommu_unmap_pages':
drivers/iommu/iommu.c:2522:19: error: 'const struct iommu_ops' has no member named 'unmap_pages'
 2522 |         return ops->unmap_pages ?
      |                   ^~
drivers/iommu/iommu.c:2523:19: error: 'const struct iommu_ops' has no member named 'unmap_pages'
 2523 |                ops->unmap_pages(domain, iova, pgsize, count, iotlb_gather) :
      |                   ^~
In file included from ./include/linux/dev_printk.h:14,
                 from ./include/linux/device.h:15,
                 from drivers/iommu/iommu.c:9:
drivers/iommu/iommu.c: In function '__iommu_unmap':
drivers/iommu/iommu.c:2536:41: error: 'const struct iommu_ops' has no member named 'unmap_pages'
 2536 |         if (unlikely(!(ops->unmap || ops->unmap_pages) ||
      |                                         ^~
./include/linux/compiler.h:78:45: note: in definition of macro 'unlikely'
   78 | # define unlikely(x)    __builtin_expect(!!(x), 0)
      |                                             ^
drivers/iommu/iommu.c: In function '__iommu_unmap_pages':
drivers/iommu/iommu.c:2525:1: error: control reaches end of non-void function [-Werror=return-type]
 2525 | }
      | ^
cc1: some warnings being treated as errors
make[7]: *** [scripts/Makefile.build:286: drivers/iommu/iommu.o] Error 1
make[6]: *** [scripts/Makefile.build:503: drivers/iommu] Error 2
make[5]: *** [Makefile:1827: drivers] Error 2
make[5]: Leaving directory '/home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176'
make[4]: *** [Makefile:23: /home/leux/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176/.modules] Error 2
make[4]: Leaving directory '/home/leux/istoreos/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/leux/istoreos/target/linux'
time: target/linux/compile#2.18#10.51#21.60
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/leux/istoreos'
make[1]: *** [target/Makefile:23: /home/leux/istoreos/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/leux/istoreos'
make: *** [/home/leux/istoreos/include/toplevel.mk:234: world] Error 2
leux@B650:~/istoreos$

jjm2473, 2023-11-29:
@leux0 iommu.c应该要回滚到最初版本的，忘了删掉patch。

执行
```
git reset --hard HEAD^
git pull
```
拉取代码试试

jjm2473, 2023-11-29:
你应该是选了编译所有内核模块吧，但是板子又没依赖 kmod-backlight

leux0, 2023-11-29:
谢谢，我选择了 > Kernel modules > Video Support backlight-pwm > kmod-backlight-pwm 这个后通过了

leux0, 2023-11-29:
没有选择所有内核模块，仅添加了个  kmod-drm-panfrost 后出现的

[image]

[image]

jjm2473, 2023-11-29:
3588目前我这里还是有错误（在打了些补丁之后）
```
[    8.939702] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply from device tree
[    8.939728] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply property in node /power-management@fd8d8000/power-controller failed
[    8.940903] panfrost fb000000.gpu: clock rate = 200000000
[    8.940940] panfrost fb000000.gpu: bus_clock rate = 198000000
[    8.941027] panfrost fb000000.gpu: Looking up mali-supply from device tree
[    8.941229] panfrost fb000000.gpu: Looking up mem-supply from device tree
[    8.952270] panfrost fb000000.gpu: gpu soft reset timed out
[    8.952952] panfrost fb000000.gpu: Fatal error during GPU init
[    8.953068] panfrost: probe of fb000000.gpu failed with error -110
```
估计是panfrost驱动少用了一些clk，或者gpu-supply这个错误。

leux0, 2023-11-29:
查看了一堆资料，[rk3588的BSP内核可以驱动mali0](https://github.com/unifreq/linux-5.10.y-rk35xx)，firmware部分[mali_csffw.bin](https://github.com/JeffyCN/mirrors/raw/fc8e00761e74ef3d559c0c0edf3700e20b47b578/firmware/g610/mali_csffw.bin)
而开源驱动panfrost  没有 /dev/mali0 

root@Ubuntu:~# dmesg | grep mali
[   10.403739] mali fb000000.gpu: Kernel DDK version g18p0-01eac0
[   10.404420] mali fb000000.gpu: bin=0
[   10.404593] mali fb000000.gpu: leakage=14
[   10.404617] debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
[   10.405949] mali fb000000.gpu: pvtm=852
[   10.406034] mali fb000000.gpu: pvtm-volt-sel=2
[   10.406926] mali fb000000.gpu: avs=0
[   10.407010] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 143; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.
[   10.407218] mali fb000000.gpu: r0p0 status 5 not found in HW issues table;
[   10.407224] mali fb000000.gpu: falling back to closest match: r0p0 status 0
[   10.407228] mali fb000000.gpu: Execution proceeding normally with fallback match
[   10.407233] mali fb000000.gpu: GPU identified as 0x7 arch 10.8.6 r0p0 status 0
[   10.407256] mali fb000000.gpu: No priority control manager is configured
[   10.407323] mali fb000000.gpu: No memory group manager is configured
[   10.407339] mali fb000000.gpu: Protected memory allocator not available
[   10.408419] mali fb000000.gpu: Capping CSF_FIRMWARE_TIMEOUT to CSF_FIRMWARE_PING_TIMEOUT
[   10.408638] mali fb000000.gpu: l=10000 h=85000 hyst=5000 l_limit=0 h_limit=800000000 h_table=0
[   10.409231] mali fb000000.gpu: Probed as mali0
[   65.336678] mali fb000000.gpu: Loading Mali firmware 0x1010000
[   65.339078] mali fb000000.gpu: Mali firmware git_sha: ee476db42870778306fa8d559a605a73f13e455c

jjm2473, 2023-11-29:
嗯，但是至少RK3568加载panfrost驱动没问题，且有/dev/dri/card*生成。

jjm2473, 2023-12-05:
@leux0 最新代码已经加上bifrost驱动了，测试rk3588有/dev/mali0，不需要修改dts

leux0, 2023-12-06:
感谢您的努力，已经成功在rk3588的iStore上运行支持调用GPU的Docker Android了

[image]

https://github.com/JeffyCN/mirrors/commits/libmali/firmware/g610
[/lib/firmware/mali_csffw.bin](https://github.com/JeffyCN/mirrors/raw/488f49467f5b4adb8ae944221698e9a4f9acb0ed/firmware/g610/mali_csffw.bin)

leux0, 2023-12-06:
https://github.com/JeffyCN/mirrors/commits/libmali/firmware/g610
[image]
[image]

leux0, 2023-12-08:
/lib/firmware/mali_csffw.bin 路径是宿主机的路径，docker镜像用的是Android的DDK宿主机不用管

这个fw文件如果跟驱动版本一致，好像会调用出问题

这是DDK版本一致时Docker调用的显示
[image]

这是DDK版本不一致时Docker调用时的显示
[image]

jjm2473, 2023-12-22:
RK35xx 系列已经全部支持 GPU 驱动

closed 2023-12-22 by jjm2473

Ybuzhenzhuo, 2024-07-04:
Excuse me, I would like to analyze the running status of OpenCl. I understand that streamlines have an opencl mode for analysis. But I cannot open opencl mode on RK3588. May I ask how everyone analyzes the OpenCl pattern, similar to the OpenCl pattern in Streamline.

a871516130, 2024-10-30:
我想在lede上使用，可以吗，现在提示kmod-rkgpu-bifrost找不到这个

jjm2473, 2024-11-01:
@a871516130 理论上没啥问题，反正驱动是从RK官方内核来的，移植过去就行。

a871516130, 2024-11-02:
移植过程中各种报错，小白搞了几天了都

a871516130, 2024-11-04:
只需要添加bifrost这个就可以了吗

jjm2473, 2024-11-07:
rk3588使用的是bifrost。不同的soc不一样，一般看soc的dts里面gpu兼容性字串，然后看看哪个驱动包含这个字串。

a871516130, 2024-11-27:
> 感谢您的努力，已经成功在rk3588的iStore上运行支持调用GPU的Docker Android了
> 
> ![25116B24AC2AEFA37AA600F5E160DF12](https://private-user-images.githubusercontent.com/19219159/288397875-5dd38ffa-adc3-44c5-b0ee-af5e78b270c8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI3MDAxNjksIm5iZiI6MTczMjY5OTg2OSwicGF0aCI6Ii8xOTIxOTE1OS8yODgzOTc4NzUtNWRkMzhmZmEtYWRjMy00NGM1LWIwZWUtYWY1ZTc4YjI3MGM4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDA5MzEwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4MDBmNDgwNjdjMjY3ZTk5NDFjMDQwN2EwMWI5ZWE3ZmYwMDMzNjcyM2I2MDliMTY4MGYwN2VjYWE1OTU4NzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Bz7UVTgt_hiaPmEVDScIJU7sAAAfqU2WU5hVemnQAXk)
> 
> ![D14B35980DB2BC5F852CA7F09B2B39FE](https://private-user-images.githubusercontent.com/19219159/288393960-b07d2569-96e5-4626-ac5b-722234810145.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI3MDAxNjksIm5iZiI6MTczMjY5OTg2OSwicGF0aCI6Ii8xOTIxOTE1OS8yODgzOTM5NjAtYjA3ZDI1NjktOTZlNS00NjI2LWFjNWItNzIyMjM0ODEwMTQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDA5MzEwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4YTBiZDJhODdjZjMzYjExMjQ5MTZjNTJkYzIwNjBlZDllMDE1MDhmNzI1MGQzMWRkNzlhNDhkOTU5ODFiY2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xYVHfft1nZiYFMxKyiDyOCennrarwe_vNrf1CfZWNoM)
> 
> [/lib/firmware/mali_csffw.bin](https://github.com/JeffyCN/mirrors/raw/488f49467f5b4adb8ae944221698e9a4f9acb0ed/firmware/g610/mali_csffw.bin)

老哥用的啥板子，我的运行之后adb连接不上

a871516130, 2024-12-13:
https://github.com/user-attachments/assets/d968c0b6-31e2-4826-9989-9f3bd1f46d12

老哥，我现在能运行起了来，但是界面循环闪退是啥情况哦

a871516130, 2024-12-16:
> /lib/firmware/mali_csffw.bin 路径是宿主机的路径，docker镜像用的是Android的DDK宿主机不用管
> 
> 这个fw文件如果跟驱动版本一致，好像会调用出问题
> 
> 这是DDK版本一致时Docker调用的显示 ![这是版本一致的显示](https://private-user-images.githubusercontent.com/19219159/288953958-36aaef28-5751-48c6-a9a9-1aaee9193951.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzMTU1NDUsIm5iZiI6MTczNDMxNTI0NSwicGF0aCI6Ii8xOTIxOTE1OS8yODg5NTM5NTgtMzZhYWVmMjgtNTc1MS00OGM2LWE5YTktMWFhZWU5MTkzOTUxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE2VDAyMTQwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThmYTUzMzc0MzIxMWIzNzYxYjkwNmJmMjZkYjQ0MjUyZmQ3M2VhODJhNDIwODM5NmYyMzcwNTI5ZTRiZmYyYjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.54ve8XX1-uz_Z305SQwqt6Gf3GMlzfd8jPP_ksWouTU)
> 
> 这是DDK版本不一致时Docker调用时的显示 ![版本不一致出错](https://private-user-images.githubusercontent.com/19219159/288954135-9408a0ad-8da3-49a3-95ca-d67253809a77.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzMTU1NDUsIm5iZiI6MTczNDMxNTI0NSwicGF0aCI6Ii8xOTIxOTE1OS8yODg5NTQxMzUtOTQwOGEwYWQtOGRhMy00OWEzLTk1Y2EtZDY3MjUzODA5YTc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE2VDAyMTQwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmYjZkMDNhMjViZDg2MTJkNmUxM2FhYzkwNjg3YWI4ZDllMGFjYjViNDYxNTkzODZmY2FmN2E3MWFmZWJiMWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9o3UMks9RBf-5guGc1hOmjEZoTBcenIawuFUWBWOXNc)

老哥，我现在到了这一步了，gpu报错的问题怎么解决啊