istoreos/istoreos #1123: 1020升级后，找不到网卡驱动

hx801217, 2023-11-21:
0928到1020，DTS源码更改了，经过比对，多了：
	uio@fe010000 {
		compatible = "rockchip,uio-gmac";
		reg = <0x00 0xfe010000 0x00 0x10000>;
		rockchip,ethernet = <0x82>;
		status = "disabled";
	};
导致认不出网卡PHY，应该怎么编译

jjm2473, 2023-11-24:
uio本来就不需要使用，我也不知道uio是干嘛的，之前的版本甚至没这个节点。如果你的固件是自己拼凑的，应该要看看uboot是不是是istoreos的。

jjm2473, 2023-11-24:
其他机型都没uio-gmac，你看其他人有问题吗？我自己也有rk3568机型。

closed 2023-11-24 by jjm2473

hx801217, 2023-11-24:
> uio本来就不需要使用，我也不知道uio是干嘛的，之前的版本甚至没这个节点。如果你的固件是自己拼凑的，应该要看看uboot是不是是istoreos的。

自己拉取了源码，编的也启动不起来。。。已经提交了commit，能帮改改不

hx801217, 2023-11-24:
> 其他机型都没uio-gmac，你看其他人有问题吗？我自己也有rk3568机型。

是3566。不过还是谢谢大大在pull requests里面的指点。

jjm2473, 2023-11-24:
OK，我有空找个3566试试，理论上3566跟3568差不多，只是少了些IO而已。

hx801217, 2023-11-24:
> OK，我有空找个3566试试，理论上3566跟3568差不多，只是少了些IO而已。

谢谢大神。

hx801217, 2023-11-24:
> OK，我有空找个3566试试，理论上3566跟3568差不多，只是少了些IO而已。

[image]
大致找到原因了， 网卡和MMC在10月8日以后，换了地址。所以编出来重复了

jjm2473, 2023-11-27:
肯定跟这个无关，alias只是一个提示用的

jjm2473, 2023-11-27:
[image-rk3566-jp-tvbox.dtb.zip](https://github.com/istoreos/istoreos/files/13469848/image-rk3566-jp-tvbox.dtb.zip)
@hx801217 你试试这个dtb行不行

hx801217, 2023-11-27:
> [image-rk3566-jp-tvbox.dtb.zip](https://github.com/istoreos/istoreos/files/13469848/image-rk3566-jp-tvbox.dtb.zip) @hx801217 你试试这个dtb行不行
谢谢大神，我来试试。另外，我在10月8日的更新发现，target/linux/rockchip/rk35xx/files/drivers/devfreq/rockchip_dmc.c有判断CPU的语句，是不是识别成3528，所以加载的是NULL.

jjm2473, 2023-11-27:
我今天找了个rk3566的板子，也就是ynnnas，测试了网络没任何问题。

刚刚发的dtb是把gmac1的clock改成output，根据之前部分rk3568机型的相关故障改的。

hx801217, 2023-11-27:
> 我今天找了个rk3566的板子，也就是ynnnas，测试了网络没任何问题。
> 
> 刚刚发的dtb是把gmac1的clock改成output，根据之前部分rk3568机型的相关故障改的。

还是不行啊。原机dtb上传如下
[jp-box.zip](https://github.com/istoreos/istoreos/files/13469940/jp-box.zip)

hx801217, 2023-11-27:
> 我今天找了个rk3566的板子，也就是ynnnas，测试了网络没任何问题。
> 
> 刚刚发的dtb是把gmac1的clock改成output，根据之前部分rk3568机型的相关故障改的。

ynnnas，是直接替换img里面的dtb就能用了？

jjm2473, 2023-11-27:
所有rk35xx机型都是同一个内核和rootfs，你说呢

jjm2473, 2023-11-27:
现在日志显示什么

hx801217, 2023-11-27:
> 现在日志显示什么

sorry，我没焊TTL，这就找个有的人

jjm2473, 2023-11-27:
把tx_delay和rx_delay改成原版的试试：
```
	tx_delay = <0x41>;
	rx_delay = <0x2e>;
```

jjm2473, 2023-11-27:
还有，这个板子是否同时有emmc和tf卡？有没有刷不同的固件？

hx801217, 2023-11-27:
> 还有，这个板子是否同时有emmc和tf卡？有没有刷不同的固件？

没有TF，只有u3 u2 各一个，有个内置sata接口，和emmc，flippy大的固件可以用，但是和istore固件目录不一样，他是按照armbian格式整的

hx801217, 2023-11-27:
> 把tx_delay和rx_delay改成原版的试试：
> 
> ```
> 	tx_delay = <0x41>;
> 	rx_delay = <0x2e>;
> ```

刚才送孩子去了，不好意思，改了原rx tx也不行，我是按照标准rockchip刷机的，这机器没有大神做UBOOT啊。
[image]

hx801217, 2023-11-27:
> tx_delay = <0x41>;
> 	rx_delay = <0x2e>;

[日志.txt](https://github.com/istoreos/istoreos/files/13470345/default.txt)

hx801217, 2023-11-27:
> 那肯定跟网卡无关了

用原机替换0928版本，完全正常，日志如下：
[正常日志.txt](https://github.com/istoreos/istoreos/files/13470520/default.txt)

jjm2473, 2023-11-27:
那就用0928呗，我没有测试机，而且其他rk3566机型都没问题。而且你的日志都停在开机过程，已经不是网络问题了。

jjm2473, 2023-11-27:
ynnnas的dts文件在这里，或许你可以对比下 https://github.com/istoreos/istoreos/blob/istoreos-22.03/target/linux/rockchip/dts/rk3568/rk3566-ynnnas.dts 

需要注意的是 ynnnas 使用 pwm 调频，不是使用 pmic（rk809等）

hx801217, 2023-11-27:
>

> ynnnas的dts文件在这里，或许你可以对比下 https://github.com/istoreos/istoreos/blob/istoreos-22.03/target/linux/rockchip/dts/rk3568/rk3566-ynnnas.dts
> 
> 需要注意的是 ynnnas 使用 pwm 调频（调节CPU/内存电压和频率），不是使用 pmic（rk809等）

谢谢大大，我认真学习学习

hx801217, 2023-11-27:
> ynnnas的dts文件在这里，或许你可以对比下 https://github.com/istoreos/istoreos/blob/istoreos-22.03/target/linux/rockchip/dts/rk3568/rk3566-ynnnas.dts
> 
> 需要注意的是 ynnnas 使用 pwm 调频（调节CPU/内存电压和频率），不是使用 pmic（rk809等）

重新编译了个ynnnas，至少istoreos ready了，还是那句，网卡用了通用驱动，跑不起啦。不知道应该怎么定义
[ynn_ynnnas 荐片盒子启动日志.txt](https://github.com/istoreos/istoreos/files/13472960/ynn_ynnnas.txt)

[   14.924089] rk_gmac-dwmac fe010000.ethernet: Failed to reset the dma
[   14.924139] rk_gmac-dwmac fe010000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   14.924151] rk_gmac-dwmac fe010000.ethernet eth0: stmmac_open: Hw setup failed
[   20.711778] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-1:00] driver [Generic PHY] (irq=POLL)
[   21.750731] rk_gmac-dwmac fe010000.ethernet: Failed to reset the dma
[   21.750819] rk_gmac-dwmac fe010000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   21.750838] rk_gmac-dwmac fe010000.ethernet eth0: stmmac_open: Hw setup failed
[   21.754470] br-lan: port 1(eth0) entered blocking state
[   21.754548] br-lan: port 1(eth0) entered disabled state
[   21.755547] device eth0 entered promiscuous mode
[   25.285444] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   25.285579] NFSD: Using legacy client tracking operations.
[   25.285594] NFSD: starting 10-second grace period (net f0000088)

jjm2473, 2023-12-15:
试试把 `compatible = "ethernet-phy-ieee802.3-c22";` 改成 `compatible = "ethernet-phy-id001c.c915", "ethernet-phy-ieee802.3-c22";`