istoreos/istoreos #1129: ARS2在更新固件OpenWrt 21.02.3 r16952-74af9267c7后出现内存占用率持续增加并自动重新启动情况，恳请给予解决为盼，谢谢。

d1857, 2023-11-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：系统一开始内存占用只有50%，运行10天后内存飙升到95%，不知为何？重启后内存占用马上降为50%，运行10天后又达到95%，已经观察了20天，真是奇怪了。我是从咱们这边买的ARS2软路由，系统使用最新的istoreos。有时候系统自动重启。

### (2) 路由器型号和固件版本

A：ARS2，固件版本：OpenWrt 21.02.3 r16952-74af9267c7

### (3) 详细日志和/或截图

A：安装软件如下图：
[image]
安装的容器如下：
[image]
系统占用内存率如下：
[image]
11月6日重启后内存占用如下：
[image]
11月15日系统内存占用又达到95%：
[image]

刚才使用命令free -h 查看，发现内存占用在一点一点增加：
[image]

现在一直别无他法，只能是观察内存占用达到95%后就手动重新启动机器，否则就自动重启了。恳请抽时间给予解决，谢谢。

jjm2473, 2023-11-24:
其实是jellyfin的内存泄漏bug，暂时没看到官方有修复。停用jellyfin就行了。使用 `top -m` 命令可以看到进程的内存占用情况（“RES”列）

jjm2473, 2023-11-24:
如果你没有使用jellyfin，同样用 top -m 命令看看哪个进程的内存占用异常

closed 2023-11-24 by jjm2473