istoreos/istoreos #1132: 【重要】防火墙安全问题

bcseputetto, 2023-11-23:
[image]
[image]
[image]
[image]
[image]
[image]

目前iStoreOS的防火墙已经对 **WAN是PPPoE协议时做出了对策**，将WAN入站默认设置为丢弃，避免用户WAN是PPPoE时获取的是公网IP，而导致默认密码或者弱密码被扫被骇的情况。
但对于WAN为DHCP客户端时则默认WAN入站为接受。
考虑到国内的家庭网络环境，WAN是DHCP客户端的情况下的确基本上没有直接能获取公网的IP的，大部分都是作为二级路由。但少部分国内用户可能跟上列聊天截图中的海外用户一样，可以直接通过DHCP获取到公网IP，例如下列江苏电信 5000Mbps的用户
[江苏电信 5000M 宽带体验](https://www.v2ex.com/t/882261)
文中提到认证方式为IPoE，而非PPPoE，光猫后面的设备DHCP可以直接获取到公网IP，最多可以提供**16 个公网 ipv4 ，16 个 /60 ipv6 pool**
在这种情况下，iStoreOS在WAN为DHCP客户端协议时，将WAN区域入站默认设置为接受，就会是一件比较危险的安全问题。特别是用户还没有能力察觉自己处于一个暴露在防火墙外边的状态。
希望能重新考虑一下WAN为DHCP客户端时的防火墙默认设置。避免在出现上图中的情况，**特别是有海外的用户拿iStoreOS作为主路由时，DHCP直接拿到公网IP的可能性会比国内大得多。**

jjm2473, 2023-11-24:
但是也不能默认禁用wan口的入站，有些人可能想当成nas使用的。对于dhcp能获取到公网ip的情况，就手动把防火墙wan区域的入站改成丢弃吧

bcseputetto, 2023-11-24:
> 但是也不能默认禁用wan口的入站，有些人可能想当成nas使用的。对于dhcp能获取到公网ip的情况，就手动把防火墙wan区域的入站改成丢弃吧

的确也有这方面的顾虑，就是担心DHCP能获取到公网IP用户会不会像聊天里的那位用户一样，对自己暴露在外没有任何认知。被入侵后怀疑是iStoreOS的问题

jjm2473, 2023-11-24:
那没办法，等之后出国际版再改这个规则吧，现在只能让用户注意这个问题，对国内用户应该没太大影响。

你说还有个ipoe协议，可以加到wan_drop的判断条件里，或者改成只有DHCP和静态IP不自动drop。

NJYZ-96, 2023-12-10:
网络配置向导那里加个提示不就行了，或者加个判断条件，DHCP+公网ip就跳出来让用户确认；是我想简单了吗？别介意哈，小白不懂

NJYZ-96, 2023-12-10:
> 那没办法，等之后出国际版再改这个规则吧，现在只能让用户注意这个问题，对国内用户应该没太大影响。
> 
> 你说还有个ipoe协议，可以加到wan_drop的判断条件里，或者改成只有DHCP和静态IP不自动drop。

网络配置向导那里加个提示不就行了，或者加个判断条件，DHCP+公网ip就跳出来让用户确认；是我想简单了吗？别介意哈，小白不懂

jjm2473, 2023-12-15:
新版本固件已经默认禁止wan口入站了，如果要开就自己在防火墙配置吧。（拨号的情况下还是会自动禁止wan口入站）

closed 2023-12-15 by jjm2473