istoreos/istoreos #1135: nanopi Rs6  DDNS  dynv6 问题

pili-book, 2023-11-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

FriendlyElec NanoPi R6S

iStoreOS 22.03.4 2023071416

### (3) 详细日志和/或截图

A：动态域名运行正常，但是在  已注册的 IP 地址  那一栏显示是无数据

bcseputetto, 2023-11-25:
检查一下 DHCP/DNS，高级设置里，过滤AAAA IPv6记录 是否勾选，尝试取消勾选后，再观察观察

pili-book, 2023-11-26:
> 检查一下 DHCP/DNS，高级设置里，过滤AAAA IPv6记录 是否勾选，尝试取消勾选后，再观察观察
弄了那个选项后确实能获取到数据了，但是又有新的问题
145744       : Waiting 300 seconds (Check Interval)
 150244       : Detect registered/public IP
 150244       : #> /usr/bin/nslookup ipv6.dynv6.com  >/var/run/ddns/ipv6.dat 2>/var/run/ddns/ipv6.err
 150244       : Registered IP '2a01:4f8:1c1c:4c96
2a01:4f9:c010:95b
2a03:4000:6:569' detected
 150244  WARN : Error in 'expand_ipv6()' - invalid IPv6 found: '2a01:4f8:1c1c:4c96
2a01:4f9:c010:95b
2a03:4000:6:569' expanded: '2a01:04f8:1c1c:4c96
2a01:04f9:c010:095b
2a03:4000:0006:0569'
 150244  WARN : Updating IP at DDNS provider failed - starting retry 59/0
 150244       : Detect local IP on 'interface'
 150244       : #> ip -o addr show dev pppoe-wan scope global >/var/run/ddns/ipv6.dat 2>/var/run/ddns/ipv6.err
 150244       : Local IP '240e:324:203:143f:62e9:7e8e:918a:f858' detected on interface 'pppoe-wan'
 150244       : Update needed - L: '240e:0324:0203:143f:62e9:7e8e:918a:f858' <> R: 'invalid'
 150244       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/ipv6.dat -o /var/run/ddns/ipv6.err --no-proxy 'http://dynv6.com/api/update?hostname=suhuanzhen.dynv6.net&token=***PW***&ipv6=240e:0324:0203:143f:62e9:7e8e:918a:f858'
 150245       : DDNS Provider answered:
addresses updated
 150245  info : Update successful - IP '240e:0324:0203:143f:62e9:7e8e:918a:f858' send
 150245  info : Forced update successful - IP: '240e:0324:0203:143f:62e9:7e8e:918a:f858' send
 150245       : Waiting 300 seconds (Check Interval)
循环5分钟检查，每次好像从服务商获得的Ip好像不正确，本来是5分钟检测一次，结果是每5分钟就强行更新不知道为啥

closed 2023-11-27 by jjm2473