istoreos/istoreos #1136: 设置IPTV，但找不到 “物理接口” -> “IGMP嗅探/IGMP Snooping” 设置

Guancheng-Li, 2023-11-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 有类似的issue，但并不能解决问题：https://github.com/istoreos/istoreos/issues/622

## 2. 详细叙述

### (1) 具体问题

A：在istoreos 中找不到 "物理接口" -> " IGMP嗅探 / IGMP Snooping" 设置

### (2) 路由器型号和固件版本

A：
设备型号：FastRhino R66s
固件版本：iStoreOS 22.03.5 2023090810
内核版本：5.10.176

### (3) 详细日志和/或截图

A：
[image]

bcseputetto, 2023-11-25:
你需要在【网络】-【接口】-【设备】 里配置 br-lan，高级设备选项里找到。
他人编译的OpenWrt通常物理设备选项都直接在lan里编辑(比如包括LAN的桥接接口)
iStoreOS你需要到设备里去编辑br-lan这个网桥设备

Guancheng-Li, 2023-12-17:
> 你需要在【网络】-【接口】-【设备】 里配置 br-lan，高级设备选项里找到。 他人编译的OpenWrt通常物理设备选项都直接在lan里编辑(比如包括LAN的桥接接口) iStoreOS你需要到设备里去编辑br-lan这个网桥设备

B站能否给出个教程呢，搞半天没搞明白

jjm2473, 2023-12-18:
网络-接口，每一个页面都点一下，每个编辑按钮都点一下，不就知道了

Guancheng-Li, 2023-12-30:
> 网络-接口，每一个页面都点一下，每个编辑按钮都点一下，不就知道了

听君一席话，如听一席话

jjm2473, 2024-01-02:
@Guancheng-Li  图形界面的东西还要一步一步教，那还不如直接给你个命令呢。而且这些功能又不是istoreos发明的，去找openwrt的教程啊。前面已经有人告诉你详细的路径了，你卡在哪一步了？