istoreos/istoreos #1137: 报错：TypeError Cannot read properties of null (reading '.name')

slayercjc, 2023-11-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络--接口 点wan_6的编辑报错

### (2) 路由器型号和固件版本

A：X86，
固件版本 iStoreOS 22.03.5 2023102711

### (3) 详细日志和/或截图

A：
[image]

bcseputetto, 2023-11-25:
你不应该去编辑这个 wan_6，它是 wan生成的动态接口，如果你不想要wan 生成这个wan_6获取IPv6，你应该去编辑 wan，高级设置，获取IPv6地址 改为已禁用，这样wan就不会自动生成 wan_6 接口

slayercjc, 2023-11-25:
> You should not edit this wan_6, it is a dynamic interface generated by the wan, if you don't want the wan to generate this wan_6 get IPv6, you should go to edit the wan, advanced settings, get the IPv6 address changed to canceled, so that the wan will not automatically generate the wan_6 interface

这个就是编辑不了的是吗？

slayercjc, 2023-11-25:
[image]
应该是取消哪个呢？我没找到获取IPV6地址那个选项啊

bcseputetto, 2023-11-25:
> ![11](https://private-user-images.githubusercontent.com/14236188/285586837-780ffefc-a19b-4356-aa70-0dbb8a08e3fe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5MTE5OTksIm5iZiI6MTcwMDkxMTY5OSwicGF0aCI6Ii8xNDIzNjE4OC8yODU1ODY4MzctNzgwZmZlZmMtYTE5Yi00MzU2LWFhNzAtMGRiYjhhMDhlM2ZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI1VDExMjgxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4YmQyNDFiMmRjMWZhMjBjOGM5ZDQ0Y2I1YzIyOWIwZjQ3YjdmMTQyMmM0YWJkOGM1ZjZlMmY5NmI4MGNiMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.D7MIGjmbQHJppoMNbBt9NR2pl4S0lONmmldlwPYttu4) 应该是取消哪个呢？我没找到获取IPV6地址那个选项啊

编辑WAN，高级设置

slayercjc, 2023-11-25:
> 编辑WAN，高级设置 ![图像](https://private-user-images.githubusercontent.com/29139573/285588379-189e29a6-5e19-4c7f-855b-5019a48a7231.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5MTkzMTMsIm5iZiI6MTcwMDkxOTAxMywicGF0aCI6Ii8yOTEzOTU3My8yODU1ODgzNzktMTg5ZTI5YTYtNWUxOS00YzdmLTg1NWItNTAxOWE0OGE3MjMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI1VDEzMzAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmODY1OGI5NDk3MWNkY2YyNWM1OGZjZWFjNjg1NmRkMDlhYTJmYzEwMGMxMDg3NTc2YmQ4ZTQ3NWY2MjM3ZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pi3oGmoMlDsJbf2Ka3PqDlrG13UJ7_-Qo0HMEyXLwFs)

好的 谢谢大佬找到了 选择已禁用就没那个wan_6了，但是我想问一下那个wan_6点编辑报错的话是正常的是吗？还是我固件有什么问题需要重新刷固件吗？

bcseputetto, 2023-11-25:
> 好的 谢谢大佬找到了 选择已禁用就没那个wan_6了，但是我想问一下那个wan_6点编辑报错的话是正常的是吗？还是我固件有什么问题需要重新刷固件吗？

不是你的问题，但这个自动生成的wan_6就算你能点开编辑，里面也是空空如也，啥都没有。

slayercjc, 2023-11-25:
> 不是你的问题，但这个自动生成的wan_6就算你能点开编辑，里面也是空空如也，啥都没有。

好的谢谢，能再问个问题吗？我这IPV6，
[image]
但是访问https://test-ipv6.com/这个网站，却说没有IPV6是怎么回事呢？
[image]

bcseputetto, 2023-11-25:
> > > > 编辑WAN，高级设置 ![图像](https://private-user-images.githubusercontent.com/29139573/285588379-189e29a6-5e19-4c7f-855b-5019a48a7231.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5MTkzMTMsIm5iZiI6MTcwMDkxOTAxMywicGF0aCI6Ii8yOTEzOTU3My8yODU1ODgzNzktMTg5ZTI5YTYtNWUxOS00YzdmLTg1NWItNTAxOWE0OGE3MjMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI1VDEzMzAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmODY1OGI5NDk3MWNkY2YyNWM1OGZjZWFjNjg1NmRkMDlhYTJmYzEwMGMxMDg3NTc2YmQ4ZTQ3NWY2MjM3ZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pi3oGmoMlDsJbf2Ka3PqDlrG13UJ7_-Qo0HMEyXLwFs)
> > > 
> > > 
> > > 好的 谢谢大佬找到了 选择已禁用就没那个wan_6了，但是我想问一下那个wan_6点编辑报错的话是正常的是吗？还是我固件有什么问题需要重新刷固件吗？
> > 
> > 
> > 不是你的问题，但这个自动生成的wan_6就算你能点开编辑，里面也是空空如也，啥都没有。
> 
> 好的谢谢，能再问个问题吗？我这IPV6， ![88](https://private-user-images.githubusercontent.com/14236188/285594308-362758c2-dd83-4738-9d23-cccbe06d2dd2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5MTk3NjMsIm5iZiI6MTcwMDkxOTQ2MywicGF0aCI6Ii8xNDIzNjE4OC8yODU1OTQzMDgtMzYyNzU4YzItZGQ4My00NzM4LTlkMjMtY2NjYmUwNmQyZGQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI1VDEzMzc0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg2MDBlZTJlNWJmM2I2ZmYxMWE2N2Y2MmM0ZWNlYmE2OWU0ZTkzN2ZmOGMyMjVjNzVlMGJiMGZkMzE3ODMxZGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ild_c57jvc2-6A-9LoVOTH7-1JUpYrAxLAuEDdS_ebk) 但是访问[https://test-ipv6.com/这个网站，却说没有IPV6是怎么回事呢？](https://test-ipv6.com/%E8%BF%99%E4%B8%AA%E7%BD%91%E7%AB%99%EF%BC%8C%E5%8D%B4%E8%AF%B4%E6%B2%A1%E6%9C%89IPV6%E6%98%AF%E6%80%8E%E4%B9%88%E5%9B%9E%E4%BA%8B%E5%91%A2%EF%BC%9F) ![image](https://private-user-images.githubusercontent.com/14236188/285594344-2c6a4695-3242-4273-b8ca-f647846b39a7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA5MTk3NjMsIm5iZiI6MTcwMDkxOTQ2MywicGF0aCI6Ii8xNDIzNjE4OC8yODU1OTQzNDQtMmM2YTQ2OTUtMzI0Mi00MjczLWI4Y2EtZjY0Nzg0NmIzOWE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI1VDEzMzc0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMxYjFlYjFjMTRkZjY4YjQ0NTUzZjI0ZWZhYTY3ZTg4NWYyM2E5ZmFiY2RlN2U3MWRlNWE5NzFkMzgyMDE5MzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bOG0gk_LkQ7PoS8c4TfUI9qopri9SyxuCOja6DwQni0)

通过网站去检测IPv6时，需要你的DNS能正常解析AAAA IPv6记录，iStoreOS默认在DHCP/DNS 高级设置里面勾选了过滤AAAA IPv6记录，你需要取消勾选后，再通过网站检测IPv6，推荐用 ipw.cn 来测试

bcseputetto, 2023-11-25:
还有，国内DNS不推荐用8.8.8.8这个谷歌DNS，这个DNS目前在中国大陆没有节点，最近的也在香港，延迟和丢包都不理想，目前网上一堆过时的教程推荐用8.8.8.8，目前来说一点都不推荐

slayercjc, 2023-11-25:
> 还有，国内DNS不推荐用8.8.8.8这个谷歌DNS，这个DNS目前在中国大陆没有节点，最近的也在香港，延迟和丢包都不理想，目前网上一堆过时的教程推荐用8.8.8.8，目前来说一点都不推荐 114DNS 是一个很老牌的DNS，但它们不怎么做技术升级，推荐用大厂阿里的公共DNS 223.5.5.5，和腾讯旗下DNSPod提供的DNS 119.29.29.29

好的 我现在把DNS改了去

bcseputetto, 2023-11-25:
你可以查看你电脑设备的网卡有没有从路由器获取到IPv6，手机也可以通过Wi-Fi的详细信息查看到，如果从路由器获取到IPv6，网站检查却没有，一般就是DNS问题，iStoreOS的话，默认过滤了IPv6记录的解析，所以网站无法测试IPv6，去取消勾选“过滤AAAA IPv6记录”

slayercjc, 2023-11-25:
> 119.29.29.29

好像是我的openclash版本的问题吧，我看他们都有IPV6设置，我的没有 是版本太老了吗？

这是别人的
[image]
这是我的。。。
[image]

请问OPENCLASH怎么更新版本呢？能教下吗 大佬

bcseputetto, 2023-11-25:
> 好像是我的openclash版本的问题吧，我看他们都有IPV6设置，我的没有 是版本太老了吗？
>
> 这是别人的 [...]

你已经装了旧版本的openclash，那你就直接去openclash的github项目，下载新版的 ipk，然后安装，它项目里写的安装依赖你不用去管，装了旧版了依赖其实都已经有了，你就装下新版ipk就行，顺便，如果ipv6测试网站走了代理也可能无法测试出ipv6，所以我推荐你用 ipw.cn来测试，因为这个 cn网站被代理的可能性很低

slayercjc, 2023-11-25:
> 你已经装了旧版本的openclash，那你就直接去openclash的github项目，下载新版的 ipk，然后安装，它项目里写的安装依赖你不用去管，装了旧版了依赖其实都已经有了，你就装下新版ipk就行，顺便，如果ipv6测试网站走了代理也可能无法测试出ipv6，所以我推荐你用 ipw.cn来测试，因为这个 cn网站被代理的可能性很低

openclash 不是删库跑路了吗？他们还有github吗

bcseputetto, 2023-11-25:
> openclash 不是删库跑路了吗？他们还有github吗

clash跑路了，openclash并没有

slayercjc, 2023-11-25:
> clash跑路了，openclash并没有

好吧  我以为都跑了。。。。。。多谢大佬教这么多

closed 2024-02-02 by jjm2473