istoreos/istoreos #1140: 关于upnp系统日志刷屏的问题

huang450404, 2023-11-26:
路由器拨号，wan口拿到的是内网ip，upnp开了强制转发，工作正常但系统日志不断出现：
Sun Nov 26 16:39:26 2023 daemon.notice miniupnpd[29016]: force_forwarding enable, private/reserved address 100.106.xxx.xx used as external IP
Sun Nov 26 16:39:36 2023 daemon.notice miniupnpd[29016]: private/reserved address 100.106.xxx.xx is not suitable for external IP
此两条信息将系统日志刷屏，有什么办法能够屏蔽此类信息？

yeefanwong, 2023-11-27:
我也有类似的问题.
硬件是H28K.

Mon Nov 27 09:25:06 2023 daemon.err miniupnpd[28554]: Failed to convert hostname '240e:3bb:633:2a0:cca1:fdff:fee8:1211' to ip address
Mon Nov 27 09:35:06 2023 daemon.err miniupnpd[28554]: Failed to convert hostname '240e:3bb:633:2a0:cca1:fdff:fee8:1211' to ip address
Mon Nov 27 09:45:06 2023 daemon.err miniupnpd[28554]: Failed to convert hostname '240e:3bb:633:2a0:cca1fdff:fee8:1211' to ip address

jjm2473, 2023-11-27:
@huang450404 你这个日志是因为你的ip不是公网ip，所以有警告。如果你不需要强制转发私有ip，可以在upnp的高级设置把强制转发关掉。如果你不需要upnp功能，还可以把整个upnp关掉。

jjm2473, 2023-11-27:
@yeefanwong 试试执行这段脚本：
```
uci -q batch <<-EOF >/dev/null
    del upnpd.config.ipv6_disable
    commit upnpd
EOF

```

然后重启下，看看还会不会出现这个日志。

yeefanwong, 2023-11-27:
> @yeefanwong 试试执行这段脚本：
> 
> ```
> uci -q batch <<-EOF >/dev/null
>     del upnpd.config.ipv6_disable
>     commit upnpd
> EOF
> ```
> 
> 然后重启下，看看还会不会出现这个日志。

测试还是会出现这个log.

jjm2473, 2023-12-15:
@huang450404 你的网心云跟upnp在同一台机器上运行吗？那网心云根本就不会用到这个upnp服务