istoreos/istoreos #1147: i226 网卡在 2.5g 状态下会间歇性的断网

BlueD, 2023-11-28:
反馈bug/问题模板，提建议请删除

## i226 网卡在 2.5g 状态下会间歇性的断网

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
主路由模式，局域网口是 2.5g，广域网口是千兆
局域网口每隔几天会断线，严重时要重启才能恢复

### (2) 路由器型号和固件版本

A：
畅网 N5105 V5 旗舰版，i266-v 四口。
iStoreOS 22.03.5 2023102711

### (3) 详细日志和/或截图

A：
Tue Nov 28 13:09:36 2023 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: u.wwwww.wtf
Tue Nov 28 13:22:39 2023 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: w.wwwww.wtf
Tue Nov 28 13:22:41 2023 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: retracker.sevstar.net
Tue Nov 28 13:23:33 2023 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: retracker.sevstar.net
Tue Nov 28 13:50:36 2023 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: u.wwwww.wtf
Tue Nov 28 14:02:27 2023 daemon.notice netifd: Network device 'eth1' link is down
Tue Nov 28 14:02:27 2023 kern.info kernel: [93492.626792] igc 0000:04:00.0 eth1: NIC Link is Down
Tue Nov 28 14:02:27 2023 kern.info kernel: [93492.627137] br-lan: port 1(eth1) entered disabled state
Tue Nov 28 14:02:31 2023 daemon.notice netifd: Network device 'eth1' link is up
Tue Nov 28 14:02:31 2023 kern.info kernel: [93496.877070] igc 0000:04:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Tue Nov 28 14:02:31 2023 kern.info kernel: [93496.877218] br-lan: port 1(eth1) entered blocking state
Tue Nov 28 14:02:31 2023 kern.info kernel: [93496.877224] br-lan: port 1(eth1) entered forwarding state
Tue Nov 28 14:02:31 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.199 04:7c:16:c5:00:e4
Tue Nov 28 14:02:31 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.199 04:7c:16:c5:00:e4 PC2600
Tue Nov 28 14:02:33 2023 daemon.notice netifd: Network device 'eth1' link is down
Tue Nov 28 14:02:33 2023 kern.info kernel: [93498.566714] igc 0000:04:00.0 eth1: NIC Link is Down
Tue Nov 28 14:02:33 2023 kern.info kernel: [93498.566887] br-lan: port 1(eth1) entered disabled state
Tue Nov 28 14:02:37 2023 daemon.notice netifd: Network device 'eth1' link is up
Tue Nov 28 14:02:37 2023 kern.info kernel: [93502.556999] igc 0000:04:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Tue Nov 28 14:02:37 2023 kern.info kernel: [93502.557147] br-lan: port 1(eth1) entered blocking state
Tue Nov 28 14:02:37 2023 kern.info kernel: [93502.557154] br-lan: port 1(eth1) entered forwarding state
Tue Nov 28 14:02:37 2023 kern.info kernel: [93502.576705] igc 0000:04:00.0 eth1: NIC Link is Down
Tue Nov 28 14:02:38 2023 daemon.notice netifd: Network device 'eth1' link is down
Tue Nov 28 14:02:38 2023 kern.info kernel: [93503.626675] br-lan: port 1(eth1) entered disabled state
Tue Nov 28 14:02:41 2023 daemon.notice netifd: Network device 'eth1' link is up
Tue Nov 28 14:02:41 2023 kern.info kernel: [93507.396940] igc 0000:04:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Tue Nov 28 14:02:41 2023 kern.info kernel: [93507.397088] br-lan: port 1(eth1) entered blocking state
Tue Nov 28 14:02:41 2023 kern.info kernel: [93507.397094] br-lan: port 1(eth1) entered forwarding state
Tue Nov 28 14:02:41 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.199 04:7c:16:c5:00:e4
Tue Nov 28 14:02:41 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.199 04:7c:16:c5:00:e4 PC2600
Tue Nov 28 14:02:42 2023 kern.info kernel: [93508.246630] igc 0000:04:00.0 eth1: NIC Link is Down
Tue Nov 28 14:02:42 2023 daemon.notice netifd: Network device 'eth1' link is down
Tue Nov 28 14:02:42 2023 kern.info kernel: [93508.406605] br-lan: port 1(eth1) entered disabled state
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876599] igc 0000:04:00.0 eth1: Register Dump
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876607] igc 0000:04:00.0 eth1: Register Name   Value
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876616] igc 0000:04:00.0 eth1: CTRL            181c0641
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876622] igc 0000:04:00.0 eth1: STATUS          00780681
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876627] igc 0000:04:00.0 eth1: CTRL_EXT        100000c0
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876632] igc 0000:04:00.0 eth1: MDIC            18017949
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876637] igc 0000:04:00.0 eth1: ICR             00000001
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876642] igc 0000:04:00.0 eth1: RCTL            0440803a
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876654] igc 0000:04:00.0 eth1: RDLEN[0-3]      00001000 00001000 00001000 00001000
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876665] igc 0000:04:00.0 eth1: RDH[0-3]        00000056 000000ac 00000068 000000e6
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876676] igc 0000:04:00.0 eth1: RDT[0-3]        00000055 000000ab 00000067 000000e5
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876688] igc 0000:04:00.0 eth1: RXDCTL[0-3]     02040808 02040808 02040808 02040808
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876699] igc 0000:04:00.0 eth1: RDBAL[0-3]      0e3b5000 0e634000 0e637000 0e36e000
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876710] igc 0000:04:00.0 eth1: RDBAH[0-3]      00000001 00000001 00000001 00000001
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876715] igc 0000:04:00.0 eth1: TCTL            a503f0fa
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876726] igc 0000:04:00.0 eth1: TDBAL[0-3]      0e3b5000 0e634000 0e637000 0e36e000
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876737] igc 0000:04:00.0 eth1: TDBAH[0-3]      00000001 00000001 00000001 00000001
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876748] igc 0000:04:00.0 eth1: TDLEN[0-3]      00001000 00001000 00001000 00001000
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876759] igc 0000:04:00.0 eth1: TDH[0-3]        000000fe 0000002e 0000009e 00000087
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876770] igc 0000:04:00.0 eth1: TDT[0-3]        00000001 0000002e 0000009e 00000087
Tue Nov 28 14:02:44 2023 kern.info kernel: [93509.876781] igc 0000:04:00.0 eth1: TXDCTL[0-3]     02100108 02100108 02100108 02100108
Tue Nov 28 14:02:44 2023 kern.err kernel: [93509.876784] igc 0000:04:00.0 eth1: Reset adapter
Tue Nov 28 14:02:47 2023 daemon.notice netifd: Network device 'eth1' link is up
Tue Nov 28 14:02:47 2023 kern.info kernel: [93513.106690] igc 0000:04:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Tue Nov 28 14:02:47 2023 kern.info kernel: [93513.106801] br-lan: port 1(eth1) entered blocking state
Tue Nov 28 14:02:47 2023 kern.info kernel: [93513.106803] br-lan: port 1(eth1) entered forwarding state
Tue Nov 28 14:02:47 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.199 04:7c:16:c5:00:e4
Tue Nov 28 14:02:47 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.199 04:7c:16:c5:00:e4 PC2600
Tue Nov 28 14:02:48 2023 kern.info kernel: [93513.516513] igc 0000:04:00.0 eth1: NIC Link is Down
Tue Nov 28 14:02:48 2023 daemon.notice netifd: Network device 'eth1' link is down
Tue Nov 28 14:02:48 2023 kern.info kernel: [93514.116528] br-lan: port 1(eth1) entered disabled state
Tue Nov 28 14:02:57 2023 daemon.notice netifd: Network device 'eth1' link is up
Tue Nov 28 14:02:57 2023 kern.info kernel: [93522.956724] igc 0000:04:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Tue Nov 28 14:02:57 2023 kern.info kernel: [93522.956874] br-lan: port 1(eth1) entered blocking state
Tue Nov 28 14:02:57 2023 kern.info kernel: [93522.956880] br-lan: port 1(eth1) entered forwarding state
Tue Nov 28 14:02:58 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.199 04:7c:16:c5:00:e4
Tue Nov 28 14:02:58 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.199 04:7c:16:c5:00:e4 PC2600
Tue Nov 28 14:02:59 2023 daemon.notice netifd: Network device 'eth1' link is down
Tue Nov 28 14:02:59 2023 kern.info kernel: [93525.326389] igc 0000:04:00.0 eth1: NIC Link is Down
Tue Nov 28 14:02:59 2023 kern.info kernel: [93525.326559] br-lan: port 1(eth1) entered disabled state
Tue Nov 28 14:03:04 2023 daemon.notice netifd: Network device 'eth1' link is up
Tue Nov 28 14:03:04 2023 kern.info kernel: [93529.616630] igc 0000:04:00.0 eth1: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Tue Nov 28 14:03:04 2023 kern.info kernel: [93529.616779] br-lan: port 1(eth1) entered blocking state
Tue Nov 28 14:03:04 2023 kern.info kernel: [93529.616785] br-lan: port 1(eth1) entered forwarding state
Tue Nov 28 14:03:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.199 04:7c:16:c5:00:e4
Tue Nov 28 14:03:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.199 04:7c:16:c5:00:e4 PC2600
Tue Nov 28 14:03:19 2023 user.info : luci: accepted login on / for root from 192.168.100.199

BlueD, 2023-11-28:
突然想起我用的是超五类线，所以先换线再说。

closed 2023-11-28 by BlueD

shabby2333, 2023-12-14:
n5105 i226换内核，至少6.2

BlueD, 2023-12-15:
反正我换完线就好了，所以就没再管它。
现在的 istore 的内核是多少，小于 6.2 吗？

BlueD, 2023-12-15:
> n5105 i226换内核，至少6.2

反正我换完线就好了，所以就没再管它。
现在 istore 的内核是多少，小于 6.2 吗？