istoreos/istoreos #1150: J1900刷了最新的固件,千兆网只有五百多的网速怎么解决

tanglei0809, 2023-12-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：J1900刷了最新的固件当主路由拨号上网,自己的小米路由器使用中继模式,网线直连,千兆宽带,电脑手机测速都只有五百多的网速
直接使用自己的小米路由器拨号上网能达到900多

### (2) 路由器型号和固件版本

A：J1900

### (3) 详细日志和/或截图

A：

tanglei0809, 2023-12-15:
> 打开后台，网络-接口-全局网络选项，勾选“数据包引导”试试，可能要重启下。
>
> （注意：非x86平台用户不需要配置，一般都有初始化脚本配置好了。）

istore路由器桥接光猫,拨号上网,现在的配置,直连istore能够正常获取ipv6地址并且访问ipv6网址,但是连接istore的二级路由器DHCP模式上网,连接二级路由器的设备无法正常获取到ipv6地址,二级路由器ipv6配置正常且已经关闭防火墙,之前没有istore路由器直接使用二级路由器拨号上网是正常的,
请问下,应该怎么配置istore路由器的网络接口

唐镭
***@***.***

jjm2473, 2023-12-16:
@tanglei0809 看置顶的issue，或者issue搜索ipv6

tanglei0809, 2023-12-19:
使用了systools里面的打开ipv6还是无法解决
![Uploading image.png…](
[image]
[image]
)

jjm2473, 2023-12-20:
你确定istoreos自己有获取到ipv6吗？

tanglei0809, 2023-12-20:
是的，istore主路由拨号有ipv6，二级小米路由也有，但是接在二级小米路由的设备没有获取到，设备直接接到主路由能获取到，二级小米路由是DHCP模式上网发自我的 iPhone

jjm2473, 2023-12-21:
@tanglei0809 既然二级路由也有ipv6了，那还跟主路由有什么关系？

closed 2023-12-21 by jjm2473

shengnannan, 2024-05-09:
同有此问题。
软路由：康耐信的n4100，英特尔2.5G网卡。
猫：vsol 2.5G
硬路由：中兴巡天PRO+ AX3000

硬路由拨号，可以跑满千兆宽带。
软路由拨号，只可以跑到730M上下。
内网测速正常。