istoreos/istoreos #1156: luck插件无法启动 老是闪一下启动了 闪一下又自动关闭了

pili-book, 2023-12-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
luck插件无法启动 老是闪一下启动了 闪一下又自动关闭了

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luck插件前几天使用同好的，后来想的麻烦就用scoat，将luck设置为不启动，今天想的还是用这个插件吧，结果无法启动 老是闪一下启动了 闪一下又自动关闭了，各种还原系统卸载什么的都不行，还是没法用，另外有个问题，为什么卸载安装后，luck原来的配置还在，登录界面的默认账号还有，为啥如此呢，卸载不干净吗

### (2) 路由器型号和固件版本

A：FriendlyElec NanoPi R6S
iStoreOS 22.03.4 2023071416 / LuCI istoreos-22.03 branch git-23.103.23275-0193674

### (3) 详细日志和/或截图

A：

Hopers, 2023-12-07:
/etc/lucky下面应该是各个组件的配置，卸载完后去定点删除把

gzxiexl, 2023-12-11:
我也碰到了 最后发现是文件权限的问题 chmod 777 搞定

pili-book, 2023-12-11:
我是后来找了个人帮看了下说是卸载后删除了配置文件夹，重新安装搞定的，你那个权限是修改哪个的权限了

gzxiexl, 2023-12-11:
就 lucky 的可执行文件 我忘了什么路径了

jjm2473, 2024-02-02:
你是不是没把socat关掉，端口冲突了？

closed 2024-02-02 by jjm2473