istoreos/istoreos #1158: Intel(R) Xeon(R) E-2174G CPU  显卡P60 如何安装jellyfin 支持硬件加速

zhibeen, 2023-12-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
Intel(R) Xeon(R) E-2174G CPU  显卡P60 如何安装jellyfin 支持硬件加速， istoreos 是否支持intel p630 核显

已经勾选了系统自带的驱动， 但是jellyfin 设置硬件加速还是无效
[image]

jjm2473, 2023-12-22:
如果不是近3年的产品，不要勾选，而且现在最新的固件可以自动判断要不要加载这个驱动，不需要用户选择

zypj1123, 2024-03-01:
> 如果不是近3年的产品，不要勾选，而且现在最新的固件可以自动判断要不要加载这个驱动，不需要用户选择

我的是n5000需要勾选吗？感觉jellyfin没有硬解

jjm2473, 2024-03-01:
自己看系统里面有没有 `/dev/dri/card*`，如果有，那基本是你的配置有问题。用istore里面的jellyfin插件部署，还有不要使用docker的国内镜像仓库，以免安装到jellyfin的老版本。

closed 2024-03-01 by jjm2473