istoreos/istoreos #1168: Station P2自带WiFi，无线网卡芯片AP6275S，请开发大神加加驱动。

xc-input, 2023-12-10:
Station P2自带WiFi，无线网卡芯片AP6275S，iStoreOS固件没有无线这一项，开发大神能给加加驱动吗，驱动在这个项目中https://github.com/xc-input/lede.git，我能力有限，不会弄，还请加一下，谢谢。

jjm2473, 2023-12-15:
你提供的链接没用，firefly的wifi驱动不在lede里面，而是在内核里面，且没有适配backport-mac80211，根本没法用。

你试试这个驱动行不行。先升级到今天的固件，下载下面的压缩包，解压以后，先安装rkwifi-firmware，然后安装kmod-。

[rkwifi.zip](https://github.com/istoreos/istoreos/files/13685470/rkwifi.zip)

注意这个驱动没有在AP6275S上测试过，所以就算系统崩溃也不要惊讶。

xc-input, 2023-12-16:
> 你提供的链接没用，firefly的wifi驱动不在lede里面，而是在内核里面，且没有适配backport-mac80211，根本没法用。
> 
> 你试试这个驱动行不行。先升级到今天的固件，下载下面的压缩包，解压以后，先安装rkwifi-firmware，然后安装kmod-。
> 
> [rkwifi.zip](https://github.com/istoreos/istoreos/files/13685470/rkwifi.zip)
> 
> 注意这个驱动没有在AP6275S上测试过，所以就算系统崩溃也不要惊讶。

您好，开发者，我用最新的的固件（12-15），并且先后按要求安装了两个ipk后，是无法正常使用的；好消息是有了无线这个选项，如图：
[image]
坏消息是无线参数是空的，如图：
[image]

jjm2473, 2023-12-16:
重启下，如果还是没有无线参数，把内核日志发出来看看

xc-input, 2023-12-17:
您好，开发者，附近是重启后的日志
[内核日志.txt](https://github.com/istoreos/istoreos/files/13695907/default.txt)
[系统日志.txt](https://github.com/istoreos/istoreos/files/13695908/default.txt)

xc-input, 2023-12-17:
> 重启下，如果还是没有无线参数，把内核日志发出来看看

您好，开发者，附近是重启后的日志
[内核日志.txt](https://github.com/istoreos/istoreos/files/13695914/default.txt)
[系统日志.txt](https://github.com/istoreos/istoreos/files/13695915/default.txt)

jjm2473, 2023-12-18:
没看到有什么报错，你点一下无线的扫描，看看能不能扫到附近的热点

jjm2473, 2023-12-18:
扫描之后再把内核日志发出来看看

lyzdroid, 2023-12-19:
> 扫描之后再把内核日志发出来看看

您好，开发者，这是扫描之后的内核日志和系统日志

[内核日志-扫描.txt](https://github.com/istoreos/istoreos/files/13713747/-.txt)

[系统日志-扫描.txt](https://github.com/istoreos/istoreos/files/13713749/-.txt)

jjm2473, 2023-12-20:
也没啥报错，可能还是fw文件的问题。你试试从firefly官方的固件里面提取`config.txt`，`fw_bcm43752a2_ag.bin`，`nvram_ap6275s.txt`，替换掉日志中出现的对应文件，注意官方固件中对应的文件可能路径不一样。

xc-input, 2023-12-20:
> 扫描之后再把内核日志发出来看看

您好，开发者，我的也扫描完了
[image]
[内核日志-扫描.txt](https://github.com/istoreos/istoreos/files/13726362/-.txt)
[系统日志-扫描.txt](https://github.com/istoreos/istoreos/files/13726363/-.txt)
我也试试看从firefly官方固件提取config.txt，fw_bcm43752a2_ag.bin，nvram_ap6275s.txt文件，导入看看效果。

ChaosHeadNo, 2024-03-11:
最新版istoreos在station p2上无法使用wifi，ipk中的kmod-报错无法安装