istoreos/istoreos #1170: NanoPi R6S 挂载网络共享插件问题bug

pili-book, 2023-12-10:
反馈bug/问题模板，提建议请删除

(1) 具体问题
A：我有一个小猫盘，能提供samba服务，想把他挂载到openwrt,但是各种填写那个内容在 挂载网络共享插件那里也无法实现挂载，也没有日志能看。。。。找不到原因 易有云文件查看是不行没挂载成功
使用mount -t cifs -o vers=2.0,username=@@@,password=@@@,iocharset=utf8 //192.168.2.4/@@@  /SmbShare
能挂载成功，说明插件肯定有问题
(2) 路由器型号和固件版本
A： FriendlyElec NanoPi R6S
StoreOS 22.03.4 2023071416 / LuCI istoreos-22.03 branch git-23.103.23275-0193674

Hopers, 2023-12-12:
密码开头是特殊符号？我在服务器fasb上做开机挂载webdav就是因为密码开头是特殊符号没开机挂载成功

pili-book, 2023-12-12:
这个是为了隐藏随便输的符号

jjm2473, 2023-12-15:
`mount -t cifs -o vers=2.0,username=@@@,password=@@@,iocharset=utf8 //192.168.2.4/@@@ /SmbShare`不是跟插件生成的命令一模一样吗？是不是你的共享名或者用户名密码有特殊字符？

pili-book, 2023-12-15:
没有特殊字符。。。你看那个截图，都是数字。。。就是搞不定怎么也不行

pili-book, 2023-12-15:
也没有啥日志什么的能看，也看不到日志说错在哪

jjm2473, 2023-12-15:
执行`/etc/init.d/cifs start`就能看到日志了

pili-book, 2023-12-16:
***@***.***:~#  /etc/init.d/cifs start
mount: bad usage
Try 'mount --help' for more information.

我查了下好像说是路径有问题，但是那个挂载路径那加斜杠不加斜杠都不行

李小虎
***@***.***

pili-book, 2023-12-16:
> 执行`/etc/init.d/cifs start`就能看到日志了

[image]
顺便wenx问下我用代码直接挂载后，过一会挂载的那就不见了，自己消失了

pili-book, 2023-12-16:
> 执行`/etc/init.d/cifs start`就能看到日志了

START=97

ENABLED=0
WORKGROUPD=0
IOCHARSET=0
GUEST=""
USERS=""
AGM=""

cifs_header() {
	local enabled
	local workgroup 

	config_get enabled $1 enabled
	config_get workgroup $1 workgroup

	ENABLED=$enabled
	WORKGROUPD=$workgroup
}

mount_natshare() {
	local server
	local name
	local natpath
	local guest
	local users
	local pwd
	local agm
	local iocharset
	local smbver

	config_get server $1 server
	config_get name $1 name
	config_get natpath $1 natpath
	config_get guest $1 guest
	config_get users $1 users
	config_get pwd $1 pwd
	config_get agm $1 agm
	config_get iocharset $1 iocharset
	config_get smbver $1 smbver
	
	mkdir -p $natpath && chmod 777 $natpath
	if [ "$smbver" = "nil" ];then
		smbverarg=""
	else
		smbverarg="vers=$smbver,"
	fi
	#echo "mount -t cifs -o ${smbverarg}username=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath"
	/usr/bin/mount -t cifs -o ${smbverarg}username=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath
打开程序观看
#echo "mount -t cifs -o ${smbverarg}username=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath"
	/usr/bin/mount -t cifs -o ${smbverarg}username=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath
要是把这里直接程用那就代码替换就能够运行，具体哪个参数错误我这小白就看不出来了

pili-book, 2023-12-16:
#echo "mount -t cifs -o ${smbverarg}username=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath"
	/usr/bin/mount -t cifs -o ${smbverarg}username=$users,password=$pwd,iocharset=$iocharset,$agm //$server/$name $natpath
                echo  "$natpat"

经使用 echo  "$natpat"输出这个变量，发现这个变量没有获取到数据

李小虎
***@***.***

closed 2023-12-16 by pili-book

jjm2473, 2023-12-16:
所以到底是什么问题？

pili-book, 2023-12-16:
ip地址后面有个小空格。。。不知道为什么会有这东西

jjm2473, 2023-12-16:
那估计是复制的时候多了空格

pili-book, 2023-12-16:
比较诡异。。。这东西我是手打的。。