istoreos/istoreos #1171: 启动项无法启动bug

pili-book, 2023-12-10:
反馈bug/问题模板，提建议请删除
(1) 具体问题

使用mount -t cifs -o vers=2.0,username=@@@,password=@@@,iocharset=utf8 //192.168.2.4/@@@ /SmbShare
能挂载成功，但是放在启动项-本地启动脚本那里确无法执行，经网上查询， /etc/rc.loca并没有启动权限，自己加上权限还是启动不成功，猜测还有别的地方有问题，应该说明插件肯定有问题
(2) 路由器型号和固件版本
A： FriendlyElec NanoPi R6S
StoreOS 22.03.4 2023071416 / LuCI istoreos-22.03 branch git-23.103.23275-0193674

Hopers, 2023-12-12:
我在服务器挂载webdav，在fasb做开机挂载时候，就是因为密码开头是特殊符号无法开机自动挂载，后面密码开头去掉特殊符号才开机自动挂载成功，您这用户名和密码开头都是特殊符号，会不会。。。

pili-book, 2023-12-12:
这是为了隐藏随便输的，实际账号密码都是数字

jjm2473, 2023-12-15:
你确定执行的时候 /SmbShare 文件夹存在？

pili-book, 2023-12-15:
有的，我新建了好几个文件夹都试过不行的

jjm2473, 2023-12-15:
改下命令，输出日志 `mount -t cifs -o vers=2.0,username=@@@,password=@@@,iocharset=utf8 //192.168.2.4/@@@ /SmbShare >/tmp/smb.log 2>&1` ，启动以后看看 /tmp/smb.log 是否存在，有没有内容

pili-book, 2023-12-15:
> 改下命令，输出日志 `mount -t cifs -o vers=2.0,username=@@@,password=@@@,iocharset=utf8 //192.168.2.4/@@@ /SmbShare >/tmp/smb.log 2>&1` ，启动以后看看 /tmp/smb.log 是否存在，有没有内容
这个log存在，内容如下
mount: /SmbShare: mount(2) system call failed: Host is down.

pili-book, 2023-12-15:
> 改下命令，输出日志 `mount -t cifs -o vers=2.0,username=@@@,password=@@@,iocharset=utf8 //192.168.2.4/@@@ /SmbShare >/tmp/smb.log 2>&1` ，启动以后看看 /tmp/smb.log 是否存在，有没有内容

pili-book, 2023-12-15:
> 改下命令，输出日志 `mount -t cifs -o vers=2.0,username=@@@,password=@@@,iocharset=utf8 //192.168.2.4/@@@ /SmbShare >/tmp/smb.log 2>&1` ，启动以后看看 /tmp/smb.log 是否存在，有没有内容

感谢，应该启动能用了，我延时了1分钟运行，成功了

closed 2023-12-15 by pili-book