istoreos/istoreos #1172: 23231208的86_64_uefi版本安装后IPV6有问题。

gdqpwl, 2023-12-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-12-15:
在配置好联网以后，istore里面安装系统便利工具（systools），然后执行“开启IPv6”试试，会自动根据当前联网方式调整ipv6设置，可能可以解决问题

shen-lan, 2023-12-20:
@jjm2473  
[image]
系统便利工具只有 关闭ipv6 呀

jjm2473, 2023-12-21:
@shen-lan 更新下

MAOGE2018, 2023-12-28:
单网口小主机挂在局域网里面，也没有开启ipv6选项。
[image]

主机名 | iStoreOS
-- | --
型号 | SUPoX COMPUTER CO.,LTD SUPoX Intel J3160NX7
架构 | Intel(R) Celeron(R) CPU J3160 @ 1.60GHz
目标平台 | x86/64
固件版本 | iStoreOS 22.03.5 2023121510 / LuCI istoreos-22.03 branch git-23.342.39264-37c783a
内核版本 | 5.10.176
本地时间 | 2023-12-28 12:23:02
运行时间 | 2d 1h 1m 33s
平均负载 | 8.54, 10.82, 10.65

MAOGE2018, 2023-12-28:
[image]
更新系统便利化工具后正常有了开启ipv6选项了。

Joker-lm, 2024-01-09:
> ![image](https://private-user-images.githubusercontent.com/39059232/293138416-9b5a5448-686e-4874-b8a7-13848f0d10a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]
为啥的我的没有系统便利工具

shen-lan, 2024-01-09:
@Joker-lm  系统便利化工具  是商店里的一个插件，你都没安装这个插件，怎么有？

138161, 2024-04-05:
第一步，先搞定wan6口
1、网络--接口--wan6--编辑
1.1、常规设置
     协议：DHCPv6客户端
         设备：eth0
         开机自动运行：选择
         请求IPv6地址：try
         请求指定长度的IPv6前缀：自动
       
1.2、高级设置
         强制链路：不选
         请求DHCP时发送的客户端ID：空
         使用默认网关：选择
         自动获取DNS服务器：选择
         DNS 权重：空
     使用网关跃点：空
     覆盖 IPv4 路由表：未指定
     覆盖 IPv6 路由表：未指定
     IPv6 源路由：选择
     委托 IPv6 前缀：选择
     IPv6 分配长度：已禁用
     IPv6 前缀过滤器：-- 请选择 --
     IPv6 后缀：::1
     IPv6 优先级：空
     
                          
1.3、防火墙设置
     创建和分配防火墙区域：wan和wan6
     
1.4、DHCP服务器--常规设置
     忽略此接口：不选
     
1.5、DHCP服务器--IPv6设置
     指定的主接口：选择
     RA 服务：中继模式
     DHCPv6 服务：中继模式
     NDP 代理：中继模式
     学习路由：选择
     

第二步，搞定lan口
2、网络--接口--lan--编辑
2.1、常规设置
     协议：静态地址
     设备：br-lan
     开机自动运行：选择
     IPv4 地址：192.168.100.1
     IPv4 子网掩码：255.255.255.0
     IPv4 网关：灰色无法更改
     IPv4 广播地址：灰色无法更改
     IPv6 地址：空
     IPv6 网关：空
     IPv6 路由前缀：空
     
2.2、高级设置
     强制链路：选择
     使用默认网关：选择
     使用自定义的 DNS 服务器：空
     DNS 搜索域名：空
     DNS 权重：空
     使用网关跃点：空
     覆盖 IPv4 路由表：未指定
     覆盖 IPv6 路由表：未指定
     委托 IPv6 前缀：选择
     IPv6 分配长度：已禁用
     IPv6 前缀过滤器：-- 请选择 --
     IPv6 后缀：::1
     IPv6 优先级：空
     
                       
[... 167 lines cut ...]
tcpdump  -nn -i eth0 icmp6 -v
tcpdump  -nn -i eth0 icmp6 -vv

查看邻居mac地址
ip -6 neighbor show 

检查哪些端口和应用在运行，如过滤19422相关内容
netstat -anp -p udp
netstat -anp -p udp | grep 19422

ipv6 DNS服务器
2001:4860:4860::8888
2001:4860:4860::8844
240e:4c:4008::1
240e:4c:4808::1
2001:dc7:1000::1
240C::6666
240C::6644
2400:3200::1
2402:4e00::
2400:da00::6666

重启网络服务
/etc/init.d/firewall restart
/etc/init.d/network restart

138161, 2024-04-11:
https://github.com/istoreos/istoreos/issues/1480
只有一个ipv6地址，没有PD前缀，可临时解决

jagerzhang, 2024-07-24:
求助一下大佬们，passwall 在ipv6梯子下无法使用负载均衡，不知道怎么设置才能正常：https://github.com/xiaorouji/openwrt-passwall/issues/3336

SparkCD, 2024-10-11:
istore本身能获取到ipv6   但是从lan连接到istore的设备没法获取到ipv6怎么解决

Edgar-qiqi, 2024-10-31:
[image]

防火墙 - NAT 规则 
不支持对ipv6进行重写源

yanyunsong, 2025-03-17:
防火墙-端口转发，不支持ipv6协议。
畅网N150，iStoreOS 22.03.7

[image]

SilenceCheung, 2025-04-11:
R5S 在系统便利化工具里面选择开启ipv6的地址选项，几个选项全部都设置了，但是链接网络的设备都还是无法获取到ipv6地址，这是为什么，系统也是最新下载的，没有做其他的设置，只下载了系统便利化工具进行开启的，PPOE拨号的，此前用的高大全openwrt是正常，换成iStoreOS 就不行了一直没有办法获取

SilenceCheung, 2025-04-11:
[image]  这些选项全部试过，都没有办法获取ipv6地址

[image]

kmod-ipt-nat6 found
IPv6 NAT mode
Warning: Option @zone[1].masq6 is unknown
Warning: Option @zone[1].masq6_privacy is unknown
Warning: Section @redirect[0] (ISOS) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[1] (NAS) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[2] (teslamate) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[3] (Emby115) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[4] (GBOX) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[5] (xiaoya) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[6] (DSM-SSH) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[7] (AX11000) does not specify a protocol, assuming TCP+UDP
Warning: Section @redirect[8] (iStoreOS-SSH) does not specify a protocol, assuming TCP+UDP
 * Clearing IPv4 filter table
 * Clearing IPv4 nat table
 * Clearing IPv4 mangle table
 * Clearing IPv4 raw table
 * Populating IPv4 filter table
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Rule 'Allow-IGMP'
   * Rule 'Allow-IPSec-ESP'
   * Rule 'Allow-ISAKMP'
   * Rule 'linkease'
   * Redirect 'ISOS'
   * Redirect 'NAS'
   * Redirect 'teslamate'
   * Redirect 'Emby115'
   * Redirect 'GBOX'
   * Redirect 'xiaoya'
   * Redirect 'DSM-SSH'
   * Redirect 'AX11000'
   * Redirect 'iStoreOS-SSH'
   * Forward 'lan' -> 'wan'
   * Forward 'docker' -> 'wan'
   * Forward 'docker' -> 'lan'
   * Forward 'lan' -> 'docker'
   * Zone 'lan'
   * Zone 'wan'
   * Zone 'docker'
 * Populating IPv4 nat table
   * Redirect 'ISOS'
   * Redirect 'NAS'
   * Redirect 'teslamate'
   * Redirect 'Emby115'
   * Redirect 'GBOX'
   * Redirect 'xiaoya'
   * Redirect 'DSM-SSH'
   * Redirect 'AX11000'
   * Redirect 'iStoreOS-SSH'
   * NAT 'DockerNAT'
   * Zone 'lan'
   * Zone 'wan'
Warning: wan will enable FULLCONE-NAT
   * Zone 'docker'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
   * Zone 'docker'
 * Populating IPv4 raw table
   * Zone 'lan'
     - Using automatic conntrack helper attachment
   * Zone 'wan'
   * Zone 'docker'
 * Clearing IPv6 filter table
 * Clearing IPv6 nat table
[... 113 lines cut ...]
   * Rule 'Allow-IPSec-ESP'
   * Rule 'Allow-ISAKMP'
   * Rule 'linkease'
   * Forward 'lan' -> 'wan'
   * Forward 'docker' -> 'wan'
   * Forward 'docker' -> 'lan'
   * Forward 'lan' -> 'docker'
   * Zone 'lan'
   * Zone 'wan'
   * Zone 'docker'
 * Populating IPv6 nat table
   * Zone 'lan'
   * Zone 'wan'
   * Zone 'docker'
 * Populating IPv6 mangle table
   * Zone 'lan'
   * Zone 'wan'
   * Zone 'docker'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/usr/share/miniupnpd/firewall.include'
Done

[image]

jjm2473, 2025-04-25:
注意这里提到的ipv6问题全是关于内网的，如果你的wan口本身就没法获取ipv6且没有ipv6-pd，那怎么配置也不会有效的。