istoreos/istoreos #1173: r5s  jellyfin 播放视频“该客户端与媒体不兼容，服务器未发送兼容的媒体格式”

y0un9y0n9, 2023-12-10:
### (1) 具体问题

A：
播放错误
该客户端与媒体不兼容，服务器未发送兼容的媒体格式。

### (2) 路由器型号和固件版本

A：
机型：r5s
系统版本：iStoreOS 22.03.5 2023120811
jellyfin版本：v1.1.8
### (3) 详细日志和/或截图

A：
[2023-12-10 22:19:29.209 +08:00] [ERR] [61] Jellyfin.Api.Helpers.TranscodingJobHelper: FFmpeg exited with code 1
[2023-12-10 22:19:29.267 +08:00] [ERR] [61] Jellyfin.Server.Middleware.ExceptionMiddleware: Error processing request. URL "GET" "/videos/702433d7-5098-79cb-ebd6-efd01d3b4276/live.m3u8".
MediaBrowser.Common.FfmpegException: FFmpeg exited with code 1
   at Jellyfin.Api.Helpers.TranscodingJobHelper.StartFfMpeg(StreamState state, String outputPath, String commandLineArguments, HttpRequest request, TranscodingJobType transcodingJobType, CancellationTokenSource cancellationTokenSource, String workingDirectory)
   at Jellyfin.Api.Controllers.DynamicHlsController.GetLiveHlsStream(Guid itemId, String container, Nullable`1 static, String params, String tag, String deviceProfileId, String playSessionId, String segmentContainer, Nullable`1 segmentLength, Nullable`1 minSegments, String mediaSourceId, String deviceId, String audioCodec, Nullable`1 enableAutoStreamCopy, Nullable`1 allowVideoStreamCopy, Nullable`1 allowAudioStreamCopy, Nullable`1 breakOnNonKeyFrames, Nullable`1 audioSampleRate, Nullable`1 maxAudioBitDepth, Nullable`1 audioBitRate, Nullable`1 audioChannels, Nullable`1 maxAudioChannels, String profile, String level, Nullable`1 framerate, Nullable`1 maxFramerate, Nullable`1 copyTimestamps, Nullable`1 startTimeTicks, Nullable`1 width, Nullable`1 height, Nullable`1 videoBitRate, Nullable`1 subtitleStreamIndex, Nullable`1 subtitleMethod, Nullable`1 maxRefFrames, Nullable`1 maxVideoBitDepth, Nullable`1 requireAvc, Nullable`1 deInterlace, Nullable`1 requireNonAnamorphic, Nullable`1 transcodingMaxAudioChannels, Nullable`1 cpuCoreLimit, String liveStreamId, Nullable`1 enableMpegtsM2TsMode, String videoCodec, String subtitleCodec, String transcodeReasons, Nullable`1 audioStreamIndex, Nullable`1 videoStreamIndex, Nullable`1 context, Dictionary`2 streamOptions, Nullable`1 maxWidth, Nullable`1 maxHeight, Nullable`1 enableSubtitlesInManifest)
   at lambda_method1286(Closure , Object )
   at Microsoft.AspNetCore.Mvc.Infrastructure.ActionMethodExecutor.TaskOfActionResultExecutor.Execute(IActionResultTypeMapper mapper, ObjectMethodExecutor executor, Object controller, Object[] arguments)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeActionMethodAsync>g__Awaited|12_0(ControllerActionInvoker invoker, ValueTask`1 actionResultValueTask)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeNextActionFilterAsync>g__Awaited|10_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Rethrow(ActionExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeInnerFilterAsync>g__Awaited|13_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResourceFilter>g__Awaited|25_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResourceExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeFilterPipelineAsync>g__Awaited|20_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Microsoft.AspNetCore.Routing.EndpointMiddleware.<Invoke>g__AwaitRequestTask|6_0(Endpoint endpoint, Task requestTask, ILogger logger)
   at Jellyfin.Server.Middleware.ServerStartupMessageMiddleware.Invoke(HttpContext httpContext, IServerApplicationHost serverApplicationHost, ILocalizationManager localizationManager)
   at Jellyfin.Server.Middleware.WebSocketHandlerMiddleware.Invoke(HttpContext httpContext, IWebSocketManager webSocketManager)
   at Jellyfin.Server.Middleware.IpBasedAccessValidationMiddleware.Invoke(HttpContext httpContext, INetworkManager networkManager)
   at Jellyfin.Server.Middleware.LanFilteringMiddleware.Invoke(HttpContext httpContext, INetworkManager networkManager, IServerConfigurationManager serverConfigurationManager)
   at Microsoft.AspNetCore.Authorization.Policy.AuthorizationMiddlewareResultHandler.HandleAsync(RequestDelegate next, HttpContext context, AuthorizationPolicy policy, PolicyAuthorizationResult authorizeResult)
   at Microsoft.AspNetCore.Authorization.AuthorizationMiddleware.Invoke(HttpContext context)
   at Jellyfin.Server.Middleware.QueryStringDecodingMiddleware.Invoke(HttpContext httpContext)
   at Swashbuckle.AspNetCore.ReDoc.ReDocMiddleware.Invoke(HttpContext httpContext)
   at Swashbuckle.AspNetCore.SwaggerUI.SwaggerUIMiddleware.Invoke(HttpContext httpContext)
   at Swashbuckle.AspNetCore.Swagger.SwaggerMiddleware.Invoke(HttpContext httpContext, ISwaggerProvider swaggerProvider)
   at Microsoft.AspNetCore.Authentication.AuthenticationMiddleware.Invoke(HttpContext context)
   at Jellyfin.Server.Middleware.RobotsRedirectionMiddleware.Invoke(HttpContext httpContext)
   at Jellyfin.Server.Middleware.LegacyEmbyRouteRewriteMiddleware.Invoke(HttpContext httpContext)
   at Microsoft.AspNetCore.ResponseCompression.ResponseCompressionMiddleware.InvokeCore(HttpContext context)
   at Jellyfin.Server.Middleware.ResponseTimeMiddleware.Invoke(HttpContext context, IServerConfigurationManager serverConfigurationManager)
   at Jellyfin.Server.Middleware.ExceptionMiddleware.Invoke(HttpContext context)

y0un9y0n9, 2023-12-10:
[2023-12-10 22:34:08.650 +08:00] [INF] [94] Jellyfin.Api.Helpers.TranscodingJobHelper: "/usr/bin/ffmpeg" "-analyzeduration 200M -f matroska -autorotate 0 -i file:\"/mnt/mmc2-4/alist/视频/IMDbTOP001-TOP250/七宗罪.Se7en.1995.BluRay.1080p.x265.10bit.2Audio.MNHD-FRDS.mkv\" -map_metadata -1 -map_chapters -1 -threads 0 -sn   -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type mpegts -start_number 0 -hls_base_url \"hls/59632b730a69a3725af7b1964c014acc/\" -hls_segment_filename \"/config/transcodes/59632b730a69a3725af7b1964c014acc%d.ts\" -hls_playlist_type event -hls_list_size 0 -y \"/config/transcodes/59632b730a69a3725af7b1964c014acc.m3u8\""
[2023-12-10 22:34:08.747 +08:00] [ERR] [94] Jellyfin.Api.Helpers.TranscodingJobHelper: FFmpeg exited with code 1
[2023-12-10 22:34:08.789 +08:00] [ERR] [94] Jellyfin.Server.Middleware.ExceptionMiddleware: Error processing request. URL "GET" "/videos/cc3a6182-c917-5d98-d386-d564849a1a93/live.m3u8".
MediaBrowser.Common.FfmpegException: FFmpeg exited with code 1
   at Jellyfin.Api.Helpers.TranscodingJobHelper.StartFfMpeg(StreamState state, String outputPath, String commandLineArguments, HttpRequest request, TranscodingJobType transcodingJobType, CancellationTokenSource cancellationTokenSource, String workingDirectory)
   at Jellyfin.Api.Controllers.DynamicHlsController.GetLiveHlsStream(Guid itemId, String container, Nullable`1 static, String params, String tag, String deviceProfileId, String playSessionId, String segmentContainer, Nullable`1 segmentLength, Nullable`1 minSegments, String mediaSourceId, String deviceId, String audioCodec, Nullable`1 enableAutoStreamCopy, Nullable`1 allowVideoStreamCopy, Nullable`1 allowAudioStreamCopy, Nullable`1 breakOnNonKeyFrames, Nullable`1 audioSampleRate, Nullable`1 maxAudioBitDepth, Nullable`1 audioBitRate, Nullable`1 audioChannels, Nullable`1 maxAudioChannels, String profile, String level, Nullable`1 framerate, Nullable`1 maxFramerate, Nullable`1 copyTimestamps, Nullable`1 startTimeTicks, Nullable`1 width, Nullable`1 height, Nullable`1 videoBitRate, Nullable`1 subtitleStreamIndex, Nullable`1 subtitleMethod, Nullable`1 maxRefFrames, Nullable`1 maxVideoBitDepth, Nullable`1 requireAvc, Nullable`1 deInterlace, Nullable`1 requireNonAnamorphic, Nullable`1 transcodingMaxAudioChannels, Nullable`1 cpuCoreLimit, String liveStreamId, Nullable`1 enableMpegtsM2TsMode, String videoCodec, String subtitleCodec, String transcodeReasons, Nullable`1 audioStreamIndex, Nullable`1 videoStreamIndex, Nullable`1 context, Dictionary`2 streamOptions, Nullable`1 maxWidth, Nullable`1 maxHeight, Nullable`1 enableSubtitlesInManifest)
   at lambda_method1286(Closure , Object )
   at Microsoft.AspNetCore.Mvc.Infrastructure.ActionMethodExecutor.TaskOfActionResultExecutor.Execute(IActionResultTypeMapper mapper, ObjectMethodExecutor executor, Object controller, Object[] arguments)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeActionMethodAsync>g__Awaited|12_0(ControllerActionInvoker invoker, ValueTask`1 actionResultValueTask)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeNextActionFilterAsync>g__Awaited|10_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Rethrow(ActionExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeInnerFilterAsync>g__Awaited|13_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResourceFilter>g__Awaited|25_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResourceExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeFilterPipelineAsync>g__Awaited|20_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Microsoft.AspNetCore.Routing.EndpointMiddleware.<Invoke>g__AwaitRequestTask|6_0(Endpoint endpoint, Task requestTask, ILogger logger)
   at Jellyfin.Server.Middleware.ServerStartupMessageMiddleware.Invoke(HttpContext httpContext, IServerApplicationHost serverApplicationHost, ILocalizationManager localizationManager)
   at Jellyfin.Server.Middleware.WebSocketHandlerMiddleware.Invoke(HttpContext httpContext, IWebSocketManager webSocketManager)
   at Jellyfin.Server.Middleware.IpBasedAccessValidationMiddleware.Invoke(HttpContext httpContext, INetworkManager networkManager)
   at Jellyfin.Server.Middleware.LanFilteringMiddleware.Invoke(HttpContext httpContext, INetworkManager networkManager, IServerConfigurationManager serverConfigurationManager)
[... cut ...]
   at Jellyfin.Server.Middleware.RobotsRedirectionMiddleware.Invoke(HttpContext httpContext)
   at Jellyfin.Server.Middleware.LegacyEmbyRouteRewriteMiddleware.Invoke(HttpContext httpContext)
   at Microsoft.AspNetCore.ResponseCompression.ResponseCompressionMiddleware.InvokeCore(HttpContext context)
   at Jellyfin.Server.Middleware.ResponseTimeMiddleware.Invoke(HttpContext context, IServerConfigurationManager serverConfigurationManager)
   at Jellyfin.Server.Middleware.ExceptionMiddleware.Invoke(HttpContext context)

y0un9y0n9, 2023-12-11:
/videos/f20275f9-d8d7-83ab-1358-448763c5a1a4/live.m3u8

{"Protocol":0,"Id":"f20275f9d8d783ab1358448763c5a1a4","Path":"/mnt/mmc2-4/alist/\u89C6\u9891/IMDbTOP001-TOP250/\u5B89\u5FB7\u70C8\u5362\u5E03\u5ED6\u592B.Andrei.Rublev.1966.BluRay.1080p.x265.10bit.MNHD-FRDS.mkv","EncoderPath":null,"EncoderProtocol":null,"Type":0,"Container":"mkv","Size":null,"Name":"\u5B89\u5FB7\u70C8\u5362\u5E03\u5ED6\u592B.Andrei.Rublev.1966.BluRay.1080p.x265.10bit.MNHD-FRDS","IsRemote":false,"ETag":"49815f0961335822b66178bbfe712910","RunTimeTicks":null,"ReadAtNativeFramerate":false,"IgnoreDts":false,"IgnoreIndex":false,"GenPtsInput":false,"SupportsTranscoding":true,"SupportsDirectStream":true,"SupportsDirectPlay":true,"IsInfiniteStream":false,"RequiresOpening":false,"OpenToken":null,"RequiresClosing":false,"LiveStreamId":null,"BufferMs":null,"RequiresLooping":false,"SupportsProbing":true,"VideoType":0,"IsoType":null,"Video3DFormat":null,"MediaStreams":[],"MediaAttachments":[],"Formats":[],"Bitrate":null,"Timestamp":null,"RequiredHttpHeaders":{},"TranscodingUrl":null,"TranscodingSubProtocol":null,"TranscodingContainer":null,"AnalyzeDurationMs":null,"DefaultAudioStreamIndex":null,"DefaultSubtitleStreamIndex":null}

/usr/bin/ffmpeg -analyzeduration 200M  -i file:"/mnt/mmc2-4/alist/视频/IMDbTOP001-TOP250/安德烈卢布廖夫.Andrei.Rublev.1966.BluRay.1080p.x265.10bit.MNHD-FRDS.mkv" -map_metadata -1 -map_chapters -1 -threads 0 -sn   -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type mpegts -start_number 0 -hls_base_url "hls/8147d08e201449c20b7f7ecd39987d0b/" -hls_segment_filename "/config/transcodes/8147d08e201449c20b7f7ecd39987d0b%d.ts" -hls_playlist_type event -hls_list_size 0 -y "/config/transcodes/8147d08e201449c20b7f7ecd39987d0b.m3u8"

ffmpeg "-analyzeduration" "10000000" "-probesize" "10000000" "-i" "file:/mnt/mmc2-4/alist/视频/IMDbTOP001-TOP250/安德烈卢布廖夫.Andrei.Rublev.1966.BluRay.1080p.x265.10bit.MNHD-FRDS.mkv" "-map_metadata" "-1" "-map_chapters" "-1" "-threads" "0" "-sn" "-copyts" "-avoid_negative_ts" "disabled" "-max_muxing_queue_size" "2048" "-f" "hls" "-max_delay" "5000000" "-hls_time" "3" "-hls_segment_type" "mpegts" "-start_number" "0" "-hls_base_url" "hls/8147d08e201449c20b7f7ecd39987d0b/" "-hls_segment_filename" "/config/transcodes/8147d08e201449c20b7f7ecd39987d0b%d.ts" "-hls_playlist_type" "event" "-hls_list_size" "0" "-y" "/config/transcodes/8147d08e201449c20b7f7ecd39987d0b.m3u8"
ffmpeg version git-2022-06-18-83b0e20 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 8.3.0 (OpenWrt GCC 8.3.0 r0-83b0e20)
  configuration: --enable-cross-compile --cross-prefix=aarch64-openwrt-linux-gnu- --arch=aarch64 --cpu=cortex-a53 --target-os=linux --prefix=/usr --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-runtime-cpudetect --enable-lto --enable-neon --enable-vfp --enable-gnutls --disable-encoder=svq1 --enable-libopus --enable-hardcoded-tables --enable-libmp3lame --enable-gpl --enable-libx264 --enable-version3 --enable-nonfree --enable-libdrm --enable-librga --enable-rkmpp --enable-libfdk-aac
  libavutil      57. 33.101 / 57. 33.101
  libavcodec     59. 42.101 / 59. 42.101
  libavformat    59. 30.100 / 59. 30.100
  libavdevice    59.  8.101 / 59.  8.101
  libavfilter     8. 46.103 /  8. 46.103
  libswscale      6.  8.102 /  6.  8.102
  libswresample   4.  8.100 /  4.  8.100
  libpostproc    56.  7.100 / 56.  7.100
file:/mnt/mmc2-4/alist/视频/IMDbTOP001-TOP250/安德烈卢布廖夫.Andrei.Rublev.1966.BluRay.1080p.x265.10bit.MNHD-FRDS.mkv: Input/output error

closed 2023-12-14 by y0un9y0n9

Pradur241, 2023-12-30:
大佬这问题咋解决的？