istoreos/istoreos #1174: istoreos可以给rtl8811cu 添加驱动吗？open wrt 可发射WiFi

Gifl, 2023-12-10:


Gifl, 2023-12-10:
USB接口

jjm2473, 2023-12-15:
驱动名称是什么，现在已经有rtl8812au，可能是同一个驱动？

lfcz, 2024-02-29:
> 驱动名称是什么，现在已经有rtl8812au，可能是同一个驱动？

VID:0BDA，PID:C811，RTL8821CU和RTL8811CU都是一样的，共用驱动，目前现有系统已支持，但是我用H28K搭配cf-811c（8811CU）测试表现为可以正常识别和打开wifi，但是电脑（win10 22h2 ax201）无法连接wifi，手机（红米note12t miui14, pixiel 5 原生安卓12）是连接后十秒钟就掉线重连，三五次之后就搜不到无线信号了。考虑是原来的usb网卡可能存在问题，又购买了几个别的网卡，几天后在进行测试

jjm2473, 2024-02-29:
usb网卡不稳定很正常，不少机器都有这问题，能不用usb就别用

lfcz, 2024-02-29:
> usb网卡不稳定很正常，不少机器都有这问题，能不用usb就别用

想用来在外面偶尔应急使用，如果别的usb无线网卡也不能稳定运行的话，考虑换r5c了

lfcz, 2024-03-03:
> usb网卡不稳定很正常，不少机器都有这问题，能不用usb就别用

网卡到了。经过半天的测试，新版的CF-811ac依旧是存在不稳定的情况，而基于RTL8812AU的NEC WL900U可以完美且稳定驱动。

chenhaoo2020, 2024-03-24:
我也使用R5C的iStoreOS 22.03.6 2024031514固件尝试了一下RTL8811CU，发现AP模式完全没办法使用，我的手机和电脑完全无法看到无线网卡，使用lsusb -tv结果是
`root@iStoreOS:~# lsusb -tv
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
    |__ Port 1: Dev 5, If 0, Class=, Driver=rtw_8821cu, 480M
        ID 0bda:c811  
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  `

lfcz, 2024-03-24:
> 我也使用R5C的iStoreOS 22.03.6 2024031514固件尝试了一下RTL8811CU，发现AP模式完全没办法使用，我的手机和电脑完全无法看到无线网卡，使用lsusb -tv结果是 `root@iStoreOS:~# lsusb -tv /: Bus 08.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M [...]

应该是rtl8811cu本身集成度高，对外围电路虽然要求不高，但是稳定性比较差，造成了各种奇怪问题，如果想用usb网卡的话还是推荐rtl8812au，很稳定，就是体积稍大一些

closed 2024-04-07 by jjm2473

jjm2473, 2025-09-28:
如果openwrt支持的话，可以直接在软件包安装驱动