istoreos/istoreos #1175: 编译出错，求助

hx801217, 2023-12-10:
cripts/Kconfig.include:39: compiler 'aarch64-openwrt-linux-musl-gcc' not found
make[6]: *** [scripts/kconfig/Makefile:71: syncconfig] Error 1
make[5]: *** [Makefile:613: syncconfig] Error 2
Makefile:708: include/config/auto.conf.cmd: No such file or directory
make[4]: *** [Makefile:731: include/config/auto.conf.cmd] Error 2
make[4]: Leaving directory '/home/hx801217/桌面/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176'
make[3]: *** [Makefile:23: /home/hx801217/桌面/istoreos/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.176/.modules] Error 2
make[3]: Leaving directory '/home/hx801217/桌面/istoreos/target/linux/rockchip'
make[2]: *** [Makefile:11: compile] Error 2
make[2]: Leaving directory '/home/hx801217/桌面/istoreos/target/linux'
time: target/linux/compile#10.02#12.79#19.21
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:30: target/linux/compile] Error 1
make[1]: Leaving directory '/home/hx801217/桌面/istoreos'
make: *** [/home/hx801217/桌面/istoreos/include/toplevel.mk:234：target/linux/compile] 错误 2

ks-bao, 2023-12-13:
'aarch64-openwrt-linux-musl-gcc' not found 这个都没得，这个工具链都没看到~~~
放几天没人帮忙我们就关闭了哈

jjm2473, 2023-12-15:
靠，不要在桌面上编译。在一个没有中文，没有空格的目录下编译OK？

closed 2023-12-15 by jjm2473