istoreos/istoreos #1177: APC SPRM1K，使用rs232和串口转USB进行通讯，在 iStoreOS 上面无法识别为 ttyUSB0

ks-bao, 2023-12-13:
有研究UPS吗？我有一台APC SPRM1K，使用rs232和串口转USB进行通讯，之前在TrueNAS上用USB连接，通过xr_serial驱动识别为ttyUSB0，UPS服务使用apcsmart驱动成功连接并正常工作，然后我组了一台万兆软路由，本来打算直接安装iStoreOS物理机并通过软路由管理局域网设备的nut_client，但是进入系统后，发现UPS被kmod-usb-acm驱动识别为ttyACM0，按照正常配置nut_server的方法会产生大量反复报错，基本如下：
Tue Dec 12 10:43:09 2023 daemon.err upsmon【17847】:  730.070153	Poll UPS 【ups@localhost】 failed - Driver not connected
Tue Dec 12 10:43:14 2023 daemon.err odhcpd【10352】: Failed to send to ff02::1%lan@br-lan (Address not available)
Tue Dec 12 10:43:14 2023 daemon.err upsmon【17847】:  735.070642	Poll UPS 【ups@localhost】 failed - Driver not connected

卸载掉这个驱动后，会由kmod-usb-serial-xr_usb_serial_common驱动识别为ttyXRUSB0，修改配置后还是一样的报错，随后我看知乎上有人用unraid连接成功了，于是我也安装unraid尝试了一下，结果unraid也一样，使用cdc_acm识别为ttyACM0，无论是自带的UPS子程序还是社区的nut插件都无法连接UPS，系统日志里大概说的是在ttyACM0端口上没有检测到apcsmart协议的设备，让我重试，然而不管重试几次都是一样的[藏狐]

然后我不信邪的在软路由上尝试安装TrueNAS Scale，结果........成了，就用普通的方法就连接上了，我又拿另一台安装了unraid的NUC8i5BEH测试了一下，和我在万兆软路由设备上的unraid情况一模一样[辣眼睛]

ks-bao, 2023-12-13:
来自我咨询的回复：
[image]

ks-bao, 2023-12-13:
留几天没响应就关闭 issue 哈。

AUKcl, 2023-12-13:
这个USB转串口的模块不是我单独配的，它是在UPS内部就完成了转换，我也不清楚用的哪个芯片，在京东上的详情页也注明了USB和RS232不能同时使用
[image]

UPS具体信息可以查看[官方用户指南](https://download.schneider-electric.com/files?p_enDocType=User+guide&p_File_Name=AHUG-AYV8V3_R0_CH.pdf&p_Doc_Ref=SPD_AHUG-AYV8V3_CH)

我使用包装自带的USB-B to USB-A线连接主板

AUKcl, 2023-12-13:
NUT是在软路由上搭建的server端，方便通知局域网内的其他服务器，UPS本身并没有控制连接的设备，只是通过USB通讯接口让软路由进行状态监测而已，用法和常见的APC BK650是一样的

jjm2473, 2023-12-15:
试试RS232转USB

AUKcl, 2023-12-17:
我买了一个主板DB9 to RS232的连接线，安装到COM2上之后修改配置文件端口号为ttyS1和ttyS0均无效，新的CH340转接线在路上，等到货后再拿来试试

另外，我刚刚注意到，此款UPS使用的芯片为[XR21V1410](https://www.maxlinear.com/product/interface/uarts/usb-uarts/xr21v1410)
[image]

在Unraid论坛上也有人的UPS使用了接近的芯片，具体可看这篇帖子：[REQUEST TO ADD XR_SERIAL DRIVERS TO SUPPORT MORE UPS.](https://forums.unraid.net/topic/147253-request-to-add-xr_serial-drivers-to-support-more-ups/#comment-1321771)，看来问题主要在xr_serial驱动支持上面，不知道这个驱动和kmod-usb-serial-xr_usb_serial_common有什么区别，在iStoreOS上UPS使用kmod-usb-serial-xr_usb_serial_common同样无法工作，但软件源里面就没有其他和xr_serial相似的驱动了

我先试试在Unraid上安装xr_serial驱动，如果成功了再来反馈

AUKcl, 2023-12-17:
[HUAWEI_UPS2000(8)](https://networkupstools.org/docs/man/huawei-ups2000.html)这篇文章中，华为这款UPS和我同款芯片，问题表现也完全一致，可以确定就是xr_serial驱动的问题

跟着[Github](https://github.com/fanhuanji/rtl8125-driver-for-unraid-6.8.2/blob/main/compile_drivers.md)这篇文章编译了xr_serial.ko驱动

* Unraid 版本：6.12.6
* Linux内核版本：6.1.64

结果无法加载，NG

又查了一下openwrt的xr_serial驱动，几乎没有资料，只有等大佬开发驱动了

jjm2473, 2023-12-18:
还不如搞个RS232转USB，至少有很多方案可选，不用限制在xr_serial上

AUKcl, 2023-12-18:
是这样的，只是想着买线要多等几天，那在这期间软件方面的问题尽量尝试解决，就当研究技术了，成功的话也可以多个选择

不过可惜还是失败了🤣

明天转接线就到了，这次应该不会有问题了

AUKcl, 2023-12-19:
买的USB to RS232(CH340)线缆终于到了，即插即用，直接把配置改为/dev/ttyUSB0就正常通讯了，日志如下：

```shell
Dec 19 15:30:19 UNRAID02 kernel: usb 3-9: new full-speed USB device number 6 using xhci_hcd
Dec 19 15:30:19 UNRAID02 usb_manager: Info: rc.usb_manager usb_add 1a86_USB2.0-Ser_ /dev/bus/usb/003/006 003 006
Dec 19 15:30:19 UNRAID02 usb_manager: Info: rc.usb_manager Autoconnect No Mapping found 1a86_USB2.0-Ser_ /dev/bus/usb/003/006 003 006 port 3-9
Dec 19 15:30:19 UNRAID02 kernel: usbcore: registered new interface driver ch341
Dec 19 15:30:19 UNRAID02 kernel: usbserial: USB Serial support registered for ch341-uart
Dec 19 15:30:19 UNRAID02 kernel: ch341 3-9:1.0: ch341-uart converter detected
Dec 19 15:30:19 UNRAID02 kernel: ch341-uart ttyUSB0: break control not supported, using simulated break
Dec 19 15:30:19 UNRAID02 kernel: usb 3-9: ch341-uart converter now attached to ttyUSB0
Dec 19 15:30:19 UNRAID02 usb_manager: Info: rc.usb_manager tty_add 1a86_USB2.0-Ser_ /dev/ttyUSB0  
Dec 19 15:30:19 UNRAID02 usb_manager: Info: rc.usb_manager Processing tty attach 1a86_USB2.0-Ser_ /dev/ttyUSB0  
Dec 19 15:30:19 UNRAID02 usb_manager: Info: rc.usb_manager Autoconnect No Mapping found 1a86_USB2.0-Ser_ /dev/ttyUSB0   port 
```