istoreos/istoreos #1181: 如何安装lrzsz

jiaxinonly, 2023-12-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法安装lrzsz包，第一次使用istoreOS，之前用其他版本的openwrt都可以opkg install lrzsz安装，在istoreOS上不行，也更新了软件包列表，手动下载ipk文件，没有找到22.03.5版本的x86包，我需要连接openwrt传送文件

### (2) 路由器型号和固件版本

A：畅网N100  iStoreOS 22.03.5, 2023112413

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-12-15:
lszrz好像是串口的吧，传文件用ssh应该更快啊

jiaxinonly, 2023-12-15:
> 你说的“其他版本的openwrt”是openwrt 22.03.x吗？如果是，那应该一样可以安装。

可能不是，是eSir大佬的高大全版本，很久没更新了，因为没有n100的核显驱动，重新安装了最新版本的istoreos

jiaxinonly, 2023-12-15:
> lszrz好像是串口的吧，传文件用ssh应该更快啊

这是个linux的本地和远程传送文件的工具，比ssh更加方便

jiaxinonly, 2023-12-15:
如果我安装不了，使用其他任人编译的openwrt 22.03.x版本的openwrt都会无法安装对吗？我需要等待适配的包更新出来？

jjm2473, 2023-12-15:
靠，你写错包名了啊，是lrzsz，不是lszrz。

ssh的速度肯定比这个快，毕竟局域网网络怎么都能达到千兆，串口一般1M以内

jjm2473, 2023-12-15:
openwrt实际上有这个软件包，但是好像没索引，而且国内的镜像仓库也没同步过来 
https://archive.openwrt.org/releases/22.03.5/packages/x86_64/packages/lrzsz_0.12.21-1_x86_64.ipk

jiaxinonly, 2023-12-15:
> 靠，你写错包名了啊，是lrzsz，不是lszrz。
> 
> ssh的速度肯定比这个快，毕竟局域网网络怎么都能达到千兆，串口一般1M以内

截图上确实是打错了，但是lrzsz我也试过，晚上我回去再试试，我了解ssh是使用scp命令传送文件，手敲比较麻烦，lrzsz很方便，当然也可以使用ssh客户端的传输功能，另外nohup也安装不了，这个对我比较重要 :cry:

jjm2473, 2023-12-15:
你不是使用一些gui软件吗，应该支持scp协议。或者下载个filezilla作为文件管理工具。

jjm2473, 2023-12-15:
nohup的包名是“coreutils-nohup”

jiaxinonly, 2023-12-15:
> openwrt实际上有这个软件包，但是好像没索引，而且国内的镜像仓库也没同步过来 https://archive.openwrt.org/releases/22.03.5/packages/x86_64/packages/lrzsz_0.12.21-1_x86_64.ipk
> 
> 下载以后在istore的手段安装那边安装下

非常感谢 :+1:，我找错了地方，我一直在routing下找https://archive.openwrt.org/releases/22.03.5/packages/x86_64/routing/ 怪不得找不到

closed 2023-12-15 by jiaxinonly