istoreos/istoreos #1183: X86硬件 20230609之后的固件，Docker版的nastool无法联通telegram，需要代理的IP无法联网

zzj6116, 2023-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：20230609之后的固件使用Docker版本nastool 2.91版的telegram通知功能，一直都能正常使用，后来全新刷机后，安装同样版本的nastool，telegram通知功能无法联网。检测网络连通性，显示api.telegram.org无法连通，Docker网络使用的都是默认的设置，一直以为是nastool的问题，前几天在绿联DX4600的Docker上使用nastool没有这个问题，可正常连通api.telegram.org，win版本的nastool同样可以正常连通api.telegram.org，所以判断是istoreos的问题，盼官方能解决这个问题。

### (2) 路由器型号和固件版本

A：硬件是J4125的x86机器，固件版本是iStoreOS 22.03.5 2023120811

### (3) 详细日志和/或截图

[A：
[image]
[image]
[image]
[image]
[image]

jjm2473, 2023-12-15:
你的翻墙插件运行在哪，跟nastool同一机器吗？翻墙插件配置本机代理模式是什么？还有执行`ip addr ; route`把结果发出来看看。

zzj6116, 2023-12-15:
> ip addr ; route

感谢您的回复，翻墙插件用的是Are you ok 下载的新版的ShadowSocksR Plus+ ，运行在iStoreOS物理机上，和Nastool在同一机器，iStoreOS是主路由，此路由下，所有的设备翻墙都是正常的，只有iStoreOS内安装的Docker版的nastool翻墙不正常，同样的网段下，绿联dokcer安装的nastool翻墙正常，执行命令行显示如下：
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         hn.kd.ny.adsl   0.0.0.0         UG    0      0        0 pppoe-wan
10.0.0.0        *               255.255.255.0   U     0      0        0 br-lan
42.228.239.254  *               255.255.255.255 UH    0      0        0 pppoe-wan
172.17.0.0      *               255.255.0.0     U     0      0        0 docker0
192.168.191.0   *               255.255.255.0   U     0      0        0 zteb4nen5y
[image]
[image]
[image]

jjm2473, 2023-12-15:
nastools是自己手动部署的吗，还是用istore的插件部署的？手动部署有没有设置dns？

zzj6116, 2023-12-15:
> nastools是自己手动部署的吗，还是用istore的插件部署的？手动部署有没有设置dns？
nastools是用istore的插件部署的，但是镜像因为拉不到官方镜像了，使用的的第三方备份的官方2.91镜像，后来也试着升级到官方的最后一版nastools 3.2的镜像，但是问题依旧，还是显示显示api.telegram.org无法连通。相同的镜像，在绿联的docker设备上没有这个问题。
[image]
[image]

jjm2473, 2023-12-15:
因为绿联是在内网啊，翻墙插件又不是在绿联上运行。
翻墙插件的本机和内网代理策略是不一样的。

你试试istoreos里面执行`curl https://api.telegram.org/`

jjm2473, 2023-12-15:
哦，我还想到一个问题，你的翻墙插件配置的代码模式是redirect还是tproxy，也许换一个就行了

jjm2473, 2023-12-15:
我在其他机器上遇到过tproxy无法代理docker流量的情况，改成redirect就行了

zzj6116, 2023-12-15:
您好，请问这个怎么改？

jjm2473, 2023-12-15:
passwall的话，高级设置里。其他插件就不知道了

zzj6116, 2023-12-15:
感谢，果然可以了，那看来是插件的原因。

jjm2473, 2023-12-15:
不一定是插件问题，可能内核tproxy就是不支持docker虚拟网络。好像passwall默认的tcp代理模式是redirect。

zzj6116, 2023-12-15:
passwall默认的tcp代理模式就是redirect
 
ShadowSocksR Plus+没有设置redirect的选项，

目前更换插件为passwall2，问题解决了

closed 2024-02-02 by jjm2473