istoreos/istoreos #1184: 不能识别网卡Intel(R) Dual Band Wireless-AC 8265和Realtek USB GbE Family Controller

masx200, 2023-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
不能识别网卡Intel(R) Dual Band Wireless-AC 8265和Realtek USB GbE Family Controller
### (2) 路由器型号和固件版本

A：
处理器 Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz   1.99 GHz
固件 [istoreos-22.03.5-2023120811-x86-64-squashfs-combined-efi.img.gz](https://fw0.koolcenter.com/iStoreOS/x86_64_efi/istoreos-22.03.5-2023120811-x86-64-squashfs-combined-efi.img.gz)
### (3) 详细日志和/或截图

A：

masx200, 2023-12-15:
[image][image][image]

jjm2473, 2023-12-15:
执行`lspci -nnk`看看具体是什么设备

masx200, 2023-12-15:
> 执行`lspci -nnk`看看具体是什么设备

```
02:00.0 Network controller [0280]: Intel Corporation Wireless 8265 / 8275 [8086:24fd] (rev 78)
	Subsystem: Intel Corporation Dual Band Wireless-AC 8265 [8086:0010]
	Kernel driver in use: iwlwifi
	Kernel modules: iwlwifi
```

masx200, 2023-12-15:
```

/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 2109:0817 VIA Labs, Inc. 
        |__ Port 3: Dev 3, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
            ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
```

jjm2473, 2023-12-15:
执行`dmesg | grep iwlwifi`看看报错

jjm2473, 2023-12-15:
USB网卡肯定已经识别到了，可能你没加到逻辑网络里，所以没使用罢了

jjm2473, 2023-12-15:
`ls -l /sys/class/net/ | grep eth`看看

masx200, 2023-12-15:
> `ls -l /sys/class/net/ | grep eth`看看
这个是我在openkylin系统里面看到的驱动信息
```
ethtool -i wlp2s0
driver: iwlwifi
version: 6.1.0-3-generic
firmware-version: 36.ca7b901d.0 8265-36.ucode
expansion-rom-version: 
bus-info: 0000:02:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```
```
ethtool -i enx00e04c683554
driver: r8152
version: v1.12.13
firmware-version: rtl8153b-2 v1 10/23/19
expansion-rom-version: 
bus-info: usb-0000:3b:00.0-2.3
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

jjm2473, 2023-12-15:
在istoreos里面执行看看，r8152驱动肯定有，iwlwifi也有，但是可能需要fw文件

masx200, 2023-12-15:
在istoreos里面的错误信息日志找到了:

```
[   45.078784] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-36.ucode failed with error -2
[   45.079354] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-36.ucode
[  105.528817] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-35.ucode failed with error -2
[  105.528823] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-35.ucode
[  123.261994] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-34.ucode failed with error -2
[  123.261996] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-34.ucode
[  123.280904] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-33.ucode failed with error -2
[  123.280905] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-33.ucode
[  123.282220] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-32.ucode failed with error -2
[  123.282221] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-32.ucode
[  123.283568] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-31.ucode failed with error -2
[  123.283569] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-31.ucode
[  123.284904] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-30.ucode failed with error -2
[  123.284905] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-30.ucode
[  123.286257] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-29.ucode failed with error -2
[  123.286259] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-29.ucode
[  123.287592] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-28.ucode failed with error -2
[  123.287594] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-28.ucode
[  123.288993] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-27.ucode failed with error -2
[  123.288995] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-27.ucode
[  123.290315] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-26.ucode failed with error -2
[  123.290317] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-26.ucode
[  123.291747] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-25.ucode failed with error -2
[  123.291749] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-25.ucode
[  123.293210] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-24.ucode failed with error -2
[  123.293211] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-24.ucode
[  123.294697] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-23.ucode failed with error -2
[  123.294698] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-23.ucode
[  123.296117] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-8265-22.ucode failed with error -2
[  123.296118] iwlwifi 0000:02:00.0: Falling back to sysfs fallback for: iwlwifi-8265-22.ucode
[  123.297576] iwlwifi 0000:02:00.0: no suitable firmware found!
[  123.298263] iwlwifi 0000:02:00.0: minimum version required: iwlwifi-8265-22
[  123.298960] iwlwifi 0000:02:00.0: maximum version supported: iwlwifi-8265-36
[  123.299605] iwlwifi 0000:02:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```

jjm2473, 2023-12-16:
软件包那边点一下更新列表，然后找到iwlwifi-firmware-iwl8265安装应该就行了

masx200, 2023-12-17:
> 软件包那边点一下更新列表，然后找到iwlwifi-firmware-iwl8265安装应该就行了

安装驱动之后可以用2.4g频道，但是不能用5g频道，可以在下个版本把这个驱动添加到固件里了吗

jjm2473, 2023-12-18:
Intel的网卡都不支持5G AP模式，可能支持5G客户端。如果要5G AP模式，用MT7915/MT7916/MT7921/MT7922。

iwlwifi-firmware-iwl8265 可以集成到固件，只是Intel的网卡本来就没啥用。

closed 2023-12-18 by jjm2473