istoreos/istoreos #1185: iPad端safari浏览器访问webUI，1：“终端”无显示，2：一些下拉菜单显示不全

N1k1Zhang, 2023-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：iPad端safari浏览器访问webUI，1：“终端”无显示，2：一些下拉菜单显示不全

### (2) 路由器型号和固件版本

A：设备型号
Raspberry Pi 4 Model B Rev 1.4
固件版本
iStoreOS 22.03.5 2023120811

### (3) 详细日志和/或截图

A：
[image]
[image]

jjm2473, 2023-12-15:
换浏览器

closed 2023-12-15 by jjm2473

N1k1Zhang, 2023-12-15:
> 换浏览器

chrome浏览器也是一样的问题

jjm2473, 2023-12-16:
Chrome会出现终端字体黑色？你的chrome什么版本啊

N1k1Zhang, 2023-12-16:
> Chrome会出现终端字体黑色？你的chrome什么版本啊

[image]
一样是没有显示，版本是刚刚更新的，一样的问题，版本号如图。另外，我密码是比较长的随机密码，每次终端输入都很麻烦，能不能实现点击终端就自动登陆，我以前的固件就可以

jjm2473, 2023-12-18:
> 点击终端就自动登陆

这是不安全的，随便一个浏览器打开7681端口，不需要密码就能登录了。

jjm2473, 2023-12-18:
这个网页版的终端是TTYD，你可以到ttyd的项目反馈这个bug

N1k1Zhang, 2023-12-18:
好的谢谢 Sent from my iPhone