istoreos/istoreos #1190: 想尝试移植编译到香橙派OrangePi R1 Plus LTS上，希望能够得到帮助

KJZH001, 2023-12-17:
## 1.关于你要提交的问题
我之前在vmware和hyperv中用虚拟机的方式使用openwrt了很长一段时间，因为目前有了硬路由需求，所以搞了一块arm板子来

这块板子似乎并没有被openwrt官方和istore所支持，但是香橙派官方给出了能够编译且运行的固件和源码（基于openwrt原版分支）

https://github.com/orangepi-xunlong/openwrt

所以我想尝试学习编译，看看能否将istore移植过去

目前希望能够知道，openwrt支持什么型号的芯片，是由什么来确定的呢？如果我想把相关的驱动搬过去，要怎么做（或者说，有哪些地方能够找到有关可供学习参考的资料呢？）

我明白，尝试移植驱动可能不是那么容易，但是我确实想试一试，希望能够得到帮助，谢谢！

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：见上文

### (2) 路由器型号和固件版本

A：Orange Pi R1 Plus LTS
Rockchip RK3328
四核 ARM Cortex-A53 64 位处理器,主频最高达1.5GHz
Mali-450MP2
1GB LPDDR3（与 GPU 共享）

### (3) 详细日志和/或截图

A：None

jjm2473, 2023-12-18:
3328直接参考r2s，同一个cpu

KJZH001, 2023-12-18:
> 3328直接参考r2s，同一个cpu

感谢回复和解答，既然是同一个cpu的话是不是也意味着我可以直接尝试把r2s的镜像刷进去看看能不能运行呢？

jjm2473, 2023-12-18:
@KJZH001 如果你不担心烧坏板子，可以试试。

其实理论上只要替换dtb文件就行了。

KJZH001, 2023-12-18:
> @KJZH001 如果你不担心烧坏板子，可以试试。
> 
> 其实理论上只要替换dtb文件就行了。

请问您的意思是我只需要把`target/linux/rockchip`这个下面的文件从香橙派的分支搬到istore这边就能编译了吗？

不过我注意到这两边的结构差异有些大

https://github.com/orangepi-xunlong/openwrt/tree/master/target/linux/rockchip
[image]

https://github.com/istoreos/istoreos/tree/istoreos-22.03/target/linux/rockchip
[image]

同时在这个目录下面也没有dts相关的东西，我也去查看了lede的分支，似乎也不太一样
https://github.com/coolsnowwolf/lede/tree/master/target/linux/rockchip
[image]

这个问题可能有些蠢了，希望我没有让你感到冒犯

对于烧毁板子的风险，这个我确实不太想尝试，毕竟才刚买过来，希望就这么给玩炸了，而且对于这块完全没接触过的领域我也确实不知道风险有多大，在无法评估风险的情况下尝试这并不明智，也感谢你的提醒

KJZH001, 2023-12-19:
> 你还是找厂家适配吧，或者让厂家找我们合作，不过我觉得研究 RK3328 已经没有性价比了，不如 RK3528。

好吧，如果真的不行的话我也会考虑放弃的

不过我也在istore的自述文件中找到了这么一段话

```
iStoreOS 来源于官方的 OpenWRT Release 分支源码，都是通过 OpenWRT 标准组件形式实现，100% 兼容 OpenWRT 的源主干分支。我们修改或者开发的部分，都以插件形式，具体源代码如下：

[iStoreOS 固件](https://github.com/istoreos/istoreos) 核心固件源代码
[nas-packages-luci](https://github.com/linkease/nas-packages-luci) 我们自己开发的插件 UI 代码
[nas-packages](https://github.com/linkease/nas-packages) 我们自己开发插件的程序代码，部分程序并不开源
[istore](https://github.com/linkease/istore) 软件中心核心代码，包含备份插件功能等，全部开源
[istore-packages](https://github.com/linkease/istore-packages) 软件中心的一些非 OpenWRT 官方包
[openwrt-app-actions](https://github.com/linkease/openwrt-app-actions) 其他一些软件包
iStoreOS 就是在 OpenWRT 最基础最原始的固件基础上，加上了上面插件的能力来实现。
```
既然修改部分是用插件实现的，那么我把这些插件加在香橙派的源码上编译是否可行呢？兼容性的问题这里先暂不考虑，我也不一定需要把istore部分的所有插件编译进去

对于我来说其实比较想要的就是istore的主题和应用商店，前者是考虑给家人使用起来也方便一点，因为有个`小白路由版本`的主题，这个版本的主题我相信我爸妈他们也能轻松操作

我昨天已经用香橙派官方的镜像在未做额外修改的情况下顺利的跑完了一次编译作为测试，成功的得到了固件，至少香橙派给出的这套源码是确实能用的

关于dtb，我今天也找了很长时间的资料，中文互联网上对于这块资料确实比较稀缺，也可能是我找的方法不对，谷歌上也没有找到太好的资料

至少今天我确认了两件事情，对于这种轻量级设备，使用dts文件描述文件树告诉cpu有哪些外围设备需要加载，dtb是从dts编译得到的二进制文件，同时dtb文件大部分情况下也能从现有的linux镜像里拆出来（香橙派给出了编译好的Debian镜像，我认为可以直接从这里面拆或者从openwrt的分支固件里拆），或者在能进系统的情况下也能从/sys/firmware/devicetree/base直接读取出来

jjm2473, 2023-12-20:
如果只是需要主题和istore，那可以只安装插件。

查看对应项目的说明文档：
https://github.com/linkease/nas-packages-luci
https://github.com/linkease/istore

closed 2024-01-05 by jjm2473

LCW0NJUPT, 2024-02-16:
虽然这个issue已经close，但是还想问下楼主，我的设备和需求与您一致，您后来采用的什么方案呢？

KJZH001, 2024-02-19:
> 虽然这个issue已经close，但是还想问下楼主，我的设备和需求与您一致，您后来采用的什么方案呢？

我最终的选择是使用了其他的分支的固件，因为我偶然间发现[mmortalwrt](https://drive.moeworld.tech/zh-CN/%E7%B3%BB%E7%BB%9F%E9%95%9C%E5%83%8F/OpenWrt/Orange%20Pi%20R1%20Plus%20LTS-immortalwrt%20Official/)还在对这块板子做维护支持，并且更新到了主线最新的23版本

当然，我之前也尝试编译过一些镜像，仓库在这里https://github.com/KJZH001/Actions-OpenWrt

编译后的成品固件（包括把iStore商店编译进去的版本）和其他的相关玩意你也能在这里找到

https://drive.moeworld.tech/zh-CN/%E7%B3%BB%E7%BB%9F%E9%95%9C%E5%83%8F/OpenWrt/

**但是，我也没测试过我编译出的这些固件究竟能不能用，自行判断是否值得尝试使用，而且我很长一段时间也不打算去修这些东西了**

LCW0NJUPT, 2024-07-10:
几个月后重新回复您，经过反复对比，发现还是原版openwrt运行最稳定，发热也较小（正好到了夏天）；相比而言immortalwrt大而全，但镜像的测试并不足够，在我的设备上经常出现卡死的问题。
推荐您使用这个原版openwrt镜像，我现在手动安装了一些第三方插件，可以基本实现所需。
https://github.com/kgoerbig/openwrt-orangepi-r1-plus/releases/tag/v22.03.2

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/2560/1

lingyue103, 2026-09-26:
我在我的分支上已经实现了移植，可以看看
https://github.com/lingyue103/istoreos