istoreos/istoreos #1191: 静态路由不生效

jinxiao, 2023-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：静态路由不生效。在界面，网络-路由，配置静态路由和IPV4规则，使用命令route查看未生效

### (2) 路由器型号和固件版本

A：x86_64 iStoreOS 22.03.5 2023112413 / LuCI istoreos-22.03 branch git-23.271.10849-b445bc4

### (3) 详细日志和/或截图

A：
[image]
使用route查看，不生效

[image]

jjm2473, 2023-12-22:
为啥你要配置规则，而且还是main表，应该只要配置路由就行了吧

jjm2473, 2023-12-22:
把规则都删掉，看看你的静态路由怎么配置的

closed 2024-01-03 by jjm2473