istoreos/istoreos #1193: 树莓派4B重启，树莓派自带的WIFI失效

N1k1Zhang, 2023-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ｘ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：树莓派4B重启时，若USB３．０接口有优盘插入，则重启后树莓派自带的WiFi失效（无法被搜索到），若拔出USB３.０接口的优盘再重启，则一切正常。PS：USB２．０接口我没试过。

### (2) 路由器型号和固件版本

A：设备型号
Raspberry Pi 4 Model B Rev 1.4
固件版本
iStoreOS 22.03.5 2023121510

### (3) 详细日志和/或截图

A：
无

jjm2473, 2023-12-22:
树莓派是 openwrt 维护的，istoreos 基本没改啥，所以你最好试试 openwrt 22.03.x 原版固件是否有同样的问题，如果有，那可以反馈给 openwrt 社区

N1k1Zhang, 2023-12-24:
好的谢谢

N1k1Zhang, 2023-12-24:
> 树莓派是 openwrt 维护的，istoreos 基本没改啥，所以你最好试试 openwrt 22.03.x 原版固件是否有同样的问题，如果有，那可以反馈给 openwrt 社区

我研究了一下，可能原因是重启系统后，默认树莓派自带的WIFI就是关闭状态，因为我自己连的USB WiFi每次重启都可以自动启动。在哪里设置可以是自带WiFi默认重启后开启状态，或者你们的固件可以加入这个默认选项，这对于单网口设备将会非常方便，谢谢

jjm2473, 2023-12-28:
openwrt本来就会在加载wifi驱动以后等待1秒钟的，如果树莓派自带的wifi初始化速度比较慢，可能就会出现这种情况。
你可以在后台的“系统”-“启动项”，“本地启动脚本”的 `exit 0` 前面添加一行 `sleep 5 ; wifi up`，如果能解决问题，那就是网卡初始化太慢了。不管能不能解决，都反馈给openwrt社区，也许他们有更好的解决方案。