istoreos/istoreos #1195: 管理页面无法登录

Alstonsz, 2023-12-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
X86系统,在线升级最新系统,导致管理页面无法访问.
### (1) 具体问题
系统都能正常工作.管理页面192.168.100.1、外网(已经做端口转发,未升级之前可以正常访问),无法登录访问.
A：

### (2) 路由器型号和固件版本
X86;无法登录页面,看不到固件版本.是因为在线更新固件后无法访问管理后台.
A：

### (3) 详细日志和/或截图

A：

kelvin80man, 2023-12-20:
我也遇到这个情况，管理页面无法登录访问。通过vi etc/config/network 修改了 IP地址也不行

jjm2473, 2024-01-03:
重启试试

lwj1994, 2024-01-17:
直接  ssh wan 口的 ip  也无法登录，局域网内根本找不到这个 ip

jjm2473, 2024-01-17:
@lwj1994 如果你是x86机型，在1月12日到1月16日之间升级的，且网卡是RTL8125，那你遇到的问题估计是这个 https://github.com/istoreos/istoreos/issues/1257 ，连显示器看看`ip addr`命令会不会卡死。

jjm2473, 2024-01-17:
其他人如果只是访问不了后台，联网什么的还正常，是不是之前安装了nginx？如果是，参考 https://github.com/istoreos/istoreos/issues/863 修复

lwj1994, 2024-02-25:
> ip addr

没有卡死，但是 ssh 192.168.6.157 连不了
[image]

jjm2473, 2024-02-25:
@lwj1994 防火墙默认禁止wan口访问后台

closed 2024-02-25 by jjm2473

lwj1994, 2024-03-05:
> @lwj1994 防火墙默认禁止wan口访问后台
这个问题依然存在。192.168.100.1 就是进不去.

jjm2473, 2024-03-05:
都说了wan口默认禁止访问，这不是“问题”，是设计好的。你在lan口连接就可以通过192.168.100.1访问后台。

根据你的截图，lan口都没接网线。如果你不知道哪个是lan口，那就都试一下，反正也就两个网口。

jayansi, 2024-12-30:
系统原来能用，使用手动升级iStoreOS 22.03.7, 2024122712时，升级过程一直卡在升级过程，强行重启后后台进不去。