istoreos/istoreos #1196: DNS超时，浏览器 ERR_NAME_NOT_RESOLVED, nslookup timeout was 2 seconds

linxingwu, 2023-12-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
突然部分域名无法访问，github ping不通，nslookup timeout was 2 seconds。但是百度 知乎 google还可以访问（openclash 开关都一样）
但是恢复之前的备份就会好两天，然后又出现一样的问题。

### (2) 路由器型号和固件版本

A：R2S
[Powered by LuCI istoreos-22.03 branch (git-23.342.39264-37c783a)](https://github.com/openwrt/luci)

### (3) 详细日志和/或截图

A：
Nslookup
服务器:  iStoreOS.lan
Address:  192.168.100.1

DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.
    timeout was 2 seconds.
*** 请求 iStoreOS.lan 超时
[image]

linxingwu, 2023-12-19:
尝试过在pc上直接把DNS改成ali的公共DNS，也是超时。麻烦大佬指点一下迷津

jjm2473, 2023-12-22:
不要使用openclash这种垃圾软件，卸载掉重启应该就好了

closed 2023-12-22 by jjm2473

linxingwu, 2023-12-22:
那你有推荐的替代软件吗？

jjm2473, 2023-12-22:
passwall，ssrp都比这个靠谱