istoreos/istoreos #1198: LAN 灯不亮，无法访问网关

linxingwu, 2023-12-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
间隔一段时间lan灯就不亮了，无法上网，路由器后台也无法登录

### (2) 路由器型号和固件版本

A：R2S
[Powered by LuCI istoreos-22.03 branch (git-23.342.39264-37c783a)](https://github.com/openwrt/luci)

### (3) 详细日志和/或截图

A：
Thu Dec 21 10:18:37 2023 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: started, version 2.89 cachesize 150
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: DNS service limited to local subnets
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset no-nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#7874
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Thu Dec 21 10:18:40 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Dec 21 10:18:41 2023 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Start!
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]: ttyd 1.6.3 (libwebsockets 4.2.1-09e6c20)
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]: tty configuration:
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:   start command: /bin/login
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:   close signal: SIGHUP (1)
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:   terminal type: xterm-256color
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]: LWS: 4.2.1-09e6c20, loglevel 7
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]: NET CLI SRV H1 H2 WS ConMon IPV6-off
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:    /usr/lib/libwebsockets-evlib_uv.so
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:  ++ [wsi|0|pipe] (1)
Thu Dec 21 10:18:41 2023 daemon.err ttyd[14252]: [2023/12/21 10:18:41:4646] N:  Using foreign event loop...
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:  ++ [vh|0|netlink] (1)
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:  ++ [vh|1|default||7681] (2)
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]: lws_socket_bind: nowsi: source ads 0.0.0.0
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:  ++ [wsi|1|listen|default||7681] (2)
Thu Dec 21 10:18:41 2023 daemon.notice ttyd[14252]:  Listening on port: 7681
Thu Dec 21 10:18:47 2023 daemon.info procd: - init complete -
Thu Dec 21 10:25:17 2023 user.info : luci: accepted login on / for root from 192.168.100.121

HarryToo, 2023-12-23:
我的R2S也是每隔两三天lan口就变红灯，最长没撑过5天，每次只能断电重启，也不知道啥问题

linxingwu, 2023-12-23:
我还有自动重启，还是有问题

iamalexblue, 2023-12-27:
一样的问题

dteen, 2023-12-28:
我也是R2S也是   每隔两三天lan口 就失联，远程访问发现整个lan口不见了。找不到lan。。。怎么办

iamalexblue, 2023-12-28:
我现在是每天wifi 断一回，搞得我现在都没法把硬路由插到R2S 上用，只能单独使用了，之前的版本没有这个问题，就是最新版的固件的问题，现在只能等更新了

Mioscout, 2023-12-30:
R2C有同样问题
目前我的临时解决方法是去网口配置对调Wan和Lan
这个没记错是R2S/R2C老毛病了，默认lan口是USB通道引出去的有时候会不怎么稳定，所以早期固件基本上都对调wan，lan
对调之后出现过一次原lan口现wan口降速到100Mbps，拔插了一下好了，总共已经稳定运行10天未断线

iamalexblue, 2023-12-30:
奈斯，那我回去试试

获取 Outlook for iOS<https://aka.ms/o0ukef>

iamalexblue, 2024-01-19:
> R2C有同样问题
> 
> 目前我的临时解决方法是去网口配置对调Wan和Lan
> 
> 这个没记错是R2S/R2C老毛病了，默认lan口是USB通道引出去的有时候会不怎么稳定，所以早期固件基本上都对调wan，lan
> 
> 对调之后出现过一次原lan口现wan口降速到100Mbps，拔插了一下好了，总共已经稳定运行10天未断线（出现降速的时候跑了5天，拔插之后跑了5天，一共10天没重启过）
> 
> 系统是1215版本，最新版本没更新
> 
> PS.istore自带那个网口配置可能不会吧wan6给改过去，要自己手动去改一下

你好，请问wan lan 我自己手动对调之后好像没有办法自动分配ip了，不知道是哪里填写出了错误，请问可否方便发一下wan lan 对调的具体步骤教程，多谢啦

Mioscout, 2024-01-20:
> 你好，请问wan lan 我自己手动对调之后好像没有办法自动分配ip了，不知道是哪里填写出了错误，请问可否方便发一下wan lan 对调的具体步骤教程，多谢啦

不就是去网口配置里面换一下，然后去接口里面把wan6改到wan同一口上然后重启一下吗？
我这边没有出现分配IP问题

iamalexblue, 2024-01-26:
> 最近也遇到过失联的情况，就是r2s重启后， 红色的指示灯不亮，感觉是没有启动，运气好的话，重复断点尝试，会偶尔能启动起来

我已经放弃 istoreos 了，前几天还是每天重启一次，昨天我愣是重启了四次，而且有一次是在我玩游戏的时候直接寄了，游戏掉线，最后狠下心直接换了另一个固件，我看istore os 也没有把这个问题优先级提的很高，之前还是抱着一丝期待一直等修复，等了好几个版本更新，一直都这个样子，甚至更烂了，所以放弃了。还有之前提的问题 #1002 到现在也没有修复。

Jesn, 2024-01-26:
我用这个 https://openwrt.ai/自动编译的就没问题，使用istoreos就启动不起来，重新刷固件还是起不来，不知道低级格式化下是否会好，晚上回去试下

iamalexblue, 2024-01-26:
> 我用这个 https://openwrt.ai/ 自动编译的就没问题，使用istoreos就启动不起来，重新刷固件还是起不来，不知道低级格式化下是否会好，晚上回去试下

哈哈哈巧了，我也用的这个，目前使用体验良好，我不打算再折腾 istoreos 了

Jesn, 2024-01-30:
> > 我用这个 https://openwrt.ai/ 自动编译的就没问题，使用istoreos就启动不起来，重新刷固件还是起不来，不知道低级格式化下是否会好，晚上回去试下
> 
> 哈哈哈巧了，我也用的这个，目前使用体验良好，我不打算再折腾 istoreos 了

你编译的这个  防火墙规则有问题么？我的防火墙规则是老的，`OpenClash` `paswall` 都上不了网，最后看issue 把`OpenClash` 改为tun模式才可以上网的

iamalexblue, 2024-01-30:
我的防火墙没有问题喔，openclash 更新了下内核（目前用的Meta内核）之后添加订阅就能直接用了，目前已稳定运行四天，没有任何问题

获取 Outlook for iOS<https://aka.ms/o0ukef>

Jesn, 2024-01-30:
> 我的防火墙没有问题喔，openclash 更新了下内核（目前用的Meta内核）之后添加订阅就能直接用了，目前已稳定运行四天，没有任何问题 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

我的目前也非常稳定，就是风扇会一直在运行，后期想找个pwm的脚本控制下风扇

iamalexblue, 2024-01-30:
> > 我的防火墙没有问题喔，openclash 更新了下内核（目前用的Meta内核）之后添加订阅就能直接用了，目前已稳定运行四天，没有任何问题 获取 Outlook for iOS<https://aka.ms/o0ukef>
> 
> > […](#)
> 
> 
> 
> 
> 我的目前也非常稳定，就是风扇会一直在运行，后期想找个pwm的脚本控制下风扇

我是目前还没有配风扇，温度在45-47度左右徘徊，目前istore 装了几个插件，后台跑了一个`Cloudflared`, docker 跑了一个小工具，内存占用50%左右，cpu占用20%左右，感觉是有点小热，但还能接受。

ThomasEricZ, 2024-04-21:
我的R2S刷istoreos,运行passwall也会不定时的lan失联,查看初步分析为DHCP服务崩溃。  
我现在使用的是友善官方的FriendlyWrt固件,passwall是手动通过软件包安装的(安装过程与原版安装passwall一致),PS: 我安装的时候没有负载均衡组件,需要手动搜索Haproxy安装,重启R2S就可已使用了。  
目前运行稳定运行半个月,没有出现失联的情况了。  
有同样问题的人可以参考一下。