istoreos/istoreos #1199: 一些关于组网的建议（mesh、有线、无线、ap、ac）

Akimio521, 2023-12-21:
我是有多台路由设备的，之前把一台软路由刷成iStoreOS后感觉体验不错，不仅功能展现直观，配置容易，也保留了OpenWRT很多高级设置，于是我就把家里所有的路由器从集客刷成了iStoreOS，但是在配置的时候我发现iStoreOS并没有快速组网的功能，这算是整个OpenWRT生态的痛点吧，希望可以学习小米，中兴等成品路由器厂商的组网方式（无论是有线还是无线）

此外，虽然可以通过easymesh等插件来组网，但是需要单独设置各个路由器等Wi-Fi密码信道，kvr等各种参数，并不是很方便，如果iStoreOS可以在系统初始化的时候监测一下运行环境就好了，如果上级有已经配置好的iStoreOS设备可以直接获取配置好的wifi密码等参数，快速配置并配置好组网信息（AP模式）

在主路由方面，希望可以学习集客，在web界面就可以看到各个加入iSroreOS的组网路由器，其连接设备数量、与主路由的连接情况等等信息

感谢iStoreOS团队的付出

jjm2473, 2023-12-22:
openwrt其实有个BATMAN的东西，你可以找资料看看

Akimio521, 2023-12-22:
> openwrt其实有个BATMAN的东西，你可以找资料看看

据我了解Batman只是一个负责mesh有线回传的一个接口，不知道我的理解有没有问题
但我想要一个AC可以管理所有AP的插件，类似集群，在主路由的web页面可以直接管理其他路由的设置，例如主路由设置好了wifi的账号密码可以直接同步到子路由上，在主路由可以修改子路由的信道之类的，在主路由的web页面上也可以看到连接到子路由的设备数量之类的

jjm2473, 2023-12-22:
软路由的wifi性能都比较一般吧，还不如使用ubnt这些成熟的ac/ap产品，而且ubnt的ac还可以用docker部署运行在软路由上

Akimio521, 2023-12-22:
> 软路由的wifi性能都比较一般吧，还不如使用ubnt这些成熟的ac/ap产品，而且ubnt的ac还可以用docker部署运行在软路由上

我使用的就是360t7，k2p这种ac+ap刷的istoreos，对于300m的宽带也算是够用了

jjm2473, 2023-12-22:
你可以给openwrt社区提一下建议，毕竟openwrt的开发者比istoreos多得多，而且openwrt本身支持的硬件大部分就是带wifi的，wifi对于istoreos来说反而不是很重要。

你说的360t7，k2p固件应该也只是其他的openwrt固件加上了istore应用商店而已，还不算是真正的istoreos

Akimio521, 2023-12-28:
> 你可以给openwrt社区提一下建议，毕竟openwrt的开发者比istoreos多得多，而且openwrt本身支持的硬件大部分就是带wifi的，wifi对于istoreos来说反而不是很重要。
> 
> 
> 
> 你说的360t7，k2p固件应该也只是其他的openwrt固件加上了istore应用商店而已，还不算是真正的istoreos

并不是只是加了istore商店，而是是直接从istoreos给的源码直接编译的

jjm2473, 2023-12-28:
istoreos只适配了x86和部分arm64平台，其他平台完全没任何适配，就算能编译也不一定能启动，能启动也是功能不完整的，例如沙箱，升级保留插件等功能，只是相当于原版openwrt固件加上了istore应用商店而已。而且k2p的开源版wifi驱动也不好用，还不如用lean仓库的k2p固件，有闭源wifi驱动。