istoreos/istoreos #1200: h69k 升级不到最新的系统

tyinet, 2023-12-21:
h69k 检查更新下载固件，点击更新的时候，不管去不去掉保留配置，h69k就重启了一下。开机还是旧的固件，一直死循环，不管怎么web界面点击升级都不行，请教这个是什么问题呢？

web 界面显示的固件版本
istoreos-22.03.5-2023121510-h6xk-squashfs-combined.img.gz

现在都固件版本
iStoreOS 22.03.5 r20370-712ed14f8b

johnshuxian, 2023-12-21:
我也是一样的问题

jjm2473, 2023-12-22:
`iStoreOS 22.03.5 r20370-712ed14f8b`这个版本号不像是我们发布的

jjm2473, 2023-12-22:
你们一开始刷的固件是从哪里下载的？

LWEI2020, 2023-12-22:
我也是一样的问题，现在固件版本 iStoreOS 22.03.5 2023102711；最新固件是istoreos-22.03.5-2023121510-x86-64-squashfs-combined.img.gz

johnshuxian, 2023-12-23:
> 我也是一样的问题，现在固件版本 iStoreOS 22.03.5 2023102711；最新固件是istoreos-22.03.5-2023121510-x86-64-squashfs-combined.img.gz

我也听停在这个版本了 升级不了

johnshuxian, 2023-12-25:
> 你们一开始刷的固件是从哪里下载的？

就是从官网上下载的 还更新了几次

jjm2473, 2023-12-28:
之后的固件会加上保留系统升级日志的功能，不过也只有升级到下个版本固件之后才有这个功能

jjm2473, 2023-12-28:
你们有在软件包升级过系统的软件包吗？尤其是base-files这个包。

tyinet, 2023-12-28:
我尝试恢复出厂设置，通过升级，也是升级不上去。

tyinet, 2023-12-28:
base-files | 1510-r20370-712ed14f8b  版本

jjm2473, 2023-12-28:
@tyinet 都说了你的固件都不是官方的，版本号都不是日期形式，你还是先刷个官方固件再说吧。

jjm2473, 2023-12-28:
从istore安装“系统便利工具”（可以搜索SysTools），然后打开插件，选择“修复系统软件”，然后点击“执行”，完成以后会自动重启。重启以后再尝试升级固件。

__不要再升级不是自己安装的软件包，openwrt有些软件包跟istoreos不兼容，或者就是还有bug没修。__

wanggy429, 2023-12-28:
尝试了修复系统软件，还是不行，basefile版本：12-2023120811，当前版本：StoreOS 22.03.5 2023120811，检测到的新版本：istoreos-22.03.5-2023121510-x86-64-squashfs-combined.img.gz

LWEI2020, 2023-12-29:
尝试了修复系统软件，还是不行
X86 8+128 6网口
现在固件版本 iStoreOS 22.03.5 2023102711；最新固件是istoreos-22.03.5-2023121510-x86-64-squashfs-combined.img.gz
重启后  仍然是原来版本
希望能研究一下

Ocean-Hw, 2024-01-05:
X86 物理机 同样问题 没有更新过base-files
系统便利工具修复系统软件后安装更新无效  更新重启后依然是原版本  
base-files版本12-2023112413
系统版本 22.03.5 2023112413
这种只能格式化重装吗

jjm2473, 2024-01-05:
@Ocean-Hw 试试先停掉docker，然后再升级。为了节省测试时间，先下载好最新固件，然后在后台的“系统”-“备份与升级”里面升级，这样就算失败了也不用再重新下载

Ocean-Hw, 2024-01-08:
> @Ocean-Hw 试试先停掉docker，然后再升级。为了节省测试时间，先下载好最新固件，然后在后台的“系统”-“备份与升级”里面升级，这样就算失败了也不用再重新下载

使用/etc/init.d/dockerd stop停止docker  
/etc/init.d/dockerd status确认处于inactive状态后  在“系统”-“备份与升级”里面升级，依然重启后还是老系统。

jjm2473, 2024-01-08:
@Ocean-Hw 你是不是扩容过根目录，也就是overlay？

Ocean-Hw, 2024-01-08:
> @Ocean-Hw 你是不是扩容过根目录，也就是overlay？

没有的，overlay就是原始的2GB，剩下的空间单独格式化成/mnt/xxxx的。

jjm2473, 2024-01-08:
@Ocean-Hw 那应该不会影响升级。既然你是x86的，那就接上显示器，然后用手机录屏（如果有采集卡就更好），然后开始升级，在最终刷写固件之前开始录就行了，把升级过程的录屏发来看看，放在网盘或者 wetransfer.com ，或者邮箱发我都行。

Ocean-Hw, 2024-01-08:
> wetransfer.com

大佬您好我录了一个发到你Gmail了 麻烦帮忙看看 谢谢呀

Ocean-Hw, 2024-01-08:
> @Ocean-Hw 那应该不会影响升级。既然你是x86的，那就接上显示器，然后用手机录屏（如果有采集卡就更好），然后开始升级，在最终刷写固件之前开始录就行了，把升级过程的录屏发来看看，放在网盘或者 wetransfer.com ，或者邮箱发我都行。

https://github.com/istoreos/istoreos/issues/1200#issuecomment-1880322568

Ocean-Hw, 2024-01-08:
> @Ocean-Hw 那应该不会影响升级。既然你是x86的，那就接上显示器，然后用手机录屏（如果有采集卡就更好），然后开始升级，在最终刷写固件之前开始录就行了，把升级过程的录屏发来看看，放在网盘或者 wetransfer.com ，或者邮箱发我都行。

大佬我录了一个发您Gmail了 谢谢呀

jjm2473, 2024-01-08:
@Ocean-Hw 谢谢，我看看

jjm2473, 2024-01-08:
[image]

@Ocean-Hw 我怀疑是mount命令被替换成其他版本了，你打开终端执行下这个命令`ls -l /bin/mount; grep /bin/mount -r /usr/lib/opkg/info`看看输出什么

jjm2473, 2024-01-08:
@Ocean-Hw 执行命令没，输出什么？

Ocean-Hw, 2024-01-08:
我后面出来了 明天一早试了发给你 谢谢大哥

发自我的iPhone

wanggy429, 2024-01-08:
> <img alt="image" width="630" src="https://private-user-images.githubusercontent.com/3896703/294806723-5ee9673e-7bfe-438f-b4ed-d0edd2ac0caa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MjMwNDEsIm5iZiI6MTcwNDcyMjc0MSwicGF0aCI6Ii8zODk2NzAzLzI5NDgwNjcyMy01ZWU5NjczZS03YmZlLTQzOGYtYjRlZC1kMGVkZDJhYzBjYWEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDhUMTQwNTQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjkwZjZiYjhiZjE0MjNmN2IyYmUwZWY0MWJhNDYxMzI4ZmZkNzQ2MTgwYTc5MGUxZjNkOWI2NGE5YjNjYTYwMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.DOFe6o4gqWH8BxJNfpRyF9UhwHl-0Olc860uzM42i_o">
> @Ocean-Hw 我怀疑是mount命令被替换成其他版本了，你打开终端执行下这个命令`ls -l /bin/mount; grep /bin/mount -r /usr/lib/opkg/info`看看输出什么

印象中，我是配置rclone 的时候，以为没有安装fuse3 ，使用了符号链接命令，把fuse2链接为fuse3 ，是不是这个问题导致的呢

Ocean-Hw, 2024-01-09:
> @Ocean-Hw 执行命令没，输出什么？

我来啦
 -------------------------------------------------------
 iStoreOS 22.03.5, 2023112413
 -------------------------------------------------------
root@iStoreOS:~# ls -l /bin/mount; grep /bin/mount -r /usr/lib/opkg/info
lrwxrwxrwx    1 root     root             7 Nov 23 19:17 /bin/mount -> busybox
/usr/lib/opkg/info/busybox.list:/bin/mount
/usr/lib/opkg/info/mount-utils.list:/usr/bin/mountpoint
/usr/lib/opkg/info/mount-utils.list:/usr/bin/mount
root@iStoreOS:~#

Ocean-Hw, 2024-01-09:
> @Ocean-Hw 执行命令没，输出什么？

[image]

Ocean-Hw, 2024-01-09:
> @Ocean-Hw 执行命令没，输出什么？

大神 在吗

jjm2473, 2024-01-10:
@Ocean-Hw 看起来mount命令没问题，执行下`mount |grep /tmp`看看

jjm2473, 2024-01-10:
@wanggy429 跟rclone和fuse没啥关系，除非rclone挂载到了`/tmp/root`

Ocean-Hw, 2024-01-11:
> @Ocean-Hw 看起来mount命令没问题，执行下`mount |grep /tmp`看看

[image]

jjm2473, 2024-01-11:
@Ocean-Hw 执行`opkg info busybox; /bin/mount --help`看看

Ocean-Hw, 2024-01-11:
> @Ocean-Hw 执行`mount | grep /proc; opkg info busybox; /bin/mount --help`看看

[image]

jjm2473, 2024-01-11:
@Ocean-Hw 提取系统里的`/lib/upgrade/stage2`文件看看

Ocean-Hw, 2024-01-11:
> @Ocean-Hw 提取系统里的`/lib/upgrade/stage2`文件看看

执行这个以后显示connection closed了  再刷新显示/usr/lib/lua/luci/dispatcher.lua:430: /etc/config/luci seems to be corrupt, unable to find section 'main'
只有断电重启

jjm2473, 2024-01-11:
@Ocean-Hw 不是让你执行，是提取这个文件。这个文件执行以后会切换根目录，肯定会导致系统问题

Ocean-Hw, 2024-01-11:
> @Ocean-Hw 不是让你执行，是提取这个文件。这个文件执行以后会切换根目录，肯定会导致系统问题

#!/bin/sh

. /lib/functions.sh
. /lib/functions/system.sh

export IMAGE="$1"
COMMAND="$2"

export INTERACTIVE=0
export VERBOSE=1
export CONFFILES=/tmp/sysupgrade.conffiles

RAMFS_COPY_BIN=		# extra programs for temporary ramfs root
RAMFS_COPY_DATA=	# extra data files

include /lib/upgrade

supivot() { # <new_root> <old_root>
	/bin/mount | grep "on $1 type" 2>&- 1>&- || /bin/mount -o bind $1 $1
	mkdir -p $1$2 $1/proc $1/sys $1/dev $1/tmp $1/overlay $1/ext_overlay && \
	/bin/mount -o noatime,move /proc $1/proc && \
	pivot_root $1 $1$2 || {
		/bin/umount -l $1 $1
		return 1
	}

	/bin/mount -o noatime,move $2/sys /sys
	/bin/mount -o noatime,move $2/dev /dev
	/bin/mount -o noatime,move $2/tmp /tmp
	/bin/mount -o noatime,move $2/overlay /overlay 2>&-
	/bin/mount -o noatime,move $2/ext_overlay /ext_overlay 2>&-
	return 0
}

switch_to_ramfs() {
	RAMFS_COPY_LOSETUP="$(command -v /usr/sbin/losetup)"
	RAMFS_COPY_LVM="$(command -v lvm)"

	for binary in \
		/bin/busybox /bin/ash /bin/sh /bin/mount /bin/umount	\
		pivot_root mount_root reboot sync kill sleep		\
		md5sum hexdump cat zcat dd tar				\
		ls basename find cp mv rm mkdir rmdir mknod touch chmod \
		'[' printf wc grep awk sed cut tail			\
		mtd partx losetup mkfs.ext4 nandwrite flash_erase	\
		ubiupdatevol ubiattach ubiblock ubiformat		\
		ubidetach ubirsvol ubirmvol ubimkvol			\
		snapshot snapshot_tool date logger			\
		/usr/sbin/fw_printenv /usr/bin/fwtool			\
		$RAMFS_COPY_LOSETUP $RAMFS_COPY_LVM			\
		$RAMFS_COPY_BIN
	do
		local file="$(command -v "$binary" 2>/dev/null)"
		[ -n "$file" ] && install_bin "$file"
	done
	install_file /etc/resolv.conf /lib/*.sh /lib/functions/*.sh	\
		/lib/upgrade/*.sh /lib/upgrade/do_stage2 		\
		/usr/share/libubox/jshn.sh /usr/sbin/fw_setenv		\
		/etc/fw_env.config $RAMFS_COPY_DATA

	mkdir -p $RAM_ROOT/var/lock

	[ -L "/lib64" ] && ln -s /lib $RAM_ROOT/lib64

	supivot $RAM_ROOT /mnt || {
		v "Failed to switch over to ramfs. Please reboot."
		exit 1
	}

	/bin/mount -o remount,ro /mnt
	/bin/umount -l /mnt

	grep -e "^/dev/dm-.*" -e "^/dev/loop.*" /proc/mounts | while read bdev mp _r; do
		umount $mp
	done

	[ "$RAMFS_COPY_LOSETUP" ] && losetup -D
	[ "$RAMFS_COPY_LVM" ] && {
		mkdir -p /tmp/lvm/cache
		$RAMFS_COPY_LVM vgchange -aln --ignorelockingfailure
	}
}

kill_remaining() { # [ <signal> [ <loop> ] ]
	local loop_limit=10

	local sig="${1:-TERM}"
	local loop="${2:-0}"
	local run=true
	local stat
	local proc_ppid=$(cut -d' ' -f4  /proc/$$/stat)

	v "Sending $sig to remaining processes ..."

	while $run; do
		run=false
		for stat in /proc/[0-9]*/stat; do
			[ -f "$stat" ] || continue

			local pid name state ppid rest
			read pid rest < $stat
			name="${rest#\(}" ; rest="${name##*\) }" ; name="${name%\)*}"
			set -- $rest ; state="$1" ; ppid="$2"

			# Skip PID1, our parent, ourself and our children
			[ $pid -ne 1 -a $pid -ne $proc_ppid -a $pid -ne $$ -a $ppid -ne $$ ] || continue

			[ -f "/proc/$pid/cmdline" ] || continue

			local cmdline
			read cmdline < /proc/$pid/cmdline

			# Skip kernel threads
			[ -n "$cmdline" ] || continue

			v "Sending signal $sig to $name ($pid)"
			kill -$sig $pid 2>/dev/null

			[ $loop -eq 1 ] && run=true
		done

		let loop_limit--
		[ $loop_limit -eq 0 ] && {
			v "Failed to kill all processes."
			exit 1
		}
	done
}

istoreos_pre_upgrade() {
	if [ -n "$UPGRADE_BACKUP" ]; then
		touch /usr/lib/opkg/.upgrading 2>/dev/null && sync /
		# reset rom uuid in ext_overlay
		if [ -e /ext_overlay/etc ]; then
			[ -f /ext_overlay/upper/usr/lib/opkg/.upgrading ] || {
				rm -f /ext_overlay/upper/usr/lib/opkg/.upgrading
				mkdir -p /ext_overlay/upper/usr/lib/opkg
				touch /ext_overlay/upper/usr/lib/opkg/.upgrading 2>/dev/null
			}
			touch /ext_overlay/.upgrading
			[ -f /ext_overlay/upper/usr/lib/opkg/.upgrading ] && rm -f /ext_overlay/etc/.extroot-uuid
			sync /ext_overlay
		fi
		[ -f /overlay/upper/usr/lib/opkg/.upgrading ] || {
			rm -f /overlay/upper/usr/lib/opkg/.upgrading
			mkdir -p /overlay/upper/usr/lib/opkg
			touch /overlay/upper/usr/lib/opkg/.upgrading 2>/dev/null
			touch /overlay/.upgrading
			sync /overlay
		}
	fi
}

indicate_upgrade

while read -r a b c; do
	case "$a" in
		MemT*) mem="$b" ;; esac
done < /proc/meminfo

[ "$mem" -gt 32768 ] && \
	skip_services="dnsmasq log network"
for service in /etc/init.d/*; do
	service=${service##*/}

	case " $skip_services " in
		*" $service "*) continue ;; esac

	ubus call service delete '{ "name": "'"$service"'" }' 2>/dev/null
done

killall -9 telnetd 2>/dev/null
killall -9 dropbear 2>/dev/null
killall -9 ash 2>/dev/null

kill_remaining TERM
sleep 4
sync
/usr/bin/umount -R -d /mnt &
sleep 5
kill_remaining KILL 1

sleep 6

echo 3 > /proc/sys/vm/drop_caches

[ -n "$IMAGE" ] && istoreos_pre_upgrade

if [ -n "$IMAGE" ] && type 'platform_pre_upgrade' >/dev/null 2>/dev/null; then
	platform_pre_upgrade "$IMAGE"
fi

ROOTFS_TYPE=$(rootfs_type)
if [ -n "$ROOTFS_TYPE" -a "$ROOTFS_TYPE" != "tmpfs" -a "$ROOTFS_TYPE" != "rootfs" ]; then
	v "Switching to ramdisk..."
	switch_to_ramfs
fi

grep /ext_overlay /proc/mounts > /dev/null && {
	/bin/mount -o noatime,remount,ro /ext_overlay
	/usr/bin/umount -R -d -l /ext_overlay 2>/dev/null || /bin/umount -l /ext_overlay
}

grep /overlay /proc/mounts > /dev/null && {
	/bin/mount -o noatime,remount,ro /overlay
	/usr/bin/umount -R -d -l /overlay 2>/dev/null || /bin/umount -l /overlay
}

# Exec new shell from ramfs
exec /bin/busybox ash -c "$COMMAND"

jjm2473, 2024-01-11:
@Ocean-Hw 看起来也是一致的。下午或晚上有空吗，要不我远程看看？

Ocean-Hw, 2024-01-11:
> @Ocean-Hw 看起来也是一致的。下午或晚上有空吗，要不我远程看看？

私个您的联系方式给我？看看吧 我感觉我这个有可能不是个例。谢谢。

jjm2473, 2024-01-11:
@Ocean-Hw  嗯，我邮件发你

jjm2473, 2024-01-11:
x86用户在手动下载并升级固件的时候，要注意是否EFI固件，如果搞错，跟运行中的类型不一致，可能会刷坏系统。也就是说，如果原本刷的是efi固件，就要下载文件名带efi的固件来升级。如果不知道运行中的固件是什么类型，可以执行这个命令 `(. /lib/upgrade/common.sh; export_bootdevice && export_partdevice diskdev 0 || exit 0; blkid -o value -s PTTYPE /dev/$diskdev)` ，如果输出“gpt”则说明是EFI固件，如果输出“dos”则不是EFI固件。

leechhboy, 2024-01-13:
大神 R5C 也是升级不了新版本 docker里只安装了一个clouddrive2 是一键安装的 没有安装别的插件 固件是在TF卡里 剩余空间格式化成docker空间 每次升级需要重置  停用docker都不行
2024-01-13 00:12:05
Running Version: 'iStoreOS 22.03.5 2023122916'
Flashing Version: 'iStoreOS 22.03.6 2024011211'
Sending TERM to remaining processes ...
Sending signal TERM to S99adb-enablemo (11407)
Sending signal TERM to adb (11435)
Sending signal TERM to adb (11456)
Sending signal TERM to bpfilter_umh (1599)
Sending signal TERM to dockerd (18389)
Sending KILL to remaining processes ...

重置后升级
Switching to ramdisk...
Performing system upgrade...
Reading partition table from image...
Writing image to /dev/mmcblk0p1...
Writing image to /dev/mmcblk0p2...
Skip partition 3 >= 3 when upgrading
Writing new UUID to /dev/mmcblk0...
Upgrade completed
Saving config files...
Saving config files...
Saving config files...

jjm2473, 2024-01-13:
@leechhboy 不重置的情况下，输出`Sending KILL to remaining processes ...`之后就没了？卡死了还是自动重启了？

leechhboy, 2024-01-13:
@jjm2473 在线升级固件 保存设置 过一会就自动重启 然后到登录界面 进去以后版本号没有变  固件更新 还有提示

jjm2473, 2024-01-13:
@leechhboy 如果自动重启了，应该是遇到错误，例如某些进程杀不掉，或者没法切换根目录，这些应该会有日志。或者，你现在看看`/overlay/upgrade.log`这个文件是否存在，有的话把内容发出来

leechhboy, 2024-01-13:
@jjm2473 
2024-01-13 00:12:05
Running Version: 'iStoreOS 22.03.5 2023122916'
Flashing Version: 'iStoreOS 22.03.6 2024011211'
Sending TERM to remaining processes ...
Sending signal TERM to S99adb-enablemo (11407)
Sending signal TERM to adb (11435)
Sending signal TERM to adb (11456)
Sending signal TERM to bpfilter_umh (1599)
Sending signal TERM to dockerd (18389)
Sending KILL to remaining processes ...
Switching to ramdisk...
Performing system upgrade...
Reading partition table from image...
Writing image to /dev/mmcblk0p1...
Writing image to /dev/mmcblk0p2...
Skip partition 3 >= 3 when upgrading
Writing new UUID to /dev/mmcblk0...
Upgrade completed
Saving config files...
Saving config files...
Saving config files...

这些就是日志

jjm2473, 2024-01-13:
@leechhboy 这个日志说明刷成功了啊。

leechhboy, 2024-01-13:
@jjm2473 
直接升级
2024-01-13 00:12:05
Running Version: 'iStoreOS 22.03.5 2023122916'
Flashing Version: 'iStoreOS 22.03.6 2024011211'
Sending TERM to remaining processes ...
Sending signal TERM to S99adb-enablemo (11407)
Sending signal TERM to adb (11435)
Sending signal TERM to adb (11456)
Sending signal TERM to bpfilter_umh (1599)
Sending signal TERM to dockerd (18389)
Sending KILL to remaining processes ...

重置后
Switching to ramdisk...
Performing system upgrade...
Reading partition table from image...
Writing image to /dev/mmcblk0p1...
Writing image to /dev/mmcblk0p2...
Skip partition 3 >= 3 when upgrading
Writing new UUID to /dev/mmcblk0...
Upgrade completed
Saving config files...
Saving config files...
Saving config files...

leechhboy, 2024-01-13:
我买r5c以来 只有重置才能升级成功  里边就docker安装了一个clouddrive2  希望大佬把clouddrive2加入 istore  现在这个软件很火，挂网盘比alist好用

jjm2473, 2024-01-13:
@leechhboy  `/overlay/upgrade.log`只会记录最后一次操作升级的日志（不管是OTA还是手动上传的固件），不会出现两次升级日志混在一起的情况。

leechhboy, 2024-01-13:
@jjm2473 
大佬没关系，晚上我回去降级 安装clouddrive后在升级一次   然后给你发log 只要能修复

leechhboy, 2024-01-13:
@jjm2473 大佬看一下 降级后重新升级

2024-01-13 19:35:57
Running Version: 'iStoreOS 22.03.5 2023122916'
Flashing Version: 'iStoreOS 22.03.6 2024011211'
Sending TERM to remaining processes ...
Sending signal TERM to dockerd (15824)
Sending signal TERM to containerd (15858)
Sending signal TERM to bpfilter_umh (1604)
Sending signal TERM to S99adb-enablemo (8932)
Sending signal TERM to adb (8967)
Sending signal TERM to adb (8975)
Sending KILL to remaining processes ...

leechhboy, 2024-01-13:
@jjm2473  我又在线升级了一次  还是不行
2024-01-13 19:48:03
Running Version: 'iStoreOS 22.03.5 2023122916'
Flashing Version: 'iStoreOS 22.03.6 2024011211'
Sending TERM to remaining processes ...
Sending signal TERM to containerd (11158)
Sending signal TERM to containerd-shim (11673)
Sending signal TERM to bpfilter_umh (1603)
Sending signal TERM to S99adb-enablemo (8853)
Sending signal TERM to adb (8866)
Sending signal TERM to adb (8888)
Sending signal TERM to dockerd (9395)
Sending KILL to remaining processes ...
Sending signal KILL to dockerd (9395)

jjm2473, 2024-01-13:
@leechhboy  就卡在这里？还是自动重启了？

leechhboy, 2024-01-13:
@jjm2473 自动重启 登录后版本号不变 固件更新有提示  我又重置了 升级成功  但是这次没有upgrade.log文件

leechhboy, 2024-01-13:
每次升级不会卡死 是自动重启 然后登录进去后版本号不变 固件更新有提示

leechhboy, 2024-01-13:
@jjm2473 是的 每次都稳定复现
clouddrive2安装地址https://github.com/sublaim/clouddrive2
不需要配置 一键安装的

[21 more replies]

closed 2024-04-08 by jjm2473