istoreos/istoreos #1206: istoreos 的snmp好像有问题

wuai1024, 2023-12-24:
目前我是使用的爱快作为主路由，使用的istoreos作为旁路由做代理。

跨三层在爱快上终端列表里面有很多mac一样的设备，按照以前的方法，在istoreos里我搜索了snmp并成功进行了安装，在爱快里配置好了。

但是还是在终端有部分异常，是不是这个snmp的有问题。

[image]

这个是爱快的设置
[image]

jjm2473, 2023-12-28:
没用过snmp，也没看出截图有什么问题，是系统和终端类型显示unknown吗？可能只是因为这些机器在dhcp的时候本来就没提供主机名，或者就是istoreos重启了但是这些机器没重启或没重新dhcp。

wuai1024, 2023-12-28:
不是，你仔细看我第一张截图，显示的 终端设备，所有的mac地址一样。因为使用istoreos做旁路由，跨三层了。所以需要使用snmp来识别到正确的mac地址。

jjm2473, 2023-12-29:
这个mac地址是不是istoreos自己的？

cooldcc, 2024-01-13:
我的升级到iStoreOS 22.03.6 2024011211 / LuCI istoreos-22.03 branch git-24.009.14990-dc7527f  这个版本之后出现同样的问题。
[image]
[image]
软路由esxi底层 主爱快辅istoreos   这个重复的mac是esxi下istoreos的mac地址    用了一年了 升级is好几次没问题  就这次升级之后出现了这个问题  设置什么的都没改动   貌似是这个snmp有问题

jjm2473, 2024-01-13:
@cooldcc istoreos没有集成snmp，snmp肯定是自己安装的，不会跟着固件升级而升级，那出现问题前你有没有升级snmp呢？

cooldcc, 2024-01-15:
@jjm2473   爱快和istoreos 升级之前没有任何改动    就是直接升级istoreos 22.03.6 2024011211 / LuCI istoreos-22.03 branch git-24.009.14990-dc7527f  固件之后出现问题       现在回滚上一个版本
iStoreOS 22.03.5 2023112413 / LuCI istoreos-22.03 branch git-23.271.10849-b445bc4  就没有问题

jjm2473, 2024-01-16:
那你试试升级固件以后重装snmp

wuai1024, 2024-01-16:
我经历过无数次重装测试后，我找到原因了。是因为我打开了网卡的SR-IOV。 我把istoreos的网卡，取消了 SR-IOV后。

snmp 就能正常工作，获取到真实的mac地址了。

jjm2473, 2024-01-16:
老版本固件的sriov又没问题？

cooldcc, 2024-01-17:
@wuai1024   SR-IOV 在哪里设置？  我去瞅瞅

wuai1024, 2024-01-17:
> @wuai1024 SR-IOV 在哪里设置？ 我去瞅瞅

要网卡支持

closed 2024-02-02 by jjm2473

elan5460, 2024-07-25:
> 我的升级到iStoreOS 22.03.6 2024011211 / LuCI istoreos-22.03 branch git-24.009.14990-dc7527f 这个版本之后出现同样的问题。 ![123](https://private-user-images.githubusercontent.com/106000123/296459584-c3d65158-4da6-467b-8b38-45e256ed815f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE5MDI0MTgsIm5iZiI6MTcyMTkwMjExOCwicGF0aCI6Ii8xMDYwMDAxMjMvMjk2NDU5NTg0LWMzZDY1MTU4LTRkYTYtNDY3Yi04YjM4LTQ1ZTI1NmVkODE1Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyNVQxMDA4MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZmZmZDJjZDkxNGM2Yzg3ZjQ1ZmVkODAxMDc3ZjAxZjk5ZmYzZDNjY2I4NDM5NWIxZjJjYWRhYzc5ZDE5OGU5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.abFBJHKWai59lExM-aCFjGQYTWA9VZUYR6wPs4zELyE) ![1234](https://private-user-images.githubusercontent.com/106000123/296459605-cc143fed-b5c5-4a6c-8267-289b6fa8a96c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE5MDI0MTgsIm5iZiI6MTcyMTkwMjExOCwicGF0aCI6Ii8xMDYwMDAxMjMvMjk2NDU5NjA1LWNjMTQzZmVkLWI1YzUtNGE2Yy04MjY3LTI4OWI2ZmE4YTk2Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyNVQxMDA4MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNGMwODc4NGNmYWE5MzFjMTNiODZkYzU3MmY3YTdhYzljZTI1NzhmNDY1ZTg4MWQ5NWU1YWRhODlkNjU4MGY2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HWaSBck0NA6I1bGdicH0eD0sAfe7cv-YFNY1-7lk9oc) 软路由esxi底层 主爱快辅istoreos 这个重复的mac是esxi下istoreos的mac地址 用了一年了 升级is好几次没问题 就这次升级之后出现了这个问题 设置什么的都没改动 貌似是这个snmp有问题

兄弟，能了解一下你这个监控软件是自己写的，还是什么第三方产品吗？谢谢

cooldcc, 2024-07-26:
istoreos  这个？

elan5460, 2024-07-26:
兄弟，谢谢您回复。我最后看懂了，这个是爱快的后台界面。
对吧？

Sent from Gmail Mobile

cooldcc, 2024-07-26:
对  监控流量这是爱快后台 

 
 
 
 
 &gt; istoreos&amp;nbsp; 这个？ 
 &gt; ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------ 
 &gt; 发件人: "istoreos/istoreos" ***@***.***&amp;gt;; 
 &gt; 发送时间:&amp;nbsp;2024年7月25日(星期四) 晚上7:09 
 &gt; ***@***.***&amp;gt;; 
 &gt; ***@***.******@***.***&amp;gt;; 
 &gt; 主题:&amp;nbsp;Re: [istoreos/istoreos] istoreos 的snmp好像有问题 (Issue #1206) 
 &gt; 我的升级到iStoreOS 22.03.6 2024011211 / LuCI istoreos-22.03 branch 
 &gt; git-24.009.14990-dc7527f 这个版本之后出现同样的问题。 软路由esxi底层 主爱快辅istoreos 
 &gt; 这个重复的mac是esxi下istoreos的mac地址 用了一年了 升级is好几次没问题 就这次升级之后出现了这个问题 设置什么的都没改动 
 &gt; 貌似是这个snmp有问题 
 &gt; 兄弟，能了解一下你这个监控软件是自己写的，还是什么第三方产品吗？谢谢 
 &gt; ***@***.***&amp;gt; 
 &gt; <" rel="noopener" target="_blank">https://github.com/istoreos/istoreos/issues/1206#issuecomment-2251875048&gt;;, 
 &gt; ***@***.***&gt;

elan5460, 2024-07-26:
> 对  监控流量这是爱快后台

我有一个通过snmp流量监控的需求，所以昨天看到你这个截图想请教下。