istoreos/istoreos #1208: X86 外网测速减半

weiyideai520, 2023-12-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 外网测速减半

### (2) 路由器型号和固件版本

A：X86主机，iStoreOS 22.03.5 2023121510 / LuCI istoreos-22.03 branch git-23.342.39264-37c783a

### (3) 详细日志和/或截图

A：内外持续压力测速
[image]

B：外网测速（iStoreOS）
[image]

C：电脑直连光猫测速
[image]

weiyideai520, 2023-12-26:
网卡信息
```
root@UN45:~# ethtool -i eth0
driver: r8169
version: 5.10.176
firmware-version: rtl8168h-2_0.0.2 02/26/15
expansion-rom-version: 
bus-info: 0000:02:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
```

weiyideai520, 2023-12-27:
网络-防火墙-路由/NAT 分载

✔软件流量分载
✔硬件流量分载

后测速从5xxMbps 提升到 7xxMbps，还是没办法实现满速的 9xxMbps

jjm2473, 2023-12-28:
https://github.com/istoreos/istoreos/issues/1150

closed 2023-12-28 by jjm2473

shengnannan, 2024-05-09:
同有此问题。
软路由：康耐信的n4100，英特尔2.5G网卡。
猫：vsol 2.5G
硬路由：中兴巡天PRO+ AX3000

硬路由拨号，可以跑满千兆宽带。
软路由拨号，只可以跑到730M上下。
内网测速正常。

lnbiuc, 2025-01-21:
相同的问题，500兆宽带
[image]

[image]