istoreos/istoreos #1210: R4S 刷不带 nonopi 字样的版本都无法启动，sys灯不闪

cokyhe, 2023-12-26:
试着刷 istoreos-21.02.3-2022112305-nanopi-r4s-squashfs.img.gz，能进系统
但是...在后台升级系统到最新版本
升级后又是sys灯不亮了

jjm2473, 2023-12-26:
直接刷最新的固件不行？

cokyhe, 2023-12-26:
最新的系统不行的，只要是不带nanopi的我都挑了几个旧版本尝试过，都是sys灯没反应

jjm2473, 2023-12-28:
这个版本呢
[istoreos-21.02.3-2023060914-r4s-squashfs.img.gz](https://fw0.koolcenter.com/iStoreOS/r4s/istoreos-21.02.3-2023060914-r4s-squashfs.img.gz)

cokyhe, 2023-12-28:
> 这个版本呢 [istoreos-21.02.3-2023060914-r4s-squashfs.img.gz](https://fw0.koolcenter.com/iStoreOS/r4s/istoreos-21.02.3-2023060914-r4s-squashfs.img.gz)

牛啊~~~这个版本可以进系统，在后台升级到最新版本，也顺利升级了iStoreOS 22.03.5 2023121510 ，感谢！！！

ChristianSSH, 2024-01-19:
是1G内存的R4S吧，刷0609以上版本会出现这个系统灯不亮的问题

cokyhe, 2024-01-21:
> 是1G内存的R4S吧，刷0609以上版本会出现这个系统灯不亮的问题

4G内存

Mickpennn, 2024-02-04:
我的也是这样。不能刷入以后所有灯都在亮。但是不能进入管理界面，也没有网络。

Mickpennn, 2024-02-04:
[image]

Mickpennn, 2024-02-04:
刚刚0609刷入可以使用。但是升级到最新20240126还是无网络，也不能进入管理界面。
升级后浏览器提示
Bad Gateway
The process did not produce any response
开机后，先亮红灯，然后SYS灯闪烁，接下来WAN和LAN都闪。不过10秒后只有电源灯亮其他灯灭，然后重复上述亮灯过程。

jjm2473, 2024-02-05:
手边没有r4s，如果谁有r4s和调试工具，把日志提取出来看看。

jjm2473, 2024-02-05:
@Mickpennn 这肯定是在不断重启

icefruit-j, 2024-02-18:
[istoreos-22.03.6-2024012613-r4s-squashfs.img.gz](https://fw0.koolcenter.com/iStoreOS/r4s/istoreos-22.03.6-2024012613-r4s-squashfs.img.gz)
也是无法启动，SYS灯不亮

icefruit-j, 2024-02-19:
> 手边没有r4s，如果谁有r4s和调试工具，把日志提取出来看看。

我找了个串口板连接到R4S，日志如下：

U-Boot TPL 2021.07-OpenWrt-2023122911 (Dec 28 2023 - 06:41:56)sdram_init: LPDDR4 - 50MHz failed!
rk3399_dmc_init DRAM init failed -22
Missing DTB

jjm2473, 2024-02-19:
@icefruit-j 先刷这个版本 https://fw0.koolcenter.com/iStoreOS/r4s/istoreos-21.02.3-2023060914-r4s-squashfs.img.gz ，然后再升级到最新固件，应该就能正常启动。

还有，你能对比下openwrt官方的22.03.6固件吗？ https://downloads.openwrt.org/releases/22.03.6/targets/rockchip/armv8/openwrt-22.03.6-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz

jjm2473, 2024-02-20:
你们说的r4s是不是带emmc的版本？

jjm2473, 2024-02-20:
如果是emmc启动，应该刷这个 https://fw.koolcenter.com/iStoreOS/r4se/

icefruit-j, 2024-02-20:
> 你们说的r4s是不是带emmc的版本？

不是，镜像写TF卡里的

jjm2473, 2024-02-21:
> 一样的错误
> 
> U-Boot TPL 2021.07-OpenWrt-r20265-f85a79bcb4 (Nov 25 2023 - 18sdram_init: LPDDR4 - 50MHz failed! rk3399_dmc_init DRAM init faMissing DTB

@icefruit-j  openwrt官方的22.03.6固件也有这个错误？那就难怪了，因为uboot是从openwrt官方来的，老版本又没问题。

jjm2473, 2024-02-22:
@icefruit-j 我是说openwrt老版本的uboot没问题，因为之前istoreos 21版本也是使用openwrt相同的uboot，却能启动。建议把问题反馈给openwrt吧 https://github.com/openwrt/openwrt/issues ，最好再跟厂商也确认下硬件版本，以及为什么openwrt官方版本无法启动

moshroom96, 2024-08-16:
谢谢指导，一摸一样的问题，刷老版本刷进去了