istoreos/istoreos #1211: Docker已迁移至外部挂载后提示Bug；/overlay/upper下重启后仍自动建docker目录

hk519, 2023-12-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Docker根目录已迁移至其他硬盘分区（挂载为/storage），主页仍提示在根分区：
[image]
[image]
已将/overlay/upper/storage/docker目录删除，重启后又再次出现。（貌似在挂载其他硬盘前先启动了Docker？）

### (2) 路由器型号和固件版本

A：
固件版本：iStoreOS 22.03.5 2023121510
镜像：istoreos-22.03.5-2023121510-x86-64-squashfs-combined
VMware Workstation 17.5
[image]
[image]

hk519, 2023-12-26:
查看日志，发现Docker启动早于分区挂载，重启iStoreOS后，必须重启Docker一次才能正常使用。

jjm2473, 2023-12-29:
md0占用0B，要么迁移失败了，要么就是启动以后没有挂载

jjm2473, 2024-01-12:
应该要把挂载点设置到/mnt里面的，例如/mnt/raid。
如果硬盘加载比较慢，导致docker启动时raid还没挂载。“系统”-“杂项设置”-“启动”可以配置等待硬盘分区，下个固件支持等待raid盘

closed 2024-01-17 by jjm2473