istoreos/istoreos #1217: H69k-max，rm520n-gl 5g网卡版本速度有限制

petermaple, 2023-12-31:
(1) 具体问题

A：H69k-max，rm520n-gl 5g网卡版本，使用istoreos（20231215版本），5g网卡最大速度只有不到250Mbps。

同样位置，同sim卡，使用wifi之路下载的“原厂”固件，能够达到正常的sim卡速度（黄金会员，500Mbps）

注意到istore os使用的5g modem协议是modem manager

而“原厂”固件使用的是qmi蜂窝

如何能够排查问题？

jjm2473, 2024-01-02:
那你试试使用qmi拨号

jjm2473, 2024-01-06:
[image]

wwan接口改成qmi蜂窝不行吗？

petermaple, 2024-01-07:
> <img alt="image" width="563" src="https://private-user-images.githubusercontent.com/3896703/294645913-3785ab6e-e3ed-4283-aacc-1bcb5eff00d7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2MTM5MjcsIm5iZiI6MTcwNDYxMzYyNywicGF0aCI6Ii8zODk2NzAzLzI5NDY0NTkxMy0zNzg1YWI2ZS1lM2VkLTQyODMtYWFjYy0xYmNiNWVmZjAwZDcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDdUMDc0NzA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDhjMjI1YWFlMGFkYzk5NjA2MWNhY2QyYTY1ZmU3ZDNkOWUyMzQ2MTdlOGQ0MjJiMDhiYjkzMGMzNjlmMDMxZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.IWa4BZaZiQT85Xrj3DohpI1qSzMGvpG1sBZEmEAdgo4">
> wwan接口的协议改成qmi蜂窝不行吗？ 你先看看原厂固件的配置是什么，抄过来试试

如果istoreos选择这个协议的话，无法获得IP，似乎少了拨号的程序？

原厂这个协议不需要任何配置，原厂也没有modemmanager这个协议。

JwietPuj-Drin, 2024-02-20:
同问。

昨日购买到一台 H69K MINI ，模块广和通 FM650 ，刷 iStoreOS 20240126 ， modemmanager 报 `正在拆卸调制解调器` （好像 eth0 WAN 也拿不到上游 DHCP ，待进一步确认），刷回原厂固件一切正常。

@jjm2473

chiyuelao, 2024-12-07:
我也是小主机安装，卡在IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready，，，cpu是N2830