istoreos/istoreos #1218: 主界面实时流量统计不准

shomaruy, 2023-12-31:
反馈bug/问题模板，提建议请删除

Q：是否用关键词搜索了issue? (使用 "" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
主界面实时流量统计不准
### (1) 具体问题
主界面实时流量统计不准，跑测速时都只显示几百K

### (2) 路由器型号和固件版本
软路由 
iStoreOS 22.03.5 2023122916

### (3) 详细日志和/或截图

[image]

jjm2473, 2024-01-02:
看看右边，识别到的网关对不对

jjm2473, 2024-01-05:
还有，你是测外网还是内网的速度，这个流量统计只会计算wan口的流量

zctmdc, 2024-01-16:
## mwan 下无法识别在线的 `wan` 口
 希望可以手动指定 因为多wan 下可能需要指定 为 `lan` 或 `lan1` 者  多个 `网口`
[image]

jjm2473, 2024-01-17:
首页是根据route判断网关的，执行`route -n`看看

zctmdc, 2024-01-21:
```shell
root@HJAJLY:~# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         100.70.96.1     0.0.0.0         UG    20     0        0 pppoe-wan_ct
10.80.107.0     172.25.19.100   255.255.255.0   UG    5000   0        0 ztxxxxxx
10.136.2.0      172.25.19.100   255.255.255.0   UG    5000   0        0 ztxxxxxx
10.136.6.0      172.25.19.100   255.255.255.0   UG    5000   0        0 ztxxxxxx
10.136.7.0      172.25.19.100   255.255.255.0   UG    5000   0        0 ztxxxxxx
10.136.92.0     172.25.19.100   255.255.255.0   UG    5000   0        0 ztxxxxxx
172.25.18.0     0.0.0.0         255.255.255.0   U     0      0        0 ztyyyyyy
172.25.19.0     0.0.0.0         255.255.255.0   U     0      0        0 ztxxxxxx
172.25.20.0     0.0.0.0         255.255.255.0   U     0      0        0 ztzzzzzz
100.70.96.1     0.0.0.0         255.255.255.255 UH    0      0        0 pppoe-wan_ct
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.25.72.0     172.25.19.100   255.255.255.0   UG    5000   0        0 ztxxxxxx
192.168.3.0     0.0.0.0         255.255.255.0   U     20     0        0 eth0.53
192.168.6.0     172.25.19.99    255.255.255.0   UG    5000   0        0 ztxxxxxx
192.168.59.0    0.0.0.0         255.255.255.0   U     0      0        0 eth0.59
192.168.60.0    0.0.0.0         255.255.255.0   U     0      0        0 eth0
192.168.68.0    0.0.0.0         255.255.255.0   U     0      0        0 eth0.68
192.168.168.0   172.25.19.99    255.255.255.0   UG    5000   0        0 ztxxxxxx
192.168.180.0   172.25.19.100   255.255.255.0   UG    5000   0        0 ztxxxxxx
192.168.180.35  172.25.18.100   255.255.255.255 UGH   5000   0        0 ztyyyyyy
192.168.181.0   172.25.20.100   255.255.255.0   UG    5000   0        0 ztzzzzzz

```

jjm2473, 2024-01-23:
> 0.0.0.0         100.70.96.1     0.0.0.0         UG    20     0        0 pppoe-wan_ct

如果没有使用策略路由的话，默认网关应该是 pppoe-wan_ct

jjm2473, 2024-01-23:
执行`ubus call network.interface dump`，把openwrt的网络状态dump出来（只要wan相关的），可能是没判断route table导致的。

jjm2473, 2024-01-23:
哦，不用了，我已经测试过了，增加路由表的判断就行了

zctmdc, 2024-01-24:
希望可以增加手动指定网卡，或者多个网卡  或者 luci- :)

zctmdc, 2024-01-30:
好像不行的哦   

```bash
root@HJAJLY:~# route -n | grep '192.168.3.67'
root@HJAJLY:~# route -n | grep -e '^0.0.0.0'
0.0.0.0         100.70.96.1     0.0.0.0         UG    20     0        0 pppoe-wan_ct
```

[image]

| key | value |
| :---: | :---: |
|设备型号|QEMU Standard PC (Q35 + ICH9, 2009)|
|固件版本|iStoreOS 22.03.6 2024012613|
|内核版本|5.10.201|
|已启动|0天9小时12分31秒|
|系统时间|2024-01-31 04:35:07|

jjm2473, 2024-02-01:
执行 `ubus call network.interface dump` 看看

jjm2473, 2024-02-05:
wan_cm_cctv这个不要配置默认路由，你又不是用它来上网的。默认路由是表示你要默认通过哪个网络连接外网的意思，你难道通过iptv上外网吗？

closed 2024-02-05 by jjm2473

zctmdc, 2024-02-05:
感谢您的回复
这个 `wan_cm_cctv` 只是个名字，是 `移动网络`，最近那边的 `移动网络` 掉线了，过段时间是会修复的。
该网络使用的`静态ip`上网，所以需要配置网关。

而且，`multiWAN`已经有效识别状态，已经下线了
> [image]

jjm2473, 2024-02-09:
静态ip不需要网关好吧，不知道你怎么理解的。网关代表的就是默认情况下从哪个网络上外网。

zctmdc, 2024-02-16:
尊敬的开发者您好，新年快乐！
我需要使用 `wan_cm_cctv` 访问公网，如果不设置网关，在我的理解中应该是用不了的吧？不知道是不是我理解有误，清指出。

这是我目前的网络情况：
```mermaid
flowchart TD
  subgraph MY_NETWORK
    wan_cm_1("
        端口: wan_cm_1
        协议: DHCP
        属性: wan
        动态IP: 192.168.31.0/24
        网关: DHCP
        是否具备公网访问能力: 是
        运行商: 移动
        当前状态: 离线
    ")
    wan_cm_cctv("
        端口: wan_cm_cctv
        协议: Static
        属性: wan
        静态IP: 192.168.3.67/24
        网关: 192.168.3.1 (上级路由IP)
        是否具备公网访问能力: 是
        运行商: 移动
        当前状态: 离线
    ")
    mwan_ct("
        端口: mwan_ct
        协议: Static
        属性: wan
        静态IP: 192.168.58.10/24
        网关: 空
        是否具备公网访问能力: 否
        当前状态: 在线
    ")
    wan_ct("
        端口: wan_ct
        协议: pppoe
        属性: wan
        是否具备公网访问能力: 是
        运行商: 电信
        当前状态: 在线
    ")
    lan_ct("
        端口: lan_ct
        协议: Static
        属性: lan
        静态IP: 192.168.59.1/24
        网关: 空
        期望网络: 电信 
        --> (wan_ct)
    ")
    lan("
        端口: lan
        协议: Static
        属性: lan
        静态IP: 192.168.60.1/24
        网关: 空
        期望网络: 负载均衡 
        --> (wan_ct/wan_cm_1/wan_cm_cctv)
    ")
    lan_cctv("
        端口: lan_cctv
        协议: Static
        属性: lan
        静态IP: 192.168.68.1/24
        网关: 空
        期望网络: 负载均衡 
        --> (wan_ct/wan_cm_1/wan_cm_cctv)
    ")
    wan_cm_1-->|VLAN_31|iStoreOS
    wan_cm_cctv-->|VLAN_53|iStoreOS
    mwan_ct-->|VLAN_59|iStoreOS
    wan_ct-->|VLAN_59|iStoreOS

    iStoreOS-->|VLAN_59|lan_ct
    iStoreOS-->|VLAN_60|lan
    iStoreOS-->|VLAN_68|lan_cctv
  end
```

jjm2473, 2024-02-18:
@zctmdc 总之就是关掉wan_cm_cctv的默认网关，况且目前route也没生效，只是netifd认为你配置了。如果你确实有多wan的需求，那也不是简单的勾选多个默认网关就行的。

zctmdc, 2024-02-18:
好的 谢谢