istoreos/istoreos #1221: h28k跑网心云频繁崩溃

dalaoshi777, 2024-01-01:
设备h28k，iStore版本最新
硬盘盒为绿联带鳍片散热，ssd为梵想550pro
出现问题后内核日志，和内核日志如下

[内核日志.txt](https://github.com/istoreos/istoreos/files/13803823/default.txt)

[系统日志.txt](https://github.com/istoreos/istoreos/files/13803826/default.txt)

好像跑量太大就这样，上行宽带30mbps，

出现问题后图片
[image]
[image]

这边磁盘是不显示了，下边docker也是转圈圈，关闭网心的容器，后还是一样，点击重启，之后无法进入web，断电重启就行。
[image]
[image]

dalaoshi777, 2024-01-01:
补充一个web界面点击重启后，进不去web后台的报错
[image]

jjm2473, 2024-01-02:
看起来是USB挂掉了，你的硬盘盒是外部供电的吗

dalaoshi777, 2024-01-02:
> 看起来是USB挂掉了，你的硬盘盒是外部供电的吗

不是，供电应该是足够的，充电器是27w的，以前小米手机用的，h28k是支持pd的那一批的，这个网心，h28k讨论群里有两个跑网心的，问题跟我一模一样，也不知道是不是供电问题。

jjm2473, 2024-01-02:
你的硬盘是挂载到什么路径的？

dalaoshi777, 2024-01-02:
> 你的硬盘是挂载到什么路径的？

/mnt/usb2-1

dalaoshi777, 2024-01-05:
> 看起来是USB挂掉了，你的硬盘盒是外部供电的吗

您好目前又出现新的进展，在稳定运行三天之后又出现了，并且查看系统运行时间，可以看到他是自动重启了一次 如图 
[image]
系统日志中出现这个
Fri Jan  5 11:43:10 2024 kern.info kernel: [ 6594.307425] usb 2-1: reset high-speed USB device number 2 using ehci-platform
Fri Jan  5 11:43:10 2024 kern.info kernel: [ 6594.492494] sd 0:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_TIME_OUT driverbyte=DRIVER_OK cmd_age=30s
Fri Jan  5 11:43:10 2024 kern.info kernel: [ 6594.492530] sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 21 09 c9 08 00 00 f0 00
Fri Jan  5 11:43:10 2024 kern.err kernel: [ 6594.492554] blk_update_request: I/O error, dev sda, sector 554289416 op 0x0:(READ) flags 0x84700 phys_seg 30 prio class 0
Fri Jan  5 11:43:25 2024 daemon.err odhcpd[5368]: Failed to send to ff02::1%lan@br-lan (Address not available)
Fri Jan  5 11:43:40 2024 kern.info kernel: [ 6625.030527] usb 2-1: reset high-speed USB device number 2 using ehci-platform
Fri Jan  5 11:43:41 2024 kern.info kernel: [ 6625.210340] sd 0:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_TIME_OUT driverbyte=DRIVER_OK cmd_age=30s
Fri Jan  5 11:43:41 2024 kern.info kernel: [ 6625.210365] sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 30 4a 24 00 00 00 88 00
Fri Jan  5 11:43:41 2024 kern.err kernel: [ 6625.210382] blk_update_request: I/O error, dev sda, sector 810165248 op 0x0:(READ) flags 0x80700 phys_seg 17 prio class 0

bupahaibao, 2024-01-06:
https://github.com/istoreos/istoreos/issues/1236
看看我发布的问题，可能对你了解真正的问题有所帮助。

jjm2473, 2024-01-07:
@bupahaibao 看起来不是同一个问题，贴主的日志确实出现了块设备的io错误，应该跟网络无关。

dalaoshi777, 2024-01-08:
> #1236 看看我发布的问题，可能对你了解真正的问题有所帮助。

您好，我看了下，不过我并没有打开full cone，前几个月我已经将我的主路由刷回了官方系统，设备为红米ax6，官方固件无法打开full cone。感谢您的帮助

dalaoshi777, 2024-01-11:
> @bupahaibao 看起来不是同一个问题，贴主的日志确实出现了块设备的io错误，应该跟网络无关。

又经过了4天再次报错，不过出现的情况似乎有些不同这次首页中docker正常没有转圈圈
[image]
[image]
[image]
[image]
[内核日志.txt](https://github.com/istoreos/istoreos/files/13897444/default.txt)
[系统日志.txt](https://github.com/istoreos/istoreos/files/13897445/default.txt)

jjm2473, 2024-01-11:
外部的硬盘是不是已经掉了

dalaoshi777, 2024-01-12:
> 外部的硬盘是不是已经掉了

按道理应该不会掉，但是这次重启并没恢复，硬盘盒的灯是亮的，没有尝试断电重启，我把硬盘线拔了再插回去就好了

jjm2473, 2024-01-12:
> 重启并没恢复

更新到最新固件，杂项设置里面的“启动”页面配置下等待硬盘，这样可以避免开机时硬盘没挂载就启动docker或其他服务

dalaoshi777, 2024-01-12:
> > 重启并没恢复
> 
> 更新到最新固件，杂项设置里面的“启动”页面配置下等待硬盘，这样可以避免开机时硬盘没挂载就启动docker或其他服务

好的

dalaoshi777, 2024-01-15:
> > 重启并没恢复
> 
> 更新到最新固件，杂项设置里面的“启动”页面配置下等待硬盘，这样可以避免开机时硬盘没挂载就启动docker或其他服务

开启之后，还是遇到了同样的爆炸问题

dalaoshi777, 2024-01-16:
> > 重启也没恢复
> 
> 更新到最新固件，杂项设置里面的“启动”页面配置下等待硬盘，这样可以避免开机时硬盘没挂载就启动docker或其他服务

[image]
新版本iStore跑网心会出现这种问题，群里也有一个一摸一样的一天爆炸几次,原来的版本没那么严重原来大概5天一炸
[内核日志1.txt](https://github.com/istoreos/istoreos/files/13953585/1.txt)
[系统日志1.txt](https://github.com/istoreos/istoreos/files/13953586/1.txt)

jjm2473, 2024-01-17:
你们用的硬盘盒一样吗？或者是否同一个方案的？

执行`lsusb -tv`看看。

dalaoshi777, 2024-01-17:
> 你们的硬盘盒都一样吗？或者是否有同一个解决方案？
> 
> 执行`lsusb -tv`看看。

好的晚上看看，目前在测试是不是供电的问题，他给固态盒子加了外置电源。

dalaoshi777, 2024-01-17:
> 你们的硬盘盒都一样吗？或者是否有同一个解决方案？
> 
> 执行`lsusb -tv`看看。
硬盘盒不一样，他是迷虎牌子的，不过主控都是9210b

dalaoshi777, 2024-01-18:
> 你们的硬盘盒都一样吗？或者是否有同一个解决方案？
> 
> 执行`lsusb -tv`看看。

我换armbian也掉，群友外置电源也掉

jjm2473, 2024-01-18:
[image]

看你的lsusb已经不是使用UAS驱动，应该驱动没啥兼容性问题了

closed 2024-02-22 by dalaoshi777