istoreos/istoreos #1223: nanopi r5c 使用 mt7921 经常无法加入WIFI

tyinet, 2024-01-02:
一直挺喜欢istoreos 系统，最近买了一个新玩具 nanopi r5c 上了一块mt7921 WIFI 模块

刚开始用WIFI没啥问题，出外面一趟在回到家连WIFI，就一直提示无法加入WIFI，接网线直连又可以上网

WIFI配置都是默认的，就改了一个WIFI名称，以及配置了一个密码，重启nanopi r5c WIFI 又正常，但是不想总是经常重启它

尝试换原厂的 rk3568-XYZ-friendlywrt-23.05-docker-YYYYMMDD.img.gz  用了几天下来，很稳定没遇到过这个问题

jjm2473, 2024-01-02:
看看wifi设置的国家是不是中国，还有，如果是5G频段，使用157信道，太低的信道会因为雷达规避导致wifi启动需要一分钟，还有检查下供电是不是足的。我也跑一两天看看有没有问题。

tyinet, 2024-01-02:
好的，谢谢，我尝试看看。供电方面我觉得没问题。都是一样的供电 rk3568-XYZ-friendlywrt-23.05-docker-YYYYMMDD.img.gz  这个固件没得问题，我打了几天游戏了，很稳定。

jjm2473, 2024-01-02:
你用istoreos一般多久会出现wifi无法连接的情况

tyinet, 2024-01-02:
@jjm2473 大概半天的样子，然后手机、电脑设备再次连接WIFI的时候，就一直提示无法加入WIFI。连接不上去了。固件是最新的  istoreos-22.03.5-2023122916-r5s-squashfs-combined.img.gz

AlanWang2333, 2024-01-02:
这网卡我用着挺好用的，WIFI的加密方式最好一样，国家选中国，好像也没啥了

tyinet, 2024-01-02:
@AlanWang2333 你用的是哪个固件，方便发下嘛，我也试试呢

AlanWang2333, 2024-01-02:
> @AlanWang2333 你用的是哪个固件，方便发下嘛，我也试试呢

他系统自带的，已经集成了，官方文档搜，无线，有一堆推荐的网卡，我就是看官方文档里写着支持，我才买的

tyinet, 2024-01-03:
@AlanWang2333 你的也是nanopi r5c 设备？ 使用的 istoreos 哪个版本号呢？

AlanWang2333, 2024-01-03:
> @AlanWang2333 你的也是nanopi r5c 设备？ 使用的 istoreos 哪个版本号呢？

我用的j3160，这个上个版本就能用，这个最新版本更新了，又能用

jjm2473, 2024-01-03:
我测试了一天，没啥问题啊。（H68K也是RK3568，WIFI模块也是MT7921。）
[image]
[image]

是不是你的翻墙插件有问题？

jjm2473, 2024-01-03:
这是我的设置

[image]
[image]

tyinet, 2024-01-03:
@jjm2473 谢谢你，我没有安装科学工具，就是普通的连WIFI，打王者游戏。我尝试按照你的配置。在测试看看

tyinet, 2024-01-03:
@AlanWang2333 好的，应该是我的配置没配置好，我尝试换大佬推荐的配置，在配置测试看看

jjm2473, 2024-01-04:
运行两天，刚刚用一台手机重复连接多次，有一次连接失败，但是再点一下又连上了，之后重复测试又没出现失败。

AlanWang2333, 2024-01-04:
> 这是我的设置
> 
> <img alt="image" width="693" src="https://private-user-images.githubusercontent.com/3896703/293826635-a3747321-f79b-4558-9c3d-731eb8ddc020.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMzk2MDQsIm5iZiI6MTcwNDMzOTMwNCwicGF0aCI6Ii8zODk2NzAzLzI5MzgyNjYzNS1hMzc0NzMyMS1mNzliLTQ1NTgtOWMzZC03MzFlYjhkZGMwMjAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDRUMDMzNTA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGE2ZjQ4ZmRkMjAyMDZmMTU1ODRhOGRkODZmOGY0ODM1NWMyM2RjYzczZGZkNmJmYzFhNGJiOGNlNDA0NDYwMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.cszeHQQ6bY0iHlrnzyTvb-9jxsjUT8aVQSbPPr5dDYc"> <img alt="image" width="726" src="https://private-user-images.githubusercontent.com/3896703/293826659-7c0400fe-e7f0-41f6-8cf9-fb43c81b2c64.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMzk2MDQsIm5iZiI6MTcwNDMzOTMwNCwicGF0aCI6Ii8zODk2NzAzLzI5MzgyNjY1OS03YzA0MDBmZS1lN2YwLTQxZjYtOGNmOS1mYjQzYzgxYjJjNjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDRUMDMzNTA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWEyYTAxOWY2NzJkNjYzZDk2NmQxZGEyNzJkMjM1MWU4ZDgyZjZjZjhiYTk2MzYyNmFkODM2N2ZhYzhlMmJlZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.RSdGow0_GGVUs4aqvDSZP_GyOcH-5fsXDHb8reO6O90">

来闲鱼，我帮你改