istoreos/istoreos #1227: ipv6获取（续约）问题

ZYong9908, 2024-01-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：开启dhcpv6和slaac，windows首次连接可以正常获取dhcpv6和slaac，此时运行`ipconfig/renew6`正常。之后windows关机，隔一会(可能10分钟)重启(不重启有地址`ipconfig/renew6`问题依旧)，开机后只有一个本地ipv6地址，此时`ipconfig/renew6`会出现`在续订接口 WLAN 时出错: 信号灯超时时间已到`，但是可以获取到slaac的ipv6，无法获取到dhcpv6，查看istoreos`已分配的 DHCPv6 租约`中对应的`剩余租期`没有更新，依旧为首次获取dhcpv6时间计算而来。
进一步测试发现：
当使用duid固定ipv6后缀后，问题依旧，但此时删除固定，或开始未绑定，出问题后再绑定，再`renew6`可以获取到dhcpv6，但过一会问题依旧。
当之前获得的dhcpv6租约到期后，再次windows开机，会重复之前首次连接的现象
IP Helper服务  自动  运行正常

### (2) 路由器型号和固件版本

A：x86  
iStoreOS 22.03.5 2023122916

### (3) 详细日志和/或截图

A：执行`ipconfig/renew6`后系统日志中无输出

jjm2473, 2024-01-03:
试试重启光猫或者重新拨号

ZYong9908, 2024-01-03:
> 试试重启光猫或者重新拨号

光猫桥接，重新拨号/重启istoreos会重复之前首次连接的现象

ZYong9908, 2024-01-03:
> 试试重启光猫或者重新拨号

发现问题出现后在启动项中重启`odhcpd`会暂时解决，之后跟首次连接一样
```
root@iStoreOS:~# cat /etc/config/dhcp

config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
	option nonegcache '1'
	option allservers '1'
	option rebind_protection '0'
	option sequential_ip '1'

config dhcp 'lan'
	option interface 'lan'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	list dns '2400:3200::1'
	list dns '2402:4e00::'
	list dns '2001:4860:4860::8888'
	option ra 'server'
	option dhcpv6 'server'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	option start '10'
	option force '1'
	option ra_default '2'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
	list ra_flags 'none'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config host
	option name 'A30G'
	option mac '00*********18'
	option ip '192.168.100.2'

config host
	option mac '7C:*********3:1A'
	option ip '192.168.100.102'
	option name 'ZYong-K60'
	option dns '1'

config host
	option name 'ZYong-PC'
	option dns '1'
	option mac '4C:**:**:**:71:65'
	option ip '192.168.100.101'
	option duid '000100*****ba1f'
	option hostid '101'
```

Karmylr, 2024-01-03:
好像是 openwrt 的 odhcp 本身的问题

ZYong9908, 2024-01-03:
> 好像是 openwrt 的 odhcp 本身的问题

请问有没有解决/替代解决方案可以参考

ZYong9908, 2024-01-03:
看到一个odhcpd的issue可能相关，但是我看不很明白
https://github.com/openwrt/odhcpd/issues/197

jjm2473, 2024-01-05:
试试这个 https://github.com/istoreos/istoreos/issues/1172#issuecomment-1857547892

ZYong9908, 2024-01-05:
> 试试这个 #1172 (comment)

这个试过了的，问题依旧，同时wan,wan6.wan_6变成了wan,wan_6没发现影响。
目前已买其他品牌无线路由器测试是否为做ap的路由器有问题

ZYong9908, 2024-01-05:
更换其他品牌无线路由器后测试满一小时，暂未出现问题
原ap路由器：水星A30G，有ipv6需求用户要注意

jjm2473, 2024-01-06:
原本使用的ap是什么？

ZYong9908, 2024-01-06:
> 原本使用的ap是什么？

水星A30G

jjm2473, 2024-01-06:
好吧，我看错了，还以为是换成了 水星A30G 🤣

jjm2473, 2024-01-06:
一般情况下，不管什么AP，使用有线桥接应该就不会有问题，也就是关掉AP的DHCP服务，然后AP和软路由用LAN to LAN相连，就行了。有些路由器的AP模式可能有问题，例如华硕路由器的AP模式至少性能有问题，所以最好使用有线桥接。

ZYong9908, 2024-01-06:
> 一般情况下，不管什么AP，使用有线桥接应该就不会有问题，也就是关掉AP的DHCP服务，然后AP和软路由用LAN to LAN相连，就行了。有些路由器的AP模式可能有问题，例如华硕路由器的AP模式至少性能有问题，所以最好使用有线桥接。

有线桥接，关dhcp同时开关ipv6等都试过，都不行，换了一个后有线桥接啥问题也没有

jjm2473, 2024-01-06:
神奇，理论上有线桥接跟交换机一样，不会对流量进行管控

ZYong9908, 2024-01-06:
问过水星客服，说的是ap或者关dhcp之后，都是向上层路由获取，我感觉是他们整体逻辑有点问题

closed 2024-01-06 by jjm2473

138161, 2024-02-26:
先用电脑接，看有几个网关，我一个光猫出来两个网关
在路由器上ip -6 route看一下自己的默认网关。如果获得的是

default from 2402:f000:x:xxxx::/64 via fe80::xxxx:xxxx:xxxx:xxxx dev eth0 proto static metric 512

这样坑爹的网关，在转发NAT包的时候会有问题，需要把去掉from 2402:f000:x:xxxx::/64这一部分的以后的默认路由添加到路由表中。可以新建一个/etc/hotplug.d/iface/99-ipv6，它的内容是
有效
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
iface=wan6
[ -z "$iface" -o "$INTERFACE" = "$iface" ] || exit 0
ip -6 route add ip -6 route show default|sed -e 's/from [^ ]* //'
logger -t IPv6 "Add IPv6 default route."

这里iface是LuCI里面Network->Interfaces里面看到的名字，一般叫wan6。这个脚本的意思是在wan6起来以后读取默认网关，把带from的内容去掉，再加到系统路由表里。记得
chmod +x /etc/hotplug.d/iface/99-ipv6

有效
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
hwaddr=ifconfig |ip -6 route show default | head -n1 |awk '{print $5}'
route -A inet6 add default gw $hwaddr dev eth0

klaihao, 2024-04-09:
把组播关掉试试，我华为光猫也是这个问题，以前正常，后来有一天开了组播，之后就不行了。刚刚把组播关掉，重启网络连接，暂时是好了

138161, 2024-04-09:
https://github.com/istoreos/istoreos/issues/1480
似乎解决问题

ZYong9908, 2024-04-26:
> #1480 似乎解决问题

很遗憾，已无法测试，原A30G已经不再，且更换之后再无问题，故不打算回购测试